espressif/arduino-esp32 #6013: ESP32-C3 bootloop when flashing any Arduino sketch

adumont, 2021-12-12:
My ESP32-C3 is bootlooping when flashing any Arduino sketch.

**Board**: "Chip is ESP32-C3 (revision 2)" (as seen in Arduino IDE flash log)
**Version/Date**: "ESP-ROM:esp32c3-20200918 / Build:Sep 18 2020" (as seen in Serial output)
**IDE name**: Arduino IDE 1.8.13
**Flash Frequency**: 80MHz (unchanged)
**PSRAM enabled**: Default 4MB with SPIFFS (unchanged)
**Upload Speed**: 921600 (unchanged)
**Computer OS**: Ubuntu 16.04.7 LTS 

### Description:
My ESP32-C3 is bootlooping when flashing any Arduino sketch.

Serial output looks like this (continuously):

```ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d0f8a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x484
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2398
SHA-256 comparison failed:
Calculated: e0c357829f601add6a50d6ea19f89687ec5acf531610506a38b33d09b4a71c7e
Expected: 77332826ec7bc491df854452f675071e3a094e80858dd7d91dd7e27940204216
Attempting to boot anyway...
entry 0x403ce000
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
[...]
```
### Sketch:

It happens when I flash any sketch, even this "blank" sketch:

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Arduino Upload Log

```
/home/adumont/opt/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/adumont/opt/arduino-1.8.13/hardware -hardware /home/adumont/.arduino15/packages -tools /home/adumont/opt/arduino-1.8.13/tools-builder -tools /home/adumont/opt/arduino-1.8.13/hardware/tools/avr -tools /home/adumont/.arduino15/packages -built-in-libraries /home/adumont/opt/arduino-1.8.13/libraries -libraries /home/adumont/Arduino/libraries -fqbn=esp32:esp32:esp32c3:CDCOnBoot=default,PartitionScheme=default,CPUFreq=160,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10813 -build-path /tmp/arduino_build_460387 -warnings=none -build-cache /tmp/arduino_cache_677574 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/adumont/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2.path=/home/adumont/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=/home/adumont/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2.path=/home/adumont/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.mklittlefs.path=/home/adumont/.arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=/home/adumont/.arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=/home/adumont/.arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2.path=/home/adumont/.arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.esptool_py.path=/home/adumont/.arduino15/packages/esp32/tools/esptool_py/3.1.0 -prefs=runtime.tools.esptool_py-3.1.0.path=/home/adumont/.arduino15/packages/esp32/tools/esptool_py/3.1.0 -prefs=runtime.tools.mkspiffs.path=/home/adumont/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/adumont/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /tmp/untitled1378934422.tmp/sketch_dec12c/sketch_dec12c.ino
/home/adumont/opt/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/adumont/opt/arduino-1.8.13/hardware -hardware /home/adumont/.arduino15/packages -tools /home/adumont/opt/arduino-1.8.13/tools-builder -tools /home/adumont/opt/arduino-1.8.13/hardware/tools/avr -tools /home/adumont/.arduino15/packages -built-in-libraries /home/adumont/opt/arduino-1.8.13/libraries -libraries /home/adumont/Arduino/libraries -fqbn=esp32:esp32:esp32c3:CDCOnBoot=default,PartitionScheme=default,CPUFreq=160,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10813 -build-path /tmp/arduino_build_460387 -warnings=none -build-cache /tmp/arduino_cache_677574 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/home/adumont/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2.path=/home/adumont/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=/home/adumont/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2.path=/home/adumont/.arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.mklittlefs.path=/home/adumont/.arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=/home/adumont/.arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=/home/adumont/.arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2.path=/home/adumont/.arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2 -prefs=runtime.tools.esptool_py.path=/home/adumont/.arduino15/packages/esp32/tools/esptool_py/3.1.0 -prefs=runtime.tools.esptool_py-3.1.0.path=/home/adumont/.arduino15/packages/esp32/tools/esptool_py/3.1.0 -prefs=runtime.tools.mkspiffs.path=/home/adumont/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/home/adumont/.arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /tmp/untitled1378934422.tmp/sketch_dec12c/sketch_dec12c.ino
Using board 'esp32c3' from platform in folder: /home/adumont/.arduino15/packages/esp32/hardware/esp32/2.0.1
Using core 'esp32' from platform in folder: /home/adumont/.arduino15/packages/esp32/hardware/esp32/2.0.1
bash -c "[ ! -f \"/tmp/untitled1378934422.tmp/sketch_dec12c\"/partitions.csv ] || cp -f \"/tmp/untitled1378934422.tmp/sketch_dec12c\"/partitions.csv \"/tmp/arduino_build_460387\"/partitions.csv"
bash -c "[ -f \"/tmp/arduino_build_460387\"/partitions.csv ] || [ ! -f \"/home/adumont/.arduino15/packages/esp32/hardware/esp32/2.0.1/variants/esp32c3\"/partitions.csv ] || cp \"/home/adumont/.arduino15/packages/esp32/hardware/esp32/2.0.1/variants/esp32c3\"/partitions.csv \"/tmp/arduino_build_460387\"/partitions.csv"
bash -c "[ -f \"/tmp/arduino_build_460387\"/partitions.csv ] || cp \"/home/adumont/.arduino15/packages/esp32/hardware/esp32/2.0.1\"/tools/partitions/default.csv \"/tmp/arduino_build_460387\"/partitions.csv"
bash -c "[ -f \"/tmp/untitled1378934422.tmp/sketch_dec12c\"/bootloader.bin ] && cp -f \"/tmp/untitled1378934422.tmp/sketch_dec12c\"/bootloader.bin \"/tmp/arduino_build_460387\"/sketch_dec12c.ino.bootloader.bin || ( [ -f \"/home/adumont/.arduino15/packages/esp32/hardware/esp32/2.0.1/variants/esp32c3\"/bootloader.bin ] && cp \"/home/adumont/.arduino15/packages/esp32/hardware/esp32/2.0.1/variants/esp32c3\"/bootloader.bin \"/tmp/arduino_build_460387\"/sketch_dec12c.ino.bootloader.bin || cp -f \"/home/adumont/.arduino15/packages/esp32/hardware/esp32/2.0.1\"/tools/sdk/esp32c3/bin/bootloader_qio_80m.bin \"/tmp/arduino_build_460387\"/sketch_dec12c.ino.bootloader.bin )"
bash -c "[ ! -f \"/tmp/untitled1378934422.tmp/sketch_dec12c\"/build_opt.h ] || cp -f \"/tmp/untitled1378934422.tmp/sketch_dec12c\"/build_opt.h \"/tmp/arduino_build_460387\"/build_opt.h"
bash -c "[ -f \"/tmp/arduino_build_460387\"/build_opt.h ] || touch \"/tmp/arduino_build_460387\"/build_opt.h"
Detecting libraries used...
/home/adumont/.arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2/bin/riscv32-esp-elf-g++ -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4-dev-3569-g6a7d83af19-dirty\"" -DESP_PLATFORM -I/home/adumont/.arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32c3/include/config -I/home/adumont/.arduino15/packages/esp32/hardware/esp32/2.0.1/tools/sdk/esp32c3/include/newlib/platform_include -I/home/adumont/.arduino15/packages/esp32/hardware/esp32 [... cut ...]
Writing at 0x00019a89... (33 %)
Writing at 0x0001f8ea... (50 %)
Writing at 0x0002592a... (66 %)
Writing at 0x0002bdf9... (83 %)
Writing at 0x000338cc... (100 %)
Wrote 166928 bytes (93052 compressed) at 0x00010000 in 2.8 seconds (effective 475.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 446.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```

adumont, 2021-12-12:
When I use [cpq/mdk](https://github.com/cpq/mdk) (Minimal Dev Kit) I can flash the example c3ws2812 that blinks the rgb led on the board. 

When I then open Arduino IDE, and try to flash the empty sketch, the on board LED stops flashing, but is still powered ON even after the `Leaving... Hard resetting via RTS pin...`, so I'm wondering if the Arduino IDE does actually reset the ESP32-C3 or even flash it correctly... (despite not seeing eny error in the Arduino IDE log)

igrr, 2021-12-13:
Hi @adumont, the current version of arduino-esp32 supports only ESP32-C3 revision 3. It is not built with support for revision 2 or earlier. It might still be possible to use a revision 2 board with ESP-IDF, enabling CONFIG_ESP32C3_REV_MIN_2.

atanisoft, 2021-12-14:
@igrr would there be any issues downgrading to `CONFIG_ESP32C3_REV_MIN_2` instead of `CONFIG_ESP32C3_REV_MIN_3`? Is there any negative change to IDF behavior with using a rev3 silicone?

sofakng, 2021-12-15:
Glad I found this issue... I spent the past few hours troubleshooting.  :(

thinhx2, 2022-01-13:
I have same problem, IDF still okay with change config revision to 2. But Arduino IDE just work with revision 3.
Checked with 4 espressif c3 devkit (4 rev 2), 2 aithinker  esp32-c3s (1 rev 3,1 rev 2), 4 lilygo c3 (4 rev 3)

bb-pnw, 2022-01-21:
I just encountered this same boot-loop problem on a pair of ESP32-C3 (revision 2) chips on early ESP32-C3-DevKitM-1 boards.

-- I'm able to build and boot Arduino sketches by reverting to the Arduino-esp32 v2.0.0 library.  Not a perfect solution, but at least these early boards won't go to waste ...

adumont, 2022-01-22:
Thanks @bb-pnw! I have tried reverting to 2.0.0 and indeed it works for my ESP32-C3-DevKitM-1 board! At least I can use it.

VojtechBartoska, 2022-04-11:
Closing this one as it was covered already. Thanks for your contribution.

closed 2022-04-11 by VojtechBartoska

QGB, 2022-05-15:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403814ce
SPIWP:0xee
mode:DIO, clock
```

HugoML, 2022-06-06:
The way I solved this for ESP32-C3:
In Arduino 1.8.16,  Tools/Flash Mode/DIO

libreleee, 2025-05-16:
const int led = 8; <---   // const int led = 13 -> 8 LED  line 11

The led is connected to pin 8 and pin 13 seems to be related to Reset
When I change it to pin 8, the LED works and it is visible in the web browser.

[image]