espressif/arduino-esp32 #6015: SerialToSerialBTM with Password Pairing?

Manasmw01, 2021-12-13:
I wanted to use the SerialToSerialBTM code with the password pairing feature.
Here's the [LINK](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBTM/SerialToSerialBTM.ino)
I can see a lot of tutorials for Bluetooth password pairing in esp-idf, but not for Arduino IDE.
I'm in a hurry and there's no time to learn esp-idf settings and stuff. 
If someone has tried it already, please help me out.

st3m0r, 2021-12-13:
As far as I know you need to use arduino as component and disable ssp

Manasmw01, 2021-12-25:
I tried that, but it works only with ESP IDF v4.4 I guess. And was asking for a lot of updates.
I tried another method which is somehow working.

Manasmw01, 2021-12-25:
I've tried something and created a new issue. Please follow that: [LINK](https://github.com/espressif/arduino-esp32/issues/6061#issue-1088610953)

SuGlider, 2022-02-10:
Try calling ```setPin()``` before ```begin()```.

``` dart
void setup() {
  Serial.begin(115200);
  SerialBT.setPin(pin);    // setPin() shall come before begin()
  SerialBT.begin("ESPBT", true); 
  Serial.println("The device started in master mode, make sure remote BT device is on!");
  
  connected = SerialBT.connect(address);
  
  if(connected) {
    Serial.println("Connected Succesfully!");
  } else {
    while(!SerialBT.connected(10000)) {
      Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app."); 
    }
  }
}
```

VojtechBartoska, 2022-04-07:
Hello @Manasmw01, can we close this one a follow this issue in your new one #6061?

Manasmw01, 2022-04-08:
Yes, we can. Refer to #6061.