espressif/arduino-esp32 #6016: BLE does not reconnect after disconnect version 1.05 and up

Zparq, 2021-12-13:
Using example code provided for BLE Server, the code compiles and runs. However, with version 1.05 and newer, it cannot reconnect BLE after a disconnect event initiated from an Android device. For example if I pair the ESP32 with the phone and connect, everything works fine. I then disable the bluetooth on the android for 15 seconds and then reenable it. After this it is impossible to reconnect to the ESP32. However it is still working with version 1.04.

hyperlite2, 2022-01-12:
I can confirm that this is the case. 1.0.4 works well to reconnect BT after disconnect, but all future versions seem to break the ability to reconnect.

Dork57, 2022-02-22:
I can confirm, too. No advertising after disconnect.

me-no-dev, 2022-02-22:
have you tried restarting the advertisement after disconnect?

Dork57, 2022-02-22:
> have you tried restarting the advertisement after disconnect?

No, but you're right. Thanks. I just found the solution also in the example code of the BLE Lib.
`pServer->startAdvertising(); // restart advertising after disconnecting`

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

RodrigoALippi, 2022-04-26:
Hi!!
 By Espressif System 2.0.3-RC1...

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };
 
    void onDisconnect(BLEServer* pServer) {
      deviceConnected = false;
      pServer->getAdvertising()->start();
    }
};

Enjoy!

VojtechBartoska, 2022-04-27:
We will take a look on this, adding to our Roadmap.

gentigace, 2022-07-03:
@RodrigoALippi many thanks man this solve also my issue.

closed 2025-01-16 by Parsaabasi