espressif/arduino-esp32 #6017: cores behave differently

mkohns, 2021-12-13:
### Hardware:
PLATFORM: Espressif 32 (2.1.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.201016 (1.0.4) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - tool-openocd-esp32 2.1000.20201202 (10.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)

Board:							Wemos Mini D1 ESP32
Core Installation version:			do not know how to figure this out
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                      no
Upload Speed:					115200
Computer OS:                                           Mac OSX

### Description:
From google I got the impression that arduino setup() and loop() is running on core 1. I can proof this via xPortGetCoreID(). 
I have a FastLED application where I do all WIFI stuff on core1 and spin up a FreeRTOS task on core 0 via xTaskCreatePinnedToCore to process the FastLED stuff. I encountered, that it  "stutters" on core 0. To proof this, I included a FPS watcher, and the visual stuttering can be proofed by a falling FPS. 
I was wondering why this could be. The intention was to separate cores (1 for wifi, 0 for app) in order to get independent.
I disabled WIFI on core 0 - but stuttering on core 1 persists. So no WIFI interference.
I tested this on 4 different ESP32. All behaved the same.
Interesting to me was, that the stuttering appeared periodically.

I tried to strip down the problem to its core. Also got rid of FastLED. Now - its very easy.

### Sketch:  
```cpp
#include <Arduino.h>

void task1 (void * pvParameters ){
  unsigned long counter = 0; 
  unsigned long linecounter = 0; // just for drawing the charts
  const unsigned long REFRESH_INTERVAL = 1000000; // 1 ms
  unsigned long previousMicros = 0;
  for (;;) {
    unsigned long currentMicros = micros();
    if(currentMicros - previousMicros >= REFRESH_INTERVAL)  {
      Serial.printf("%lu;%lu;%lu\n", linecounter, currentMicros - previousMicros, counter);
      previousMicros = micros();
      counter = 0;
      linecounter++;
    }
    counter++;
  }
}

void setup() {
  Serial.begin(115200);
  while (!Serial); // wait for serial attach  
  Serial.printf("Hello From ESP32 on Core: %d\n",xPortGetCoreID());

  // disable WDT so we do not need to deal with VTaskDelay
  disableCore0WDT(); 
  disableCore1WDT();

  // switch below to run on core 0 or 1
  //xTaskCreatePinnedToCore(task1, "task1",  10000,  NULL,  10000, NULL,  1);
  xTaskCreatePinnedToCore(task1, "task1",  10000,  NULL,  10000, NULL,  0);
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

This produces this chart for core 1 (the arduino standard one)
[image]

And this chart for core 0
[image]

Please mark the different y-axis ranges.

My questions:
1) I assume in this setup nothing but my task and a system "idle" task in running on core 1. Is this true? I can not get VTaskList running to proof.
2) I assume in this setup, the Free RTOS Scheduler will never schedule any internal tasks on core1. Is this true?
3) As you can see, there are performance peaks every around 50s. How could this be introduced?  The performance peaks are "positive", the loop can pass more often (counter is bigger). 
4) Core 1 (the default core) does not have any "negative" performance peaks,  and is in total more steady. Core 0 also has "negative" performance peaks, and i assume (maybe wrong) that these are the ones I see with FastLED. How does this difference on the two cores can be explained? What can I do?

Thanks for reading and thinking!

mkohns, 2021-12-13:
To be complete: Here is a gist showing the "stuttering" on core 0 and the "no stuttering" on core 1 with FastLED. 

https://gist.github.com/mkohns/a0b87cc5dbbedda9d2d3a796a82e264d

You can run the code (and reproduce the issue) also without an LED matrix installed. 

On core 1 - you will see 2 FPS messages until the loop has settled. Then you can run it for hours without any further line.
On core 0 - you will also see 2 FPS messages until the loop has settled. But you will get new lines around every 30s-1m looking like this

```
FPS: 203
FPS: 196
FPS: 203
FPS: 196
FPS: 203
FPS: 196
FPS: 196
FPS: 203
FPS: 203
FPS: 196
```

If you would have an LED Matrix, you could see the stuttering -  when message occurs.
You may wonder the high FPS of 200 frame per second. I just have chosen this for easy reproduction. This also occurs on FPS 60. But you need to wait longer. This is how I discovered it.

igrr, 2021-12-13:
This might not explain the issue you are seeing with FastLED, but I think the issue with this test is that in 1.0.4, `micros` itself may take different amount of time depending on some conditions, and might be a bit impercise. Up until arduino-esp32 version 2.0.0, the underlying version of ESP-IDF used a 32-bit timer as the timekeeping source for `esp_timer_get_time` (and hence `micros`), which required software handling of overflows. The overflows occurred every 0xefffffff/(80 MHz), which is 50.3 seconds — roughly the period that can be seen in your measurements. The code to handle reading the timer value was [a bit involved](https://github.com/espressif/esp-idf/blob/85c43024cb133cd81d3e3a95c1da49d51291d306/components/esp32/esp_timer_esp32.c#L186-L209) and was imprecise around the points when overflow happened.

In 2.0.0 and later Arduino releases, the underlying version of ESP-IDF has switched to a 64-bit timer which no longer requires software handling of overflows, since it (for all practical purposes) never overflows. The timing of a single `esp_timer_get_time` call (and hence `micros`) also varies less now, at least in the case of your simple example.

I think we need to run this test with latest arduino-esp32 version and see if the results are any different.

mkohns, 2021-12-13:
This sounds interesting to me. Am I able to help here?
Can I upgrade to > 2.0.0 within platform.io? I already googled a bit - but perhaps you know best. I already tried to enhance my platformio.ini with

```
platform_packages =
    toolchain-xtensa32 @ ~2.80400.0
    framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.1
    platformio/tool-esptoolpy @ ~1.30100
```

But this did not do the trick:

```
Linking .pio/build/wemos_d1_mini32/firmware.elf
/Users/xyz/.platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
*** [.pio/build/wemos_d1_mini32/firmware.elf] Error 1
```

Perhaps there is a workaround I could manage?

igrr, 2021-12-13:
@mkohns I don't have experience with that myself, but it seems that this might be possible, see the recent comments in https://github.com/platformio/platform-espressif32/issues/619#issuecomment-976525911.

pablozg, 2021-12-13:
> 

This works for me.

platform                    = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master ; Core 2.0.1
platform_packages           = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git ; SDK 4.4
                             platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.2/esptool-v3.2.zip

mkohns, 2021-12-13:
Okay, the above - did not work for me. Same linker error. I got it working with this one here:

```
platform =  https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
    toolchain-xtensa32 @ ~2.80400.0
    platformio/tool-esptoolpy @ ~1.30100
```

This produces this compile output

```
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.d218e58 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - tool-openocd-esp32 2.1000.20201202 (10.0) 
 - toolchain-xtensa-esp32 8.4.0+2021r2 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
```
@igrr : Is this what you intended to test?

The result is confusing to me:
On Core 1 (default core) - it looks like this

[image]
The "down" spikes have a delta of approx. 670 counts
If you zoom into the top base-line you see

[image]
The deviation is quite small with +- 1 only dancing around 1.549.666

On Core 0, it now looks completely different:
 [image]
No peaks to top or bottom. A base line at around 1.535.249 (15.000 counts slower than core 1). And a deviation of -6.5 and +4.5.

I can not interpret the result. Can you?

I also tested the FastLED stuff with this. This is now even more confusing to me:

On Core1: It is just the same. Stable 200 FPS. No problems. (although these deep peaks are there)
On Core0: It does not hit the 200FPS anymore. It cycles between 201(!) and 196. Stuttering is different but visible.

VojtechBartoska, 2022-04-07:
Hello @mkohns, thanks for testing. We have missed that, did you try it also on v2.0.3-RC1?

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

closed 2022-04-22 by VojtechBartoska