espressif/arduino-esp32 #6019: CONFIG_ARDUINO_RUNNING_CORE on ESP32-S2

misery, 2021-12-14:
I tried newest esphome with arduino-esp32 2.0.1. They check ```CONFIG_ARDUINO_RUNNING_CORE == 1``` now and it fails as there is no ```disableCore1WDT```.
It seems that ```CONFIG_ARDUINO_RUNNING_CORE``` on ESP32-S2 needs to be ```0``` instead of ```1``` like ESP32-C3 as it is single core, too.

```
src/esphome/components/esp32/core.cpp: In function 'void esphome::arch_init()':
src/esphome/components/esp32/core.cpp:50:3: error: 'disableCore1WDT' was not declared in this scope
   disableCore1WDT();
   ^~~~~~~~~~~~~~~
src/esphome/components/esp32/core.cpp:50:3: note: suggested alternative: 'disableCore0WDT'
   disableCore1WDT();
   ^~~~~~~~~~~~~~~
   disableCore0WDT
```

```cpp
#if CONFIG_ARDUINO_RUNNING_CORE == 0
  disableCore0WDT();
#endif
#if CONFIG_ARDUINO_RUNNING_CORE == 1
  disableCore1WDT();
#endif
```

See esphome issue: https://github.com/esphome/issues/issues/2816

Jason2866, 2021-12-16:
It is inconsistent and should be changed. I changed this setting in sdkconfig for the S2 and the builded Arduino framework does work with the S2

me-no-dev, 2021-12-17:
So this is some code that breaks in esphome and is part of esphome? They can check if the chip is unicore and set the running core to 0 or do not call the functions, just like here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal.h#L109

oxan, 2021-12-17:
> So this is some code that breaks in esphome and is part of esphome?

We could workaround it in ESPHome (but it'll be an ugly `#ifdef` mess), but it doesn't make any sense to have `CONFIG_ARDUINO_RUNNING_CORE` set to `1` on a device that doesn't have core `1`. Why not fix that in Arduino? That'll also help others that use this define in the future.

Jason2866, 2021-12-17:
It just needs changes in this part https://github.com/espressif/esp32-arduino-lib-builder/blob/master/sdkconfig.esp32s2#L142-L162
and with the next lib build run it is consistent with the C3 build

me-no-dev, 2021-12-20:
Because we are about to release and we will not be updating the IDF libs, we need to change this in both places (here and the lib builder)

closed 2021-12-20 by me-no-dev