espressif/arduino-esp32 #6020: A few notes for WiFiSTA performance

YordanYanakiev, 2021-12-14:
### Related area

WiFi

### Hardware specification

ESP32/ESP8266

### Is your feature request related to a problem?

One of the root of the performance issues related to the Wifi.

### Describe the solution you'd like

In WiFiSTA.cpp the user "**mlord**" ( **Mark Lord** ) ( **https://github.com/mlord** ) discovered that the performance issues is related with incoherent with the normal setup initialization in the AUTHMODE.
```
    if(password != NULL && password[0] != 0){
               wifi_config->sta.threshold.authmode = WIFI_AUTH_WEP;  
                _wifi_strncpy((char*)wifi_config->sta.password, password, 64);
```
this is incoherent with the nowdays most used WPS2 PSK, and therefore it should be :
```
    if(password != NULL && password[0] != 0){
               wifi_config->sta.threshold.authmode = WIFI_AUTH_WPA2_PSK;  
                _wifi_strncpy((char*)wifi_config->sta.password, password, 64);
```
aside of that the current scan overdo it with Scanning all chaneles:
```
wifi_config->sta.scan_method = WIFI_ALL_CHANNEL_SCAN;
```
and it should be the fast scan:
```
wifi_config->sta.scan_method = WIFI_FAST_SCAN;  
```

these changes drop significally the time for connection from 3 to 17 seconds on some setups down to almost instantly
( the problem discussed in here : https://github.com/espressif/arduino-esp32/issues/2501#issuecomment-990113298 )

me-no-dev, 2021-12-14:
a change to do fast scan if channel is selected was just added today. The other change I just added here: https://github.com/espressif/arduino-esp32/pull/6022

closed 2021-12-14 by me-no-dev

mlord, 2021-12-31:
This fix in #6022 doesn't change anything -- best to revert it.

The authmode gets overwritten with WIFI_AUTH_WEP a few lines further on, which is where the code should get changed to select WIFI_AUTO_WPA2_PSK instead.

VojtechBartoska, 2022-01-04:
@YordanYanakiev and @me-no-dev, can you please validate this changes?