espressif/arduino-esp32 #6021: SD card error

bhupiister, 2021-12-14:
### Board

esp32

### Device Description

M5 core2 based on esp32

### Hardware Configuration

Nothing connected

### Version

latest master

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

standard

### PSRAM enabled

yes

### Upload speed

115200

### Description

`In file included from C:/Users/user/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/SD_MMC/src/SD_MMC.cpp:24:0:
C:/Users/user/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/SD_MMC/src/SD_MMC.cpp: In member function 'bool fs::SDMMCFS::begin(const char*, bool, bool, int)':
C:/Users/user/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/driver/include/driver/sdmmc_host.h:119:1: sorry, unimplemented: non-trivial designated initializers not supported
 }
 ^
C:/Users/user/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/SD_MMC/src/SD_MMC.cpp:45:39: note: in expansion of macro 'SDMMC_SLOT_CONFIG_DEFAULT'
     sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
                                       ^
C:/Users/user/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/sdk/esp32/include/driver/include/driver/sdmmc_host.h:119:1: sorry, unimplemented: non-trivial designated initializers not supported
 }
 ^
C:/Users/user/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/libraries/SD_MMC/src/SD_MMC.cpp:45:39: note: in expansion of macro 'SDMMC_SLOT_CONFIG_DEFAULT'
     sdmmc_slot_config_t slot_config = SDMMC_SLOT_CONFIG_DEFAULT();
                                       ^
*** [.pio\build\m5stack-core2\lib133\SD_MMC\SD_MMC.cpp.o] Error 1`

### Sketch

```cpp
Basic read-write sketch
```

### Debug Message

```plain
Program cannot build
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2021-12-14:
@bhupiister Can you share more information on your PlatformIO setup? The error you are reporting indicates that there are some missing initialization parameters in the referenced structure. This is not seen on arduino-esp32 master which builds against ESP-IDF v4.4.

bhupiister, 2021-12-14:
@atanisoft 
I have initialized it by simply writing
    SD.begin(TFCARD_CS_PIN, SPI, 40000000);

atanisoft, 2021-12-14:
Comment out the `platform_packages` line and add:
```
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
```
This will bring in the full package of dependencies etc from the 2.0.x code line.

bhupiister, 2021-12-14:
@atanisoft the part after the hash in your git link is greened out. Is that intended?

bhupiister, 2021-12-14:
@atanisoft 
Its giving weird errors now
command 'platformio-ide.upload' not found
command 'platformio-ide.clean' not found

atanisoft, 2021-12-14:
Double check that you don't have `platform` listed twice in the ini file, I just realized it was already there and I had suggested to add instead of update.

atanisoft, 2021-12-14:
> the part after the hash in your git link is greened out. Is that intended?

That would be a parse/render issue in VSCode (or similar) and not an issue for PIO runtime.

bhupiister, 2021-12-14:
No that is not an issue, i did take care of that
here is the current config

platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = m5stack-core2
framework = arduino
monitor_speed = 115200
monitor_filters = log2file, default, esp32_exception_decoder
build_type = debug
build_flags =   -DBOARD_HAS_PSRAM 
                -mfix-esp32-psram-cache-issue                
lib_deps = 
        https://github.com/yururi-nonbiry/I2C_MPU6886_4kHz.git
        m5stack/M5Core2 @ ^0.0.9
        https://github.com/lvgl/lv_arduino.git
        https://github.com/Depau/ESPAsyncWebServer.git
        https://github.com/mobizt/Firebase-ESP32.git

bhupiister, 2021-12-14:
@atanisoft Can you help with above info

atanisoft, 2021-12-14:
I'm not sure what might be broken in PIO, I'm not seeing this when I'm building locally.

bhupiister, 2021-12-14:
@atanisoft 
`platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
`
it did work after I gave time to platformIO to solve dependencies
And there are no errors before compiling. However, at the time of building, it runs out of memory fails to build. It says the program is too big. 

All I did was to change the platform to the one you recommended, is it because of a different type of FreeRTOS? I remember that the default platform has the optimized version of FreeRTOS. Or is there some sort of flag related to bord type that I am missing to define?

atanisoft, 2021-12-14:
> is it because of a different type of FreeRTOS?

Arduino ESP32 2.0.x does require a bit more space than earlier versions, but it shouldn't be a problem in most cases.

> However, at the time of building, it runs out of memory fails to build. It says the program is too big.

This indicates that the build did succeed but failed in the PIO check of the binary fitting into the partition size, this is likely a problem with the board definition on the PIO side.

bhupiister, 2021-12-15:
@atanisoft 
**A huge thanks man for the prompt reply !!!** 

I was finally able to solve my issues with the library you suggested.
`platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream`
My other problem was that I was using ESPAsyncWebServer by `Depu `but I moved back to the one by `me-no-dev`

Also, I had to change the default partition table as by default it had configured a very small flash size

```
Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x640000,
app1,     app,  ota_1,   0x650000,0x640000,
spiffs,   data, spiffs,  0xc90000,0x370000,
```
So finally I was able to make the `xRingbufferCreateStatic `with help of the new version of FreeRTOS

```
    //Allocate memory for ring buffer data structure
    StaticRingbuffer_t *buffer_struct = (StaticRingbuffer_t *)heap_caps_malloc(sizeof(StaticRingbuffer_t), MALLOC_CAP_SPIRAM);
    //Allocate memory for ring buffer's data storage area
    uint8_t *buffer_storage = (uint8_t *)heap_caps_malloc(/* sizeof(uint8_t) * */ buffSizeInBytes, MALLOC_CAP_SPIRAM);
    //Set your buffer type
    RingbufferType_t buffer_type = RINGBUF_TYPE_BYTEBUF;
    buf_handle = xRingbufferCreateStatic(buffSizeInBytes, buffer_type, buffer_storage, buffer_struct);
```
I am leaving this code here, in case someone comes finding the solution for the above issue.

closed 2021-12-15 by VojtechBartoska