espressif/arduino-esp32 #6027: esp32c3  chip revision: 3   wifi.begin  crash

3050311118, 2021-12-15:
### Board

ESP32C3

### Device Description

ESP32C3 chip revision: 3

### Hardware Configuration

NO  

### Version

latest master

### IDE Name

ARDUINO IDE

### Operating System

WINDOWS

### Flash frequency

80M

### PSRAM enabled

no

### Upload speed

115200

### Description

WIFI.begin  crash . when connectting to wifi router
bluetooth OK

### Sketch

```cpp
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>

const char* ssid = "xxxxx";
const char* password = "yyyyyy";

WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp32!");
  digitalWrite(led, 0);
}

void handleNotFound() {
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void) {
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp32")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
  delay(2);//allow the cpu to switch to other tasks
}
```

### Debug Message

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1428
load:0x403ce000,len:0xc04
load:0x403d0000,len:0x292c
SHA-256 comparison failed:
Calculated: 9f7363434bc7a1a2434ba3062500fa10b9fce5bb859899ee0424321b4ddaf742
Expected: 9b18b42e3e8e407f5e7b13f26c80172eda36d674c584e818f50843c766ebde69
Attempting to boot anyway...
entry 0x403ce000
I (53) boot: ESP-IDF v4.4-dev-2313-gc69f0ec32 2nd stage bootloader
I (53) boot: compile time 12:10:14
I (54) boot: chip revision: 3
I (54) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (60) qio_mode: Enabling default flash chip QIO
I (64) boot.esp32c3: SPI Speed      : 80MHz
I (68) boot.esp32c3: SPI Mode       : QIO
I (72) boot.esp32c3: SPI Flash Size : 4MB
W (76) boot.esp32c3: PRO CPU has been reset by WDT.
I (80) boot: Enabling RNG early entropy source...
I (85) boot: Partition Table:
I (87) boot: ## Label            Usage          Type ST Offset   Length
I (94) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (100) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (107) boot:  2 app0             OTA app          00 10 00010000 00140000
I (113) boot:  3 app1             OTA app          00 11 00150000 00140000
I (120) boot:  4 spiffs           Unknown data     01 82 00290000 00170000
I (127) boot: End of partition table
I (130) boot_comm: chip revision: 3, min. application chip revision: 0
I (136) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=12cf8h ( 77048) map
I (155) esp_image: segment 1: paddr=00022d20 vaddr=3fc8e000 size=031f8h ( 12792) load
I (157) esp_image: segment 2: paddr=00025f20 vaddr=40380000 size=0a0f8h ( 41208) load
I (166) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=96d6ch (617836) map
I (253) esp_image: segment 4: paddr=000c6d94 vaddr=4038a0f8 size=03e48h ( 15944) load
I (256) esp_image: segment 5: paddr=000cabe4 vaddr=50000000 size=00010h (    16) load
I (261) boot: Loaded app from partition at offset 0x10000
I (262) boot: Disabling RNG early entropy source...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

felmue, 2021-12-15:
Hello @3050311118 

running your code on my M5StampC3 works fine - no crash.
- have you tried to connect to a different AP?
- have you tried a different USB cable?
- have you tried a different USB power source?

Thanks
Felix

sh-user, 2021-12-15:
@3050311118 i checked your example on my board ESP32-C3-12F, at 80 MHz, SPI mode: QIO and do get cyclic reboot, only works in DIO mode, message output is incorrect `mode:DIO, clock div:1`
Your Ardoino IDE settings:
```
I (64) boot.esp32c3: SPI Speed      : 80MHz
I (68) boot.esp32c3: SPI Mode       : QIO
```

felmue, 2021-12-16:
@sh-user 

good catch. I missed that one. Any idea why the message output is incorrect?

Thanks
Felix

SuGlider, 2021-12-17:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/startup.html

There are 2 stage boot loaders. The messages are from ROM and from each stage bootloader.
First stage uses DIO with 40MHz or 80MHz (clock div 1 or 2).
Second stage can use any mode or CPU frequency, with or without OTA and so on. This is can be set within Arduino IDE.

#### ROM and First Stage Bootloader Messages:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xe (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1428
load:0x403ce000,len:0xc04
load:0x403d0000,len:0x292c
SHA-256 comparison failed:
Calculated: 9f7363434bc7a1a2434ba3062500fa10b9fce5bb859899ee0424321b4ddaf742
Expected: 9b18b42e3e8e407f5e7b13f26c80172eda36d674c584e818f50843c766ebde69
Attempting to boot anyway...
entry 0x403ce000
```

#### Second Stage Bootloader Messages:
```
I (53) boot: ESP-IDF v4.4-dev-2313-gc69f0ec32 2nd stage bootloader
I (53) boot: compile time 12:10:14
I (54) boot: chip revision: 3
I (54) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (60) qio_mode: Enabling default flash chip QIO
I (64) boot.esp32c3: SPI Speed      : 80MHz
I (68) boot.esp32c3: SPI Mode       : QIO
I (72) boot.esp32c3: SPI Flash Size : 4MB
W (76) boot.esp32c3: PRO CPU has been reset by WDT.
I (80) boot: Enabling RNG early entropy source...
I (85) boot: Partition Table:
I (87) boot: ## Label            Usage          Type ST Offset   Length
I (94) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (100) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (107) boot:  2 app0             OTA app          00 10 00010000 00140000
I (113) boot:  3 app1             OTA app          00 11 00150000 00140000
I (120) boot:  4 spiffs           Unknown data     01 82 00290000 00170000
I (127) boot: End of partition table
I (130) boot_comm: chip revision: 3, min. application chip revision: 0
I (136) esp_image: segment 0: paddr=00010020 vaddr=3c0a0020 size=12cf8h ( 77048) map
I (155) esp_image: segment 1: paddr=00022d20 vaddr=3fc8e000 size=031f8h ( 12792) load
I (157) esp_image: segment 2: paddr=00025f20 vaddr=40380000 size=0a0f8h ( 41208) load
I (166) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=96d6ch (617836) map
I (253) esp_image: segment 4: paddr=000c6d94 vaddr=4038a0f8 size=03e48h ( 15944) load
I (256) esp_image: segment 5: paddr=000cabe4 vaddr=50000000 size=00010h (    16) load
I (261) boot: Loaded app from partition at offset 0x10000
I (262) boot: Disabling RNG early entropy source...
```

hggh, 2021-12-25:
I have also a ESP32-C3-WROOM-02 rev 3 and the same problem - I have used the newest version 2.0.2 but had no look.
I have tested the flash modes QIO and DIO but chip crashes on Wifi.begin()

SuGlider, 2022-02-14:
Issue confirmed. 
It's necessary to set DIO mode in the Arduino IDE Board Menu in order to make the posted example to run.
In QIO mode, it will reset the board with Watch Dog Timer.

SuGlider, 2022-02-14:
@3050311118 @hggh @sh-user @felmue @me-no-dev 

Issue analysis indicates that the problem has nothing to do with WiFi!
It has no relationship with ```wifi.begin()``` or any WiFi library.

The problem is related to using GPIO13 as LED in the Sketch.
This GPIO is used by ESP32-C3 in QIO Flash access.

To solve it, just change the LED pin in the sketch.

``` dart
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>

const char* ssid = "xxxxx";
const char* password = "yyyyyy";

WebServer server(80);

const int led = 2;   // NEVER USE GPIO13 with ESP32 C3.

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp32!");
  digitalWrite(led, 0);
}

void handleNotFound() {
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void) {
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp32")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
  delay(2);//allow the cpu to switch to other tasks
}
```

closed 2022-02-14 by SuGlider