espressif/arduino-esp32 #6029: ESP-NOW, "esp_now_add_peer" doesn't work on version 2.0.1

DonatelloX, 2021-12-15:
### Board

ESP32

### Device Description

DevKitC

### Hardware Configuration

ESP-NOW

### Version

v2.0.1

### IDE Name

Arduino 1.8.15

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

`esp_now_add_peer` does not work in 2.0.1. Cannot use ESP-NOW.

### Sketch

```cpp
#include <WiFi.h>
#include <esp_now.h>

//ESP-NOW
// Set the SLAVE MAC Address
uint8_t slaveAddress[] = {0x98, 0xCD, 0xAC, 0x60, 0xE1, 0xE4};
// PMK and LMK keys
static const char* PMK_KEY_STR = "1234567890123456";
static const char* LMK_KEY_STR = "1234567890123456";

typedef struct struct_esp_now {
  float temperature;
  float humidity;
  char serial_char;
};
// Create a struct_message called myData
struct_esp_now ESPnow_Data;

void setup() {
  Serial.begin(115200);

  Serial.println("");
  delay(500);
 
  Serial.println("*** START ***");

  // Set device as a Wi-Fi Station
  WiFi.mode(WIFI_STA);
  // Init ESP-NOW
  if (esp_now_init() != ESP_OK) {
    Serial.println("ESP-NOW: There was an error initializing");
    return;
  }
  // Setting the PMK key
  esp_now_set_pmk((uint8_t *)PMK_KEY_STR);

  // Register the slave
  esp_now_peer_info_t slaveInfo;
  memcpy(slaveInfo.peer_addr, slaveAddress, 6);
  slaveInfo.channel = 0;
  // Setting the master device LMK key
  for (uint8_t i = 0; i < 16; i++) {
    slaveInfo.lmk[i] = LMK_KEY_STR[i];
  }
  slaveInfo.encrypt = true;
  
  //esp_now_add_peer is always different from ESP_NOW in version 2.0.1. In version 1.0.6 it works correctly.
  if (esp_now_add_peer(&slaveInfo) != ESP_OK){
    Serial.println("ESP_NOW: There was an error registering the slave");
    While(1);
  }
                    
}

void loop() {

}
```

### Debug Message

```plain
esp_now_add_peer is always different from ESP_NOW in version 2.0.1. In version 1.0.6 it works correctly.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2021-12-15:
does it work without PMK?

DonatelloX, 2021-12-15:
No, it does not work.
With `esp_now_add_peer`, I get the error:
E (634) ESPNOW: Peer interface is invalid

me-no-dev, 2021-12-15:
ahh yeah... you need to enable AP mode for ESP-NOW. Please try `WiFi.mode(WIFI_AP);`

DonatelloX, 2021-12-16:
I change `WiFi.mode(WIFI_STA);` to `WiFi.mode(WIFI_AP);` but I always get the error:
_E (634) ESPNOW: Peer interface is invalid_
This happens with version 2.0.1 while with 1.0.6 it always works well with `WiFi.mode(WIFI_STA);` or `WiFi.mode(WIFI_AP);`.

delphifeel, 2021-12-17:
Hey. Try `memset(slaveInfo, 0, sizeof(slaveInfo));` before `memcpy`

DonatelloX, 2021-12-19:
Yes @delphifeel,  now it works!
I added `memset(&slaveInfo, 0, sizeof(slaveInfo));` before `memcpy(slaveInfo.peer_addr, slaveAddress, 6);`
The new code is as follows:
```
#include <WiFi.h>
#include <esp_now.h>

//ESP-NOW
// Set the SLAVE MAC Address
uint8_t slaveAddress[] = {0x98, 0xCD, 0xAC, 0x60, 0xE1, 0xE4};
// PMK and LMK keys
static const char* PMK_KEY_STR = "1234567890123456";
static const char* LMK_KEY_STR = "1234567890123456";

typedef struct struct_esp_now {
  float temperature;
  float humidity;
  char serial_char;
};
// Create a struct_message called myData
struct_esp_now ESPnow_Data;

void setup() {
  Serial.begin(115200);

  Serial.println("");
  delay(500);
 
  Serial.println("*** START ***");

  // Set device as a Wi-Fi Station
  WiFi.mode(WIFI_STA);
  // Init ESP-NOW
  if (esp_now_init() != ESP_OK) {
    Serial.println("ESP-NOW: There was an error initializing");
    return;
  }
  // Setting the PMK key
  esp_now_set_pmk((uint8_t *)PMK_KEY_STR);

  // Register the slave
  esp_now_peer_info_t slaveInfo;
  memset(&slaveInfo, 0, sizeof(slaveInfo));
  memcpy(slaveInfo.peer_addr, slaveAddress, 6);
  slaveInfo.channel = 0;
  
  // Setting the master device LMK key
  for (uint8_t i = 0; i < 16; i++) {
    slaveInfo.lmk[i] = LMK_KEY_STR[i];
  }
  slaveInfo.encrypt = true;

  if (esp_now_add_peer(&slaveInfo) != ESP_OK){
    Serial.println("ESP_NOW: There was an error registering the slave");
    while(1);
  }
  Serial.println("Completed");
                    
}

void loop() {

}
```

closed 2022-01-04 by VojtechBartoska

DonatelloX, 2022-01-20:
After hours of operation, I reverted to version 1.0.6: many times packages are lost with both version 2.0.1 and 2.0.2.

aidarishkulov, 2022-02-22:
How to measure WiFi signal strength ESP_NOW?

DonatelloX, 2022-02-22:
I have never measured the signal strength, but keeping 2 devices close together, espnow does not work well with version 2.0.x, in version 1.0.6 it works fine up to 100 meters with external antenna.

ISbitPro, 2022-11-14:
> Hey. Try `memset(&slaveInfo, 0, sizeof(slaveInfo));` before `memcpy`

This saved me!
I have not touched my code for over a year, and then suddenly it stopped working with "E (340) ESPNOW: Peer interface is invalid"

GitNitneroc, 2024-03-03:
I'm very bad at C/C++ but doesn't this not-so-obvious request of memset() show that the api should provide a function to generate a esp_now_peer_info_t ?