espressif/arduino-esp32 #6032: setCpuFrequencyMhz() changes Serial bauds if frequency<80Mhz

Leonleon33, 2021-12-16:
### Board

ESP32 Dev module

### Device Description

ESP32 Dev module alone

### Hardware Configuration

Nothing

### Version

v2.0.1

### IDE Name

Arduino IDE

### Operating System

Ubuntu 20.04

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

When setting CPU frequency below 80Mhz, the serial monitor is garbage until I change the baud rate...
I manage to solve the problem by setting the Serial.begin(bauds)  to : 80 / cpufreq * monitor_bauds
See sketch attached.

### Sketch

```cpp
uint32_t Freq = 0;
const int bauds = 115200;
int my_bauds;
int cpufreqs[6] = {240, 160, 80, 40, 20, 10};
int i = 0;

void setup() {
  Freq = getCpuFrequencyMhz();
  if (Freq < 80) {
    my_bauds = 80 / Freq * bauds;
  }
  else {
    my_bauds = bauds;
  }

  Serial.begin(my_bauds);        // Attention dépend de la frequence CPU si elle est <80Mhz 
  delay(500);
  //
  Freq = getCpuFrequencyMhz();
  Serial.print("CPU Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getXtalFrequencyMhz();
  Serial.print("XTAL Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getApbFrequency();
  Serial.print("APB Freq = ");
  Serial.print(Freq);
  Serial.println(" Hz");
}
void loop() {
  Serial.print("\nchange CPU freq to ");
  Serial.println(cpufreqs[i]);
  delay(1000);
  setCpuFrequencyMhz(cpufreqs[i]);
  Freq = getCpuFrequencyMhz();
  Serial.print("Send to serial with my_bauds = ");
  Serial.println(my_bauds);
  Serial.print("CPU Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  //  Change bauds rate ?

  if (Freq < 80) {
    my_bauds = 80 / Freq * bauds;
  }
  else {
    my_bauds = bauds;
  }
  Serial.end();
  delay(1000);
  Serial.begin(my_bauds);
  delay(1000);
  Serial.print("\nchange my_bauds to ");
  Serial.println(my_bauds);
  Serial.print("Serial set to ");
  Serial.println(my_bauds);
  Serial.print("CPU Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Serial.print("my_bauds = ");
  Serial.println(my_bauds);

  delay(1000);
  i++;
  i = i % 6;
}
```

### Debug Message

```plain
No debug message, just monitor output set to 115200
CPU Freq = 240 MHz
XTAL Freq = 40 MHz
APB Freq = 80000000 Hz

change CPU freq to 240
Send to serial with my_bauds = 115200
CPU Freq = 240 MHz

change my_bauds to 115200
Serial set to 115200
CPU Freq = 240 MHz
my_bauds = 115200

change CPU freq to 160
Send to serial with my_bauds = 115200
CPU Freq = 160 MHz

change my_bauds to 115200
Serial set to 115200
CPU Freq = 160 MHz
my_bauds = 115200

change CPU freq to 80
Send to serial with my_bauds = 115200
CPU Freq = 80 MHz

change my_bauds to 115200
Serial set to 115200
CPU Freq = 80 MHz
my_bauds = 115200

change CPU freq to 40
f⸮⸮⸮`                 AND OTHER GARBAGE
change my_bauds to 230400
Serial set to 230400
CPU Freq = 40 MHz
my_bauds = 230400

change CPU freq to 20
f⸮⸮⸮`                 AND OTHER GARBAGE
change my_bauds to 460800
Serial set to 460800
CPU Freq = 20 MHz
my_bauds = 460800

change CPU freq to 10
f⸮⸮⸮`                 AND OTHER GARBAGE
change my_bauds to 921600
Serial set to 921600
CPU Freq = 10 MHz
my_bauds = 921600
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2021-12-17:
Thanks @Leonleon33 for reporting and solving the issue with your example.
The issue will be solved with the PR #6037
Great Job!

closed 2021-12-21 by me-no-dev

savejeff, 2022-08-17:
I encounter the same problem. 
I was able to fix this in the same way. 
Is the fix maybe not published on the main branch?

SuGlider, 2022-08-17:
This PR is already merged and it is part of Arduino Core 2.0.2+
About PlatformIO, it is already in latest realease.
https://github.com/platformio/platform-espressif32/releases

savejeff, 2022-08-18:
But why do i then need to ajust the baudrate in the way that was proposed by the issue creator?

What I'm doing is going into a different mode that is something like a standby mode. when setting a CpuFrq below 80Mhz i have to update the baudrate accordingly. 

It also seems like I2C and SPI clocks are also slowed down. SD Card writes also stop working after the reduction of the CPU clock

SuGlider, 2022-08-24:
If necessary, please open a new issue clarifying the Arduino Core Version, if this issue occurs with Arduino IDE, what is the SoC (ESP32 or the S2/S3/C3) and so on. Describe your issue and then add a minimum sketch that can be used to reproduce the issue.

savejeff, 2022-08-25:
I have allready supplied all these informations. The original sketch provided by the Issue opened still works with the latest Arduino 2.0.4 and ESP32-S3 Devkit C-1.

But ill open a new issue

kjw3898, 2022-10-21:
> I have already supplied all these information. The original sketch provided by the Issue opened still works with the latest Arduino 2.0.4 and ESP32-S3 Devkit C-1.
> 
> But ill open a new issue

try this when change cpu frequency
``` 
 
Serial.begin(115200); // first init serial 
// ...  your code
Serial.end();    // when chage cpu frequency
setCpuFrequencyMhz(40);
Serial.begin(115200);
```

aeonSolutions, 2023-01-01:
That's quite a neat hack !!
I've done some clean up to the code and make it more usable . Here:

```
/*
* LAB: 8
* Name: ESP32 Read Default Clocks
* Author: Khaled Magdy
* For More Info Visit: https://deepbluembedded.com/esp32-change-cpu-speed-clock-frequency/
*/
 //function takes the following frequencies as valid values:
//  240, 160, 80    <<< For all XTAL types
//  40, 20, 10      <<< For 40MHz XTAL
//  26, 13          <<< For 26MHz XTAL
//  24, 12          <<< For 24MHz XTAL
// For More Info Visit: https://deepbluembedded.com/esp32-change-cpu-speed-clock-frequency/

#include "Wire.h"
 
uint32_t Freq = 0;
int MCU_FREQUENCY_SERIAL_SPEED=115200;
int SERIAL_DEFAULT_SPEED = 115200;

void changeMcuFreq(int Freq){
  delay(500);
  setCpuFrequencyMhz(Freq);
  changeSerialBaudRate(Freq);
  delay(500);
} 

void serialMcuFreq(){
  Freq = getCpuFrequencyMhz();
  Serial.print("CPU Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getXtalFrequencyMhz();
  Serial.print("XTAL Freq = ");
  Serial.print(Freq);
  Serial.println(" MHz");
  Freq = getApbFrequency();
  Serial.print("APB Freq = ");
  Serial.print(Freq/1000000);
  Serial.println(" MHz");  
}

void changeSerialBaudRate(uint32_t Freq){
    if (Freq < 80) {
      MCU_FREQUENCY_SERIAL_SPEED = 80 / Freq * SERIAL_DEFAULT_SPEED;
    }
    else {
      MCU_FREQUENCY_SERIAL_SPEED = SERIAL_DEFAULT_SPEED;
    }
    Serial.end();
    delay(1000);
    Serial.begin(MCU_FREQUENCY_SERIAL_SPEED);
    delay(1000);
    Serial.print("\nSerial Baud Rate Speed is ");
    Serial.println(MCU_FREQUENCY_SERIAL_SPEED);
}

void setup()
{

  Serial.begin(MCU_FREQUENCY_SERIAL_SPEED);
  
}
 
void loop()
{
  changeMcuFreq(240);
  serialMcuFreq();    
  delay(2000);
  
  changeMcuFreq(160);
  serialMcuFreq();    
  delay(2000);

  changeMcuFreq(80);
  serialMcuFreq();    
  delay(2000);

  changeMcuFreq(40);
  serialMcuFreq();    
  delay(2000);

  changeMcuFreq(20);
  serialMcuFreq();    
  delay(2000);

  changeMcuFreq(10);
  serialMcuFreq();    
  delay(2000);

}

```

aeonSolutions, 2023-05-04:
The above solution does not work for me.  I get a crash.

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
Saved PC:0x42120ca2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
entry 0x403c98d8
```

here's my code:

```
void changeMcuFreq(INTERFACE_CLASS* interface, int Freq){
  Serial.flush();
  Serial.end();
  interface->UARTserial->flush();
  delay(200);
  setCpuFrequencyMhz(Freq);
  interface->CURRENT_CLOCK_FREQUENCY=Freq;
  if (Freq < 80) {
    interface->MCU_FREQUENCY_SERIAL_SPEED = 80 / Freq * interface->SERIAL_DEFAULT_SPEED;
  } else {
    interface->MCU_FREQUENCY_SERIAL_SPEED = interface->SERIAL_DEFAULT_SPEED;
  }
  Serial.begin(interface->MCU_FREQUENCY_SERIAL_SPEED);
  interface->UARTserial->updateBaudRate(interface->MCU_FREQUENCY_SERIAL_SPEED);
  
  interface->UARTserial->print("The current Serial Baud speed on the UART Port is ");
  interface->UARTserial->println(interface->MCU_FREQUENCY_SERIAL_SPEED);
} 
```

aeonSolutions, 2023-05-05:
So i read. 
And i confirm it is working for serial send , Serial.print() command, however it is not working for Serial.read()

JasonPittenger, 2023-05-22:
I just tried a sketch changing the clock speed from 240,160,80,40,26,24,20,13,12,10,5,4,2,1.
At 40 MHz, I had to use 2*baud, but at 20  MHz and 10 MHz, it worked using 1*baud.
Trying 26,24,13,12,5,4,2 or 1 all resulted in a clock speed of 10MHz.

aeonSolutions, 2023-05-22:
Haven't tested at 2 baud ....or 1 . 
 brb

JasonPittenger, 2023-05-22:
I had a typo.  It was 2 x baud.  IE, at 40 MHz, I had to set the serial port to 230400 to get 115200 output.  All other settings I was able to use 115200 to get 115200.

JasonPittenger, 2023-05-22:
I can confirm, serial baud is correct at 240, 160, 80, 20, and 10 MHz, but incorrect at 40 MHz.  Serial.begin must be called after changing frequency to 10 MHz (and probably when changing to 20 as well).

aeonSolutions, 2023-05-23:
here's a draft of my code with @JasonPittenger's solution:

```
void changeMcuFreq(INTERFACE_CLASS* interface, int Freq){
  Serial.flush();
  Serial.end();
  interface->UARTserial->flush();
  delay(200);
  setCpuFrequencyMhz(Freq);
  interface->CURRENT_CLOCK_FREQUENCY=Freq;
  if (Freq < 80) {
    interface->MCU_FREQUENCY_SERIAL_SPEED = 80 / Freq * interface->SERIAL_DEFAULT_SPEED;                 //230400
  }else if (Freq < 40) {
    interface->MCU_FREQUENCY_SERIAL_SPEED = interface->SERIAL_DEFAULT_SPEED;                             //115200
    Serial.begin( interface->SERIAL_DEFAULT_SPEED );  
} else {
    interface->MCU_FREQUENCY_SERIAL_SPEED = interface->SERIAL_DEFAULT_SPEED; 
  }

  interface->UARTserial->updateBaudRate(interface->MCU_FREQUENCY_SERIAL_SPEED);
  
  interface->UARTserial->print("The current Serial Baud speed on the UART Port is ");
  interface->UARTserial->println(interface->MCU_FREQUENCY_SERIAL_SPEED);
} 
```
the original code can be found here:
https://github.com/aeonSolutions/aeonlabs-ESP32-C-Base-Firmware-Libraries#readme

aeonSolutions, 2023-05-24:
Here's my code snippet updated with @JasonPittenger  solution

```
void changeMcuFreq(INTERFACE_CLASS* interface, int Freq){
  Serial.flush();
  Serial.end();
  delay(200);
  setCpuFrequencyMhz(Freq);
  interface->CURRENT_CLOCK_FREQUENCY=Freq;
  if (Freq < 80) {
    interface->MCU_FREQUENCY_SERIAL_SPEED = 2 * interface->SERIAL_DEFAULT_SPEED;                         
  }else if (Freq == 40) {
    interface->MCU_FREQUENCY_SERIAL_SPEED = 2*interface->SERIAL_DEFAULT_SPEED;                            
} else {
    interface->MCU_FREQUENCY_SERIAL_SPEED = interface->SERIAL_DEFAULT_SPEED; 
  }

  Serial.begin( interface->SERIAL_DEFAULT_SPEED );  
  interface->UARTserial->updateBaudRate(interface->MCU_FREQUENCY_SERIAL_SPEED);
  interface->UARTserial->print("The current Serial Baud speed on the UART Port is ");
  interface->UARTserial->println(interface->MCU_FREQUENCY_SERIAL_SPEED);
} 
```
the original code can be found here:
https://github.com/aeonSolutions/aeonlabs-ESP32-C-Base-Firmware-Libraries#readme

PR #6037 (2021-12-21): Fixes baudrate with CPU Freq < 80MHz

## Summary
This PR fixes an issue with UART when CPUFreq is lower than 80MHz (APB Freq)

## Impact
No impact beside the fix.

## Related links
Fix #6032