espressif/arduino-esp32 #6035: Potential internal UART buffer overflow when using all three UARTs.

gkvirkvia, 2021-12-17:
### Board

TinyPico

### Device Description

A plain TinyPico.

### Hardware Configuration

UART1 configured as input on pins 15 and 14, UART2 configured as output on pins 25 and 26.

Pins 32 and 33 are connected to bridged DE/RE inputs of two MAX485 modules.

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

My sketch is very simple - I read data on `UART1` and forward it to `Serial` and `UART2`.

When I only forward the data verbatim to `UART2` and `Serial`, i.e. do `byte = Serial1.read(), Serial2.write(byte); Serial.write(byte);`  everything works fine with the data rate that I happen to be at.

However, if I instead try to `Serial.printf` a nicer hex representation of the data I intercept, after 3 seconds or so of behaving correctly, the program goes haywire. In particular, `UART1` starts to see bogus data - `Serial1.available()` always returns true and the data seems to be coming from nowhere. I verified with a standalone logic analyzer that the input is quiet when this is happening.

What's likely happening is that the `printf` call inflates the data, overflow some sort of internal buffer when the data flow is too fast for the baud rate of `Serial` to handle and this somehow surfaces itself as `Serial1` malfunction. 

### Sketch

```cpp
#include <HardwareSerial.h>

#define SERIAL2_RE_DE 33
#define SERIAL1_RE_DE 32

void setup() {
    Serial.begin(115200);
    Serial1.begin(115200, SERIAL_8N1, 15, 14);
    Serial2.begin(115200, SERIAL_8N1, 25, 26);
    pinMode(SERIAL1_RE_DE, OUTPUT);
    pinMode(SERIAL2_RE_DE, OUTPUT);
    digitalWrite(SERIAL1_RE_DE, 0);
    digitalWrite(SERIAL2_RE_DE, 1);
    Serial.println("Booted up");
}

void loop() {
  while(Serial1.available()) {
    int byte = Serial1.read();
    Serial2.write(byte);
    Serial.printf("%02X\n", byte); // If this line is commented or even turned into Serial.write(byte),
                                   // the sketch is fine. With it uncommented, the Serial1 starts
                                   // "reading" data that's just not on the wire, even if I completely
                                   // ground the input.
  }
}
```

### Debug Message

```plain
None
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2021-12-17:
> ### Version
> v1.0.6

Hello @gkvirkvia, please if it's possible for your project consider update to the latest version of ESP32 Arduino Core to latest version [2.0.1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.1). It includes refactoring of UART and your problem can be fixed by that.

gkvirkvia, 2021-12-22:
@VojtechBartoska Unfortunately not possible as I'm using Arduino IDE which doesn't seem to allow for switching the underlying IDF versions.

me-no-dev, 2021-12-22:
you need to change the Arduino for ESP32 version, not IDF version. Make sure that you are using the correct Board Manager link: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-boards-manager

SuGlider, 2022-01-11:
@gkvirkvia 
I can't reproduce this issue, neither using 1.0.6 nor using 2. 0. 0+

It works fine even after 10 minutes sending data to UART 1.
Please provide other ways to reproduce it and more information about the application and its conditions.

In order to test it, I connected pin 15 (RX1) to pin 26 (TX2) and changed the sketch to keep sending bytes in a loopback from UART2 TX to UART1 RX. Output is visible in UART 0.

``` dart
#include <HardwareSerial.h>

#define SERIAL2_RE_DE 33
#define SERIAL1_RE_DE 32

// Loopback on 
void setup() {
    Serial.begin(115200);
    Serial1.begin(115200, SERIAL_8N1, 15, 14);
    Serial2.begin(115200, SERIAL_8N1, 25, 26);
    pinMode(SERIAL1_RE_DE, OUTPUT);
    pinMode(SERIAL2_RE_DE, OUTPUT);
    digitalWrite(SERIAL1_RE_DE, 0);
    digitalWrite(SERIAL2_RE_DE, 1);
    Serial.println("Booted up");
}

void loop() {
  static uint8_t count = 0;

  Serial2.write('0' + count);
  delay(100);
  while(Serial1.available()) {
    int byte = Serial1.read();
    //Serial2.write(byte);
    //Serial.write(byte);
    Serial.printf("%02X ", byte); // If this line is commented or even turned into Serial.write(byte),
                                   // the sketch is fine. With it uncommented, the Serial1 starts
                                   // "reading" data that's just not on the wire, even if I completely
                                   // ground the input.
  }
  count++;
  if (count + '0' > 'Z') {
    count = 0;
    Serial.println();
  }
}
```

closed 2022-01-27 by VojtechBartoska