espressif/arduino-esp32 #6038: SD.open)( --> How to parse through a file and write data at specific lines on the fly / simultaneously

basementmedia2, 2021-12-18:
### Board

ESP32

### Device Description

ESP32 D1 Mini / AZDelivery

### Hardware Configuration

microSD-Shield

### Version

v2.0.1

### IDE Name

Arduino IDE

### Operating System

Linux Manjaro

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hi,

this worked with ESP8266 (delete_item comes from ajax request and "Messung" is a struct which contains sensor data):

```
bool deleter = true;
auswertungenFile = SD.open("2021-12-10.bin", FILE_WRITE);
if (auswertungenFile) {
          auswertungenFile.seek(0); // Back to beginning of file
          while (auswertungenFile.read((uint8_t *)&Messung, sizeof(Messung)) != 0 && deleter) {
            char delete_id[7];
            sprintf(delete_id, "%d%d%d", Messung.stunde, Messung.minute, Messung.sekunde);
            if (atoi(delete_id) == delete_item) {
              // We have found the entry that has to be marked as "non-valid"
              Messung.valid = false;
              auswertungenFile.seek(auswertungenFile.position() - sizeof(Messung));
              auswertungenFile.write((uint8_t *)&Messung, sizeof(Messung));
              deleter = false;
            }
  yield();
}
auswertungenFile.close();
```

But with ESP32 the whole file gets just deleted.

Is there a way to get this working with SD.h of ESP32?

Best wishes 
Daniel

### Sketch

```cpp
See above
```

### Debug Message

```plain
No debug message, please see description above
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2021-12-23:
@P-R-O-C-H-Y Can you please take a look?

basementmedia2, 2021-12-24:
Hi,

just found the solution myself:

auswertungenFile = SD.open("2021-12-10.bin", "r+");

and then it works.

LG and Merry Christmas to all!

closed 2021-12-24 by basementmedia2