espressif/arduino-esp32 #6039: SPIFFS: mount failed, -10025

GiovanniCmpaner, 2021-12-18:
### Board

ESP32-S2

### Device Description

Genuine ESP32-S2-Saola-1M

### Hardware Configuration

No connections

### Version

v2.0.1

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Getting errors when trying to use SPIFFS with ESP32-S2.

`platformio.ini` file:
```ini
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-master
platform_packages = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
                    platformio/tool-esptoolpy @ https://github.com/tasmota/esptool/releases/download/v3.2/esptool-v3.2.zip
board = esp32dev
framework = arduino

monitor_speed = 115200
build_type = debug
board_build.speed = 921600
board_build.partitions = partitions_custom.csv
board_build.mcu = esp32s2

build_unflags = -std=gnu++11 -fno-rtti
build_flags = -std=gnu++17 -DCORE_DEBUG_LEVEL=5

[...]
```

`partition_custom.csv` file:
```csv
# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,   0x4000,
otadata,  data, ota,     0xD000,   0x2000,
phy_init, data, phy,     0xF000,   0x1000,
ota_0,    app,  ota_0,   ,         1900K,
ota_1,    app,  ota_1,   ,         1900K,
storage,  data, spiffs,  ,         128K,
```

### Sketch

```cpp
#include <Arduino.h>
#include <SPIFFS.h>

void setup()
{
    Serial.begin(115200);
    Serial.setDebugOutput(true);

    SPIFFS.begin(true);
}

void loop()
{
}
```

### Debug Message

```plain
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x4e4
load:0x4004c000,len:0x82c
load:0x40050000,len:0x277c
entry 0x4004c174
E (232) SPIFFS: mount failed, -10025
[   663][E][esp32-hal-misc.c:125] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT
E (1445) SPIFFS: mount failed, -10025
[  1875][E][SPIFFS.cpp:114] format(): Formatting SPIFFS failed! Error: -1
[  1876][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

GiovanniCmpaner, 2021-12-18:
Related issues:
https://github.com/espressif/arduino-esp32/issues/638
https://github.com/platformio/platform-espressif32/issues/619

GiovanniCmpaner, 2021-12-18:
The workaround I found is this:
https://github.com/platformio/platform-espressif32/issues/546

1. Add to `platformio.ini`:
`board_build.flash_mode = qio`
2. This will break the boot from ESP32-S2
3. Then upload the bootloader:
`python.exe esptool.py --chip esp32s2 --port "COM5" --baud 921600 write_flash 0x1000 bootloader_qio_40m.bin`
4. Full paths omitted from tools and bin, you can get them on verbose upload

Then, SPIFFS works correctly.

atanisoft, 2021-12-19:
Try renaming "storage" to "spiffs" in your partition csv file.

GiovanniCmpaner, 2021-12-19:
Even after renaming to `spiffs`, with `flash_mode = dio`, I still get `SPIFFS: mount failed, -10025`.

Jason2866, 2021-12-21:
It is probably a issue writing the needed correct boot loader. Solved in latest development platform for esp32.
You can try (uses a updated platform version)
```
platform                    = https://github.com/tasmota/platform-espressif32/releases/download/v3.4.1/Tasmota-platform-espressif32.zip
platform_packages           = framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
```

GiovanniCmpaner, 2021-12-21:
This indeed solved the problem.
Is the bootloader a responsibility from PlatformIO, meaning, should I open an issue there?

Jason2866, 2021-12-21:
@GiovanniCmpaner  No, since it is fixed in actual master / development branch.
Using core 2.0.x is still beta since PlatformIO did not released since IDF 4.4 is still not released from espressif

closed 2021-12-21 by GiovanniCmpaner

QGB, 2022-09-25:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4ab8657f6f5aa80b81c7dd649db3bd64a98999ed7c65acc5a2ab77ce80caf525
Expected: 6babbc7b9ea63e04e0fc89ec224f6268aa3f84392018f73ab33721f768ce2eb6
Attempting to boot anyway...
entry 0x403ce000
E (154) SPIFFS: mount failed, -10025
E (18136) SPIFFS: mount failed, -10025
[ 18109][E][SPIFFS.cpp:114] format(): Formatting SPIFFS failed! Error: -1
[ 18110][E][SPIFFS.cpp:89] begin(): Mounting SPIFFS failed! Error: -1

```

happy12, 2023-01-15:
try formatting the system files on the first run. More info here: https://www.tutorialspoint.com/esp32_for_iot/esp32_for_iot_spiffs_storage.htm