espressif/arduino-esp32 #6041: How to set time, with ESP32 core time management functions

fflori4n, 2021-12-19:
### Related area

documentation, example, core time management

### Hardware specification

ESP32 generic

### Is your feature request related to a problem?

There are functions used in this example:
https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Time/SimpleTime/SimpleTime.ino
which are used to set the time variable for the ESP32 from an NTP response -- configTime() -- , and to access the time variable -- getLocalTime() --.
These functions are part of the ESP core, but have no real documentation at all. 
is there a setLocalTime() function, that allows the user to set the time variable to a specific unix time? e.g. instead of updating the time from an NTP server, use external RTC module time, or time from GPS to set this internal time variable.

### Describe the solution you'd like

Please provide more examples, some sort of documentation for the available functions, or add the  location of these ESP core time functions in the header of the example. Add a setLocalTime() function if it's missing.

### Additional context

Other people searching for the same thing:

https://arduino.stackexchange.com/questions/71797/esp32-documentaiont-for-time-h
https://www.esp32.com/viewtopic.php?t=6043
https://stackoverflow.com/questions/54458116/how-can-i-set-the-esp32s-clock-without-access-to-the-internet

some people are using Posix functions form <sys/time.h>, or external libraries such as ESP32 time, but if getLocalTime() and configTime() exist, setLocalTime() should also be implemented

GiovanniCmpaner, 2021-12-19:
Try something like this (in C):
```
#include <time.h>

void adjustTime(){
    tm parts{};
    parts.tm_sec   = 0;   // seconds after the minute (0-59)
    parts.tm_min   = 23;  // minutes after the hour (0-59)
    parts.tm_hour  = 12;  // hours since midnight (0-23)
    parts.tm_mday  = 2;   // day of the month (1-31)
    parts.tm_mon   = 9;   // months since January (0-11)
    parts.tm_year  = 121; // years since 1900    

    time_t time = mktime(&parts);

    timeval val{};
    val.tv_sec = time;
    val.tv_usec = 0;

    settimeofday(&val, nullptr);
}
```
An example adjusting time from using a string in C++:
```
#include <ctime>
#include <string>
#include <sstream>
#include <iomanip>

// format: yyyy-MM-dd HH:mm:ss
void adjustTimeFromString( const std::string& str )
{
    auto stream = std::istringstream{str};
    auto parts = std::tm{};
    stream >> std::get_time( &parts, "%Y-%m-%d %H:%M:%S" );
    auto time = std::mktime( &parts );
    auto val = timeval{mktime(&parts)};
    settimeofday(&val, nullptr);
}
```
References:
https://www.cplusplus.com/reference/ctime/mktime/
https://www.cplusplus.com/reference/ctime/tm/
https://linux.die.net/man/2/settimeofday

fflori4n, 2021-12-19:
Thanks, these examples work great. So it is the preferred way to use POSIX functions to set and get time on the ESP.

closed 2021-12-19 by fflori4n