espressif/arduino-esp32 #6042: Unable to use traditional OTA with Rainmaker

rmetzner49, 2021-12-19:
### Board

ESP32-Dev Module, Wroom32

### Device Description

Dev Module on Breadboard

### Hardware Configuration

No

### Version

other

### IDE Name

Arduino 1.8.18

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Rainmaker appears to use a Custom Partition Table that perhaps breaks Arduino OTA.

Solution:  Use Rainmaker OTA which requires an "Image" upload to the cloud.

Question:  Which file (HEX?) is the "Image" that Rainmaker's OTA is looking for?  Can this be accomplished with Arduino?

### Sketch

```cpp
// ***************************beginOTA**************************************
uint8_t mac[6];
void beginOTA() {
  WiFi.macAddress(mac);
  ArduinoOTA
  .onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  })
  .onEnd([]() {
    Serial.println("\nEnd");
  })
  .onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  })
  .onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });

  ArduinoOTA.begin();
}//------------------------------------------------------------------------

// **************************WiFisetup****************************************
void WiFisetup(){
  WiFi.begin(ssid, password);                         //connect to WiFi
  Serial.print("Attempting Connection to WiFi Network: ");
  Serial.println(ssid);
  while (WiFi.status() != WL_CONNECTED) {
    delay(2000);
    Serial.print("**");
    Serial.println(WiFi.status());
  }
  Serial.println(" ");
  Serial.print("CONNECTED to ");
  Serial.print(ssid);
  Serial.print(" using Local IP address: ");
  Serial.println(WiFi.localIP());
  delay(500);
  beginOTA();
  delay(500);
  ArduinoOTA.setHostname("WaterWheel-300_ESP32");
}// --------------------------- end WiFi setup
```

### Debug Message

```plain
Traditional Upload TIMES OUT, never receives permission
```

### Other Steps to Reproduce

None

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

pedrominatel, 2021-12-20:
Hi @shahpiyushv, can you help us with this question?

rmetzner49, 2021-12-20:
@pedrominatel , It appears Rainmaker has a way to OTA by uploading an "image" to the server and then scheduling the update as a job.  My efforts tanked at not being able to supply a "server_certificate".  Would this be the original "provisioning barcode" used?  The instructions are pretty vague.  Piyush needs to walk us through the method.

pedrominatel, 2021-12-21:
Do you mean this explanation? https://rainmaker.espressif.com/docs/ota.html

Adding ESP Rainmaker docs for the Arduino support it's on my plans, but I haven't started yet.

rmetzner49, 2021-12-21:
Yes.  "(valid only for OTA_USING_PARAMS), you can copy the appropriate server.crt file into your example's main/ folder and embed it into the binary using the below line in main/CMakeLists.txt".  Since with Arduino I don't have a CMakeLists.txt file, I'm unable to add this.  I also make the assumption that the "image" needed looks something like "WaterWheel_300.ino.esp32.bin" when using the Arduino option to export a binary.

So if that is correct, the only thing I'm missing is the "server_certificate" and how to create it using Arduino.  I guess I have to wait until you are able to add Arduino support.

rmetzner49, 2021-12-21:
Is the "server_certificate" the barcode generated while setting up the node?

rmetzner49, 2021-12-22:
I tried some more by adding the line "esp_rmaker_ota_config_t ota_config = {
        .server_cert = ESP_RMAKER_OTA_DEFAULT_SERVER_CERT,
    };
That solved the missing server cert problem but raised another question.  The above is suggested for OTA_USING_PARMS, but the Rainmaker Dashboard says to use OTA_USING_TOPICS.  Which one is correct?

shahpiyushv, 2021-12-23:
@rmetzner49 , had you already tried [`RMaker.enableOTA(OTA_USING_PARAMS)`](https://github.com/espressif/arduino-esp32/blob/caef4006af491130136b219c1205bdcf8f08bf2b/libraries/RainMaker/examples/RMakerSwitch/RMakerSwitch.ino#L65) in your Arduino RainMaker example?

Both, OTA_USING_PARAMS  and OTA_USING_TOPICS can be used for OTA. "Using Params" is for end user workflows with no admin user involved (using RainMaker CLI as you can see [here](https://rainmaker.espressif.com/docs/ota.html#ota-using-params)), whereas "Using Topics" is meant for admin users using the Dashboard as you can see [here](https://rainmaker.espressif.com/docs/ota.html#ota-using-topics). Reading the docs [here](https://rainmaker.espressif.com/docs/user-roles.html) will help understand what these 2 user roles are. However, for the simplest case (and specifically for Arduino use case), end user and admin user are the same and any of the 2 can be used, but the method of triggering the OTA Job is different.

In any case, I have not tested the OTA in RainMaker Arduino recently, so am not 100% sure if it works well right now. I will check soon and update. You may also try as per my comments above and check.

rmetzner49, 2021-12-24:
@shahpiyushv , I did NOT try the OTA_USING_PARAMS method because it appeared setting up the CLI is fairly involved.  So I opted for the Dashboard thinking it was less complex.  I think I  now understand PARAMS is for the CLI and TOPICS is for the Dashboard.  I will try both ways to see if I can push out an update.  Please verify the .server_cert = ESP_RMAKER_OTA_DEFAULT_SERVER_CERT needs to be declared before RMaker.enableOTA(OTA_USING_TOPICS).  I will let you know if I have any success.

rmetzner49, 2021-12-24:
@shahpiyushv   When I enable RMaker using TOPICS, a fault is created that reboots about 3 second intervals.  I will try making the CLI tomorrow.  Going back to PARAMS we run normally.  The register dump if you're interested is:

Core  0 register dump:
PC      : 0x800d2e1d  PS      : 0x00060030  A0      : 0x801603f6  A1      : 0x3ffd8b00  
A2      : 0x3ffd7834  A3      : 0x00000000  A4      : 0x3ffd4b80  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x8016490c  A9      : 0x3ffd8ad0  
A10     : 0x3ffd7834  A11     : 0x3ffd8b00  A12     : 0x3ffc51b0  A13     : 0xb33fffff  
A14     : 0x00000001  A15     : 0x00000001  SAR     : 0x0000000c  EXCCAUSE: 0x00000014  
EXCVADDR: 0x800d2e1c  LBEG    : 0x40090c88  LEND    : 0x40090c9e  LCOUNT  : 0xffffffff  

Backtrace:0x400d2e1a:0x3ffd8b000x401603f3:0x3ffd8b30 0x401606c2:0x3ffd8b60

rmetzner49, 2021-12-24:
@shahpiyushv I placed RMaker in a separate .h file and call the function from the sketch setup
// *************************** RMakerSetup *************************************
void RMakerSetup(){
  pinMode(RM_RST,INPUT_PULLUP);                  // D2 Input
  pinMode(RM_LED,OUTPUT);                        // D13 Output

    Node my_node;    
    my_node = RMaker.initNode("ESP RainMaker Node");

    //Create custom dimmer device
    my_device.addNameParam();
    my_device.addPowerParam(DEFAULT_POWER_MODE);
    my_device.assignPrimaryParam(my_device.getParamByName(ESP_RMAKER_DEF_POWER_NAME));

    //Create and add a custom level parameter
    Param level_param1("This Fill-L", "custom.param.This Fill-L", value(3333), PROP_FLAG_READ | PROP_FLAG_WRITE);
    my_device.addParam(level_param1);
    Param level_param2("Day's Fill-L", "custom.param.Day's Fill-L", value(6666), PROP_FLAG_READ | PROP_FLAG_WRITE);
    my_device.addParam(level_param2);
    my_device.addCb(write_callback);
    
    //Add custom dimmer device to the node   
    my_node.addDevice(my_device);
    Serial.printf("Device added!\n");

    //OTA is optional 
    esp_rmaker_ota_config_t ota_config = {.server_cert = ESP_RMAKER_OTA_DEFAULT_SERVER_CERT,};

    RMaker.enableOTA(OTA_USING_PARAMS);
    //If you want to enable scheduling, set time zone for your region using setTimeZone(). 
    //The list of available values are provided here https://rainmaker.espressif.com/docs/time-service.html
    // RMaker.setTimeZone("Asia/Shanghai");
    // Alternatively, enable the Timezone service and let the phone apps set the appropriate timezone
    RMaker.enableTZService();
    RMaker.enableSchedule();
    RMaker.start();
    Serial.printf("RMaker started!\n");
    WiFi.onEvent(sysProvEvent);
  #if CONFIG_IDF_TARGET_ESP32
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_BLE, WIFI_PROV_SCHEME_HANDLER_FREE_BTDM, WIFI_PROV_SECURITY_1, pop, service_name);
  #else
    WiFiProv.beginProvision(WIFI_PROV_SCHEME_SOFTAP, WIFI_PROV_SCHEME_HANDLER_NONE, WIFI_PROV_SECURITY_1, pop, service_name);
  #endif
}//---------------------------------------------------------------------------

rmetzner49, 2021-12-24:
@shahpiyushv , @pedrominatel Where does the requirements.txt file live that is used with the CLI installation?

shahpiyushv, 2021-12-27:
@rmetzner49 , the  [requirements.txt](https://github.com/espressif/esp-rainmaker-cli/blob/305d461be9e4e7d2d02aa33d7c1537fd54d85735/requirements.txt) file is under `esp-rainmaker/cli`

rmetzner49, 2021-12-27:
@pedrominatel thank you, but following the steps produced errors after a long list of installed files.

  ----------------------------------------
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly

Any idea why I wouldn't be able to use TOPICS?  That seems much less complex.