espressif/arduino-esp32 #6044: captive portal example not working

plusmartin, 2021-12-20:
### Board

ESP32 Dev Module

### Device Description

nothing connected to dev module

### Hardware Configuration

nothing connected to dev module

### Version

v2.0.1

### IDE Name

PlatformIO

### Operating System

macos

### Flash frequency

default

### PSRAM enabled

yes

### Upload speed

default

### Description

trying the captive portal example as is, but returns guru error.

### Sketch

```cpp
same as here: https://github.com/espressif/arduino-esp32/blob/master/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino
```

### Debug Message

```plain
--- Available filters and text transformations: colorize, debug, default, direct, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/cu.usbserial-0001  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
���bb���4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060e30  A0      : 0x80110781  A1      : 0x3ffb3b40  
A2      : 0x3ffcd050  A3      : 0x3ffcd6f4  A4      : 0x3ffccf60  A5      : 0x3ffccf40  
A6      : 0x05040808  A7      : 0x0f040808  A8      : 0x80110624  A9      : 0x3ffb3b00  
A10     : 0x3ffcd060  A11     : 0x3ffcd6f4  A12     : 0x3ffb3b4c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb3b40 0x4011077e:0x3ffb3b80 0x4011d605:0x3ffb3ba0 0x4012260d:0x3ffb3be0 0x401278aa:0x3ffb3c00 0x40111187:0x3ffb3c20 0x400886e1:0x3ffb3c50
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2021-12-20:
Hi @plusmartin, there is nothing wrong with that particular example. It seems the problem is in your environment, Arduino ESP32 Core v2.0.1 is not supported yet by PlatformIO. You can check and find workaround here: https://github.com/platformio/platform-espressif32/issues/619.

VojtechBartoska, 2022-01-03:
@plusmartin Did you take a look on mentioned link above?

closed 2022-01-10 by VojtechBartoska

VojtechBartoska, 2022-01-27:
Issue name changed and pinned to the top as a reference for PlatformIO set up workaround for Arduino core ESP32 v2.0.0 and latest.

chegewara, 2022-04-11:
The whole topic is a bit outdated, since arduino-esp32 v2.0.3+ are using new toolchain. So, if anyone find this topic, this is working to me with current master and v2.0.3-RC1:

```
[env]
platform = https://github.com/platformio/platform-espressif32.git#v3.5.0

platform_packages =
  ; use a special branch
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#master
  toolchain-xtensa32@~2.80400.0
```

VojtechBartoska, 2022-05-02:
Official support announced: https://piolabs.com/blog/news/platformio-oss-april-2022-updates.html