espressif/arduino-esp32 #6046: Update error: Could Not Activate The Firmware

SiavashSkynet, 2021-12-20:
### Board

ESP 32 Dev Module

### Device Description

There is not any unusual configuration.

### Hardware Configuration

The mentioned module is just connected to my PC.

### Version

v2.0.1

### IDE Name

Arduino

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

I want to update my board over internet. So used the below sketch my always gain error : `mismatch chip ID, & Update error: Could Not Activate The Firmware ` 
I really stuck here. My partition scheme setup is on default value.
 

### Sketch

```cpp
/**
   httpUpdate.ino

    Created on: 27.11.2015

*/

#include <Arduino.h>

#include <WiFi.h>
#include <WiFiMulti.h>

#include <HTTPClient.h>
#include <HTTPUpdate.h>

WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("skynet", "skynet");

}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    WiFiClient client;

    // The line below is optional. It can be used to blink the LED on the board during flashing
    // The LED will be on during download of one buffer of data from the network. The LED will
    // be off during writing that buffer to flash
    // On a good connection the LED should flash regularly. On a bad connection the LED will be
    // on much longer than it will be off. Other pins than LED_BUILTIN may be used. The second
    // value is used to put the LED on. If the LED is on with HIGH, that value should be passed
    // httpUpdate.setLedPin(LED_BUILTIN, LOW);

    t_httpUpdate_return ret = httpUpdate.update(client, "http://skynet.com/sample.bin");
    // Or:
    //t_httpUpdate_return ret = httpUpdate.update(client, "server", 80, "/file.bin");

    switch (ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILED Error (%d): %s\n", httpUpdate.getLastError(), httpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
        Serial.println("HTTP_UPDATE_OK");
        break;
    }
  }
}
```

### Debug Message

```plain
E (502649) boot_comm: mismatch chip ID, expected 0, found 3584
HTTP_UPDATE_FAILED Error (9): Update error: Could Not Activate The Firmware
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

closed 2021-12-20 by SiavashSkynet

papakpmartin, 2023-01-17:
@SiavashSkynet Did you close this issue bacause you were able to get things working? I'm seeing the same kind of issue.

Rob58329, 2025-02-25:
The "Could Not Activate The Firmware" failure message you get when trying to upload a new sketch over OTA is usually because the sketch  has been compiled for a different ESP32 chip to the one you are trying to upload the sketch to.

(Eg. a sketch compiled for the ESP32-C3 being uploaded via OTA onto an ESP32-S3.)

The message means that the new sketch has been sucesfully stored on the ESP32's flash, but the OTA software detected it was for the wrong chip, and so has reverted to using the original (old) software again.   (The OTA software allows 2 separate sketches to be stored in flash, and just boots from the most recent valid-sketch.)