espressif/arduino-esp32 #6056: BasicHttpsClient not working in WIFI_MODE_APSTA

Zzo911, 2021-12-23:
### Board

Lilygo TG4 (ESP32 WROVER + SIM7600E)

### Device Description

Using ESP32 Core 2.0.1 and ESP32 WROVER Kit (all versions)

### Hardware Configuration

None

### Version

v2.0.1

### IDE Name

Ardiono 1.8.18

### Operating System

macOS 12.1

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

When the Wifi is set to WIFI_MODE_APSTA, the BasicHttpsClient is returning an error -1 : connection refused.
The sketch only works in WIFI_MODE_STA.
How is it possible to force the WiFiClientSecure to use the STA instead of the AP interface?

### Sketch

```cpp
void sendPushover(String pushTitle, String pushMessage, uint8_t pushPriority) {
  if (WiFi.status() == WL_CONNECTED) {
    WiFiClientSecure *client = new WiFiClientSecure;
    if (client) {
      client -> setInsecure();
      HTTPClient https;
      if (https.begin(*client, pushServer)) {
        Serial.println("Connection to pushover OK");
        https.addHeader("Content-Type", "application/json");
        String httpsRequestData = "{\"token\":\"" + pushToken + \
                                  "\",\"user\":\"" + pushUser + \
                                  "\",\"title\":\"" + pushTitle + \
                                  "\",\"message\":\"" + pushMessage + \
                                  "\",\"priority\":\"" + pushPriority + "\"}";           
        int httpsResponseCode = https.POST(httpsRequestData);
        if (httpsResponseCode > 0) {
          Serial.printf("Pushover response : %s\n", https.getString().c_str());
        } else {
          Serial.printf("Pushover error : %s\n", https.errorToString(httpsResponseCode).c_str());
        }
        https.end();
      } else {
        Serial.printf("Connection Pushover failed");
      }
      delete client;
    }
  } else {
    Serial.println("Not connected to WiFi, using SIM instead");
  }
}
```

### Debug Message

```plain
httpsResponseCode = -1
Pushover response : connection refused
```

### Other Steps to Reproduce

Changing Wifi mode to WIFI_MODE_APSTA

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

vshymanskyy, 2021-12-28:
Probably related to this: https://github.com/espressif/arduino-esp32/issues/6077

Zzo911, 2021-12-28:
More debug info from the Arduino IDE console when WIFI_MODE_APSTA :

19:56:55.137 -> [  6089][V][HTTPClient.cpp:245] beginInternal(): url: https://api.pushover.net/1/messages.json
19:56:55.137 -> [  6089][D][HTTPClient.cpp:297] beginInternal(): protocol: https, host: api.pushover.net port: 443 url: /1/messages.json
19:56:55.172 -> Connection to pushover OK
19:56:55.172 -> [  6109][D][HTTPClient.cpp:594] sendRequest(): request type: 'POST' redirCount: 0
19:56:55.172 -> 
19:56:55.172 -> [  6109][V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 236204
19:56:55.172 -> [  6117][V][ssl_client.cpp:65] start_ssl_client(): Starting socket
19:57:01.934 -> [ 12880][E][WiFiGeneric.cpp:1230] hostByName(): DNS Failed for api.pushover.net
19:57:01.934 -> [ 12881][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -1
19:57:01.934 -> [ 12883][V][ssl_client.cpp:309] stop_ssl_socket(): Cleaning SSL connection.
19:57:01.934 -> [ 12889][D][HTTPClient.cpp:1133] connect(): failed connect to api.pushover.net:443
19:57:01.970 -> [ 12896][W][HTTPClient.cpp:1436] returnError(): error(-1): connection refused
19:57:01.970 -> Pushover error : connection refused
19:57:01.970 -> [ 12914][D][HTTPClient.cpp:404] disconnect(): tcp is closed
19:57:01.970 -> 
19:57:01.970 -> [ 12914][V][ssl_client.cpp:309] stop_ssl_socket(): Cleaning SSL connection.

When using WIFI_MODE_STA it's working but I need APSTA as I'm using a portal too :

20:02:24.333 -> [  3544][V][HTTPClient.cpp:245] beginInternal(): url: https://api.pushover.net/1/messages.json
20:02:24.333 -> [  3544][D][HTTPClient.cpp:297] beginInternal(): protocol: https, host: api.pushover.net port: 443 url: /1/messages.json
20:02:24.333 -> Connection to pushover OK
20:02:24.333 -> [  3564][D][HTTPClient.cpp:594] sendRequest(): request type: 'POST' redirCount: 0
20:02:24.367 -> 
20:02:24.367 -> [  3564][V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 239560
20:02:24.367 -> [  3572][V][ssl_client.cpp:65] start_ssl_client(): Starting socket
20:02:24.401 -> [  3638][V][ssl_client.cpp:141] start_ssl_client(): Seeding the random number generator
20:02:24.439 -> [  3640][V][ssl_client.cpp:150] start_ssl_client(): Setting up the SSL/TLS structure...
20:02:24.439 -> [  3643][I][ssl_client.cpp:171] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
20:02:24.439 -> [  3651][V][ssl_client.cpp:241] start_ssl_client(): Setting hostname for TLS session...
20:02:24.439 -> [  3659][V][ssl_client.cpp:256] start_ssl_client(): Performing the SSL/TLS handshake...
20:02:25.509 -> [  4734][V][ssl_client.cpp:277] start_ssl_client(): Verifying peer X.509 certificate...
20:02:25.509 -> [  4734][V][ssl_client.cpp:286] start_ssl_client(): Certificate verified.
20:02:25.509 -> [  4737][V][ssl_client.cpp:301] start_ssl_client(): Free internal heap after TLS 197988
20:02:25.547 -> [  4745][D][HTTPClient.cpp:1140] connect():  connected to api.pushover.net:443
20:02:25.547 -> [  4753][V][ssl_client.cpp:339] send_ssl_data(): Writing HTTP request with 215 bytes...
20:02:25.547 -> [  4762][V][ssl_client.cpp:339] send_ssl_data(): Writing HTTP request with 155 bytes...
20:02:26.722 -> [  5964][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'HTTP/1.1 200 OK'
20:02:26.756 -> [  5965][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'Date: Tue, 28 Dec 2021 19:02:26 GMT'
20:02:26.756 -> [  5969][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'Content-Type: application/json; charset=utf-8'
20:02:26.756 -> [  5978][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'Transfer-Encoding: chunked'
20:02:26.756 -> [  5986][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'Connection: keep-alive'
20:02:26.792 -> [  5994][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'X-Frame-Options: SAMEORIGIN'
20:02:26.792 -> [  6002][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'X-XSS-Protection: 1; mode=block'
20:02:26.792 -> [  6010][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'X-Content-Type-Options: nosniff'
20:02:26.792 -> [  6018][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'X-Download-Options: noopen'
20:02:26.828 -> [  6026][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'X-Permitted-Cross-Domain-Policies: none'
20:02:26.828 -> [  6036][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'Referrer-Policy: strict-origin-when-cross-origin'
20:02:26.828 -> [  6045][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'Access-Control-Allow-Origin: *'
20:02:26.828 -> [  6054][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'Access-Control-Allow-Methods: POST, OPTIONS'
20:02:26.862 -> [  6063][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'Access-Control-Allow-Headers: X-Requested-With, X-Prototype-Version, Origin, Accept, Content-Type, X-CSRF-Token, X-Pushover-App, Authorization'
20:02:26.862 -> [  6081][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'Access-Control-Max-Age: 1728000'
20:02:26.862 -> [  6089][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'X-Limit-App-Limit: 10000'
20:02:26.899 -> [  6097][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'X-Limit-App-Remaining: 9888'
20:02:26.899 -> [  6105][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'X-Limit-App-Reset: 1641016800'
20:02:26.899 -> [  6113][V][HTTPClient.cpp:1233] handleHeaderResponse(): RX: 'ETag: W/"5a3865eb5766236af14e9e37c22d38f2"'
[... cut ...]
20:02:27.046 -> [  6251][D][HTTPClient.cpp:906] writeToStream():  read chunk len: 0
20:02:27.046 -> [  6256][D][HTTPClient.cpp:382] disconnect(): still data in buffer (2), clean up.
20:02:27.046 -> 
20:02:27.046 -> [  6263][D][HTTPClient.cpp:389] disconnect(): tcp keep open for reuse
20:02:27.046 -> Pushover response : {"status":1,"request":"e899782c-592a-4a8a-8c0e-cd016338dd7c"}
20:02:27.080 -> [  6280][D][HTTPClient.cpp:389] disconnect(): tcp keep open for reuse
20:02:27.080 -> [  6286][V][ssl_client.cpp:309] stop_ssl_socket(): Cleaning SSL connection.

VojtechBartoska, 2022-05-04:
I'm closing the issue as expired due to no answer.

If needed, please reopen it.

Thanks for understanding.

closed 2022-05-04 by VojtechBartoska

lukalafaye, 2023-06-28:
Getting the same issue. It's related to https://github.com/m1cr0lab-esp32/esp-now-network-and-wifi-gateway/issues/6