espressif/arduino-esp32 #6057: LED PWM Resolution is reset to 10 bits every time double ledcWriteTone(uint8_t chan, double freq) function is called

SergiiSenchurov, 2021-12-23:
### Board

esp32 devkit v1

### Device Description

Hardware esp32 devkit v1

### Hardware Configuration

GPIO10 (almost any) is configured to be used with PWM LED

### Version

v1.0.6

### IDE Name

Visual Studio Code with Arduino IDE

### Operating System

MacOS 12.0.1

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

LED PWM Resolution is reset to 10 bits every time double ledcWriteTone(uint8_t chan, double freq) function is called

Workaround:

Modify esp32-hal-ledc.c
//      // in function double ledcWriteTone(uint8_t chan, double freq)
//      // -- comment out
//      // double res_freq = _ledcSetupTimerFreq(chan, freq, 10); //Hard-coded 10 bits timer resolution
//
//      // ++ Add
//      uint8_t group=(chan/8), timer=((chan/2)%4);
//      uint8_t current_resolution = LEDC.timer_group[(group)].timer[(timer)].conf.duty_resolution;
//      double res_freq = _ledcSetupTimerFreq(chan, freq, current_resolution); 

### Sketch

```cpp
/**
 * @brief sets the led dimmer parameters
 * 
 * @param newParameters frequency and duty cycle
 */
void LED_channel::set_pwm_parameters(PWM_parameters newParameters)
{
  PWM_parameters params = assert_received_pwm_params(newParameters);
  if (is_initialized)
  {
    pwm_parameters = params;
    ledcWriteTone(ledChannel, pwm_parameters.frequency);
    ledcWrite(ledChannel, pwm_parameters.duty_cycle);

  }
}
/////////////////////////////////////////////////////////////////////////////////////////
```

### Debug Message

```plain
no error messages
Debug confirms PWM resolution change
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2021-12-24:
1.0.6 is quite old.please try 2.0.2

SergiiSenchurov, 2021-12-24:
This issue persists in 2.0.2 too

Here is the source code 

ledc_timer_config_t ledc_timer = {
        .speed_mode       = group,
        .timer_num        = timer,
        .duty_resolution  = 10, // !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
        .freq_hz          = freq, 
        .clk_cfg          = LEDC_DEFAULT_CLK
    };
    ledc_timer_config(&ledc_timer);
    channels_resolution[chan] = 10; // !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

me-no-dev, 2021-12-25:
but your source code now is not using Arduino API?

P-R-O-C-H-Y, 2022-01-13:
Hi @SergiiSenchurov, 
`ledcWriteTone` is intended to be used in 10 bits mode.
According to your sketch you just need to change frequency.
Then use `double ledcChangeFrequency(uint8_t chan, double freq, uint8_t bit_num)` insted.
Or what is your case why you want to use  `ledcWriteTone`?

closed 2022-02-25 by P-R-O-C-H-Y