espressif/arduino-esp32 #6058: ESP32 devkit v1 WiFi.h and Ethernet.h can not be simultaneously compiled

SergiiSenchurov, 2021-12-23:
### Board

esp32 devkit v1

### Device Description

ESP32 DEVKIT v1 with W5500 Ethernet shield connected over the SPI

### Hardware Configuration

//       static const uint8_t SS    = 5;
//       static const uint8_t MOSI  = 19;
//       static const uint8_t MISO  = 21;
//       static const uint8_t SCK   = 18;

### Version

v1.0.6

### IDE Name

Visual Studio Code with Arduino IDE 1.8.6

### Operating System

MacOS 12.0.1

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Simultaneous compilation of Ethernet.h and WiFi.h makes successfull compilation impossible due to the constructors syntax conflict

Workaround is

// 1. Modify "Server.h" file as follows
// class Server: public Print
// {
// public:
//     // -- Remove
//     // virtual void begin(uint16_t port=0) =0;
//     // ++ Add
//        virtual void begin() = 0;
// };
//
// 2. Modify "WiFiServer.h"
//      // -- Remove void begin(uint16_t port=0);
//      // ++ Add 
//      void begin(void);
//      void begin(uint16_t port);
//
// 3. Modify "WiFiServer.cpp"
//      // ++ Add
//          void WiFiServer::begin(void){
//          begin(0, 1);
//          }
//

### Sketch

```cpp
void setup(){

  

  Ethernet.init(SS);
  module_state.DHCP_result = Ethernet.begin(ethernet_macaddress); // Dynamic IP setup
  module_state.hardware_status = Ethernet.hardwareStatus();

  if (Ethernet.linkStatus() == LinkON)
  {
    serverEthernet.begin();
  }

  WiFi.hostname(WIFI_HOSTNAME);
  WiFi.begin(ssid, pass);
  uint8_t delays_count = 0;
  while ((WiFi.status() != WL_CONNECTED) && (delays_count < MAX_WIFI_CONNECT_RETRIES))
  {
    delay(500);
    delays_count++;
  }

  serverWiFi.begin();
  serverWiFi.setNoDelay(true);
```

### Debug Message

```plain
Debug impossible
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2021-12-23:
You can use the built-in Ethernet library via "ETH.h". The Ethernet.h file you are using is for AVR and is not guaranteed to work with ESP32.

This appears to also be a duplicate of https://github.com/espressif/arduino-esp32/issues/2704

SergiiSenchurov, 2021-12-24:
The issue persists. I cannot find eth.h library

atanisoft, 2021-12-24:
You don't need to "find" the library, it is built-in as part of the Arduino esp32 package you added to the Arduino IDE. You should be able to find a few examples of it's usage in the IDE under the ESP32 examples (look for Ethernet). 

In your code you would swap `#include <Ethernet.h>` with `#include <ETH.h>` and use the `ETH` instance (similar to `Serial`) directly.

VojtechBartoska, 2022-04-11:
Hello, can I consider this as covered?

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

closed 2022-04-22 by VojtechBartoska