espressif/arduino-esp32 #6059: ESP32-S2FN4R2 SPI MOSI VOLTAGE BELOW 1V

Dresch123, 2021-12-24:
### Board

WEMOS S2 MINI using the ESP32-S2FN4R2

### Device Description

Have reproduced this on plain module.
https://www.wemos.cc/en/latest/s2/s2_mini.html

### Hardware Configuration

Connected to TFT SCLK and SDA signal or module not connected to anything. 

### Version

v2.0.1

### IDE Name

Arduino 1.8.15

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

when doing an SPI transfer the MOSI signal (35) becomes less than one volt. SPI CLK is normal 3.3V.
When simply toggling GPIO 35 HIGH/LOW the voltage is normal 3.3V square wave. I have attached scope shots of both SCLK (36) and MOSI (35) during spi.transfer calls. A longer description is at:
https://forum.arduino.cc/t/wemos-s2-mini-spi-esp32-s2-spi-issue/938801
[image]
[image]
[image]
Here I combined the SPI transfer and then digitalWrites in the same loop.
[image]

### Sketch

```cpp
#include "FS.h"
#include "SD.h"
#include "SPI.h"
//See
//https://stackoverflow.com/questions/69101743/how-to-use-custom-pins-for-spi-on-esp32-pico-v3-02
//https://github.com/espressif/esp-idf/blob/v4.2.1/examples/peripherals/spi_master/lcd/main/spi_master_example_main.c
//spi2_host
SPIClass spi2(HSPI);
#define FSPI_MISO   37
#define FSPI_MOSI   35
#define FSPI_SCLK   36
#define FSPI_SS     34

//https://github.com/espressif/arduino-esp32/issues/4716
//SPIClass spi2(HSPI);
//#define SD_miso  17
//#define SD_mosi  16
//#define SD_sck   15
//#define SD_ss    14
 
static const int spiClk = 20000000; // 1 MHz
void setup() {
  // put your setup code here, to run once:

  //spi2.begin(SD_sck, SD_miso, SD_mosi, -1);
   //pinMode(SD_ss, OUTPUT);
}

void loop() {
uint8_t buf[100],i;
    for(i = 0; i < 100; i++)
        buf[i] = i;

    while(1){
    spi2.begin(FSPI_SCLK, FSPI_MISO, FSPI_MOSI, FSPI_SS);
    spi2.beginTransaction(SPISettings(spiClk, MSBFIRST, SPI_MODE3));
    spi2.transfer(buf,100);
    spi2.endTransaction();
    spi2.end();
    //spi2.end();
pinMode(FSPI_MOSI, OUTPUT);
digitalWrite(FSPI_MOSI,HIGH); 
digitalWrite(FSPI_MOSI,LOW);   
digitalWrite(FSPI_MOSI,HIGH);
digitalWrite(FSPI_MOSI,LOW);
digitalWrite(FSPI_MOSI,HIGH); 
digitalWrite(FSPI_MOSI,LOW);
digitalWrite(FSPI_MOSI,HIGH); 
digitalWrite(FSPI_MOSI,LOW);  
delay(5);
    }
}
```

### Debug Message

```plain
There is no error message or compile message. It builds and runs happily.
```

### Other Steps to Reproduce

I have tried this on multiple WEMOS S2 Mini boards and have gotten the  same results or no signal on MOSI.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

0x0fe, 2022-02-08:
what i your scope bandwidth?

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

Dresch123, 2022-04-11:
Is this a hardware version or software version? And where would I get this version?

VojtechBartoska, 2022-04-11:
It's a newest software version of Arduino core esp32, available here: https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1

Take a look on Docs where is explained how to use development release versions: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

Dresch123, 2022-04-11:
OK. Thanks I will. It has been a while since I looked at this issue. I did find a work around. Will have to review my notes and code.

VojtechBartoska, 2022-04-22:
any news on this? Thanks!

VojtechBartoska, 2022-05-06:
Hi @Dresch123, just gladly asking if you were able to test this? We just released 2.0.3 stable version of Arduino core so it will be now easier to test.

VojtechBartoska, 2022-05-26:
Closing as expired, if needed please reopen the issue.

closed 2022-05-26 by VojtechBartoska