espressif/arduino-esp32 #6060: X509 - Certificate verification failed, e.g. CRL, CA or signature check failed

tsctrl, 2021-12-25:
### Board

ESP32 Dev Module

### Device Description

DevKit

### Hardware Configuration

all attached but working fine

### Version

latest master

### IDE Name

IDF Component

### Operating System

Windows 10

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

9600

### Description

`
[ 73564][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():210]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[ 73624][E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -9984
[ 73719][W][HTTPClient.cpp:1400] returnError(): error(-1): connection refused
`
i would ask for anyone help that have similar kind of issue on error above ^
background:
i have a node.js server running on https local with selfsign certificates. the cert was registered to my computer and i am able to load the cert in my postman apps and able to call my local server using lan ip or localhost and also can load the response from server on browser. so i assume there is no issue on the certificate generation part.

i have similar setup as this issue and follow the step but no success:
[https://github.com/espressif/arduino-esp32/issues/5021](https://github.com/espressif/arduino-esp32/issues/5021)

### Sketch

```cpp
WiFiClientSecure *client = new WiFiClientSecure;

client->setCACert(CA_cert);
client->setCertificate(ESP_CA_cert);
client->setPrivateKey(ESP_RSA_key);

http.begin(*client, url);
```

### Debug Message

```plain
`
[ 73564][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():210]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[ 73624][E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -9984
[ 73719][W][HTTPClient.cpp:1400] returnError(): error(-1): connection refused
`
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

tsctrl, 2021-12-27:
update. i am able to use domain name but not using ip. good for my use case thus closing this issue.

closed 2021-12-27 by tsctrl

jojo2massol, 2022-08-02:
I have the same issue:
```
[ 25461][E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():270]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[ 25462][E][WifiClientSecure.cpp:135] connect(): start_ssl_client: -9984
```
I'm using a self-signed certificate, and I need to use it with the IP, since I'm in a local network.
Else, it works using another server with a domain name with a different rootCA.
For now, the only way to communicate is using the insecure mode, but no verification is made, and data can be intercepted using IP spoofing.
I don't know what to do.

nayookyaw, 2022-08-06:
I also have the same error, any Guru can solve the issue?

elap4, 2022-08-25:
Just use `client->setInsecure();`

This allows for self signed servers to be used. You could verify the fingerprint of the servers certificate if you want to ensure you're connecting to the correct server.

jojo2massol, 2022-08-31:
The point of having https is for security ...

elap4, 2022-09-01:
See the second part of the answer on generating certificates

morgan-wild, 2022-09-28:
I had the same issue

> [E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
> [E][WiFiClientSecure.cpp:133] connect(): start_ssl_client: -9984
> Error on HTTP request xxxxxxxxxxxxxxxxx

It was because I used a wrong part of my certificate (let's encrypt).

To solve it, I used [this tool](https://projects.petrucci.ch/esp32/?page=ssl) found [here](https://www.esp32.com/viewtopic.php?t=12083)

Then I typed my host, clicked 'Get certificate' and clicked on 'ISRG Root X1'.
At the bottom, there is directly the code to use for our ESP32 and it worked!

TheClockTwister, 2023-04-27:
For everyone who is still desperately trying to find their mistake at this point:

The ESP32 implementation...
- Only supports RSA key length up to 4096-bit (so no 8192-bit)
- Does not care about alternative DNS names, but only the CN (common name field)