espressif/arduino-esp32 #6061: Bluetoothserial with Password pairing only works on ESP32 v1.0.1

Manasmw01, 2021-12-25:
### Board

ESP32

### Device Description

So, there's this [Code](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBTM/SerialToSerialBTM.ino)
which I am using for getting data using UART and sending that data to the Bluetooth terminal for further processing.
The only thing I want to add here is password pairing for Bluetooth.
I tried the ESP-IDF example named gatt_security_server and it works well. But I faced issues in incorporating Bluetooth Serial in it.
I also searched a lot in this, but all I can see is open issues with no solutions.
PS: The code is working fine if I downgrade the ESP32 version to 1.0.1 in Arduino IDE. I found this method in one of the comments: [LINK](https://github.com/espressif/arduino-esp32/issues/1458?fbclid=IwAR37Havcd_iBA6pJezyD35awZXPOzv-TuWjRbxNnuaYzIAVG_cAVMpkTxwU#issuecomment-567611934)
I was wondering if you guys could help me out so that this code works with the latest ESP32 version.

### Hardware Configuration

Only serial. 

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Works with v1.0.1 but not with the latest board version

### Sketch

```cpp
https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBTM/SerialToSerialBTM.ino
```

### Debug Message

```plain
No
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

hyperlite2, 2022-01-12:
I am also having this issue. It seems to have appeared in 1.0.5 and has been carried through all the way to 2.0.2. Downgrading to 1.0.4 has fixed the pairing issues for me.

euphi, 2022-02-08:
Example code that works fine with v1.0.1 and can't connect with v2 (v2.0.2 and latest):

```C++
#include "BluetoothSerial.h"
BluetoothSerial SerialBT;

uint8_t address[6] = { 0x20, 0x13, 0x01, 0x18, 0x02, 0x26 };
const char *pin = "1234";
bool connected;

void setup() {
  Serial.begin(115200);
  SerialBT.begin("ESPBT", true); 
  SerialBT.setPin(pin);
  Serial.println("The device started in master mode, make sure remote BT device is on!");
  
  connected = SerialBT.connect(address);
  
  if(connected) {
    Serial.println("Connected Succesfully!");
  } else {
    while(!SerialBT.connected(10000)) {
      Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app."); 
    }
  }
}

void loop() {

  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(5);
}
```

Tested with platformio:

```ini
[env:esp32-v1]
platform=espressif32
board = esp32doit-devkit-v1
framework = arduino
build_flags = -DCORE_DEBUG_LEVEL=6

[env:esp32-v2.0.2]
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2idf/platform-espressif32-2.0.2.zip
board = esp32doit-devkit-v1
framework = arduino
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
build_flags = -DCORE_DEBUG_LEVEL=6

[env:esp32-v2-upstream]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32doit-devkit-v1
framework = arduino
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
build_flags = -DCORE_DEBUG_LEVEL=6

```

Debug output is not helpful:

V1 (OK):
```

entry 0x400806a8
[I][BluetoothSerial.cpp:571] _init_bt(): device name set
[I][BluetoothSerial.cpp:243] esp_spp_cb(): ESP_SPP_INIT_EVT
[I][BluetoothSerial.cpp:126] btSetPin(): pin set
The device started in master mode, make sure remote BT device is on!
[I][BluetoothSerial.cpp:803] connect(): master : remoteAddress
[I][BluetoothSerial.cpp:324] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT
[I][BluetoothSerial.cpp:326] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT: spp connect to remote
[I][BluetoothSerial.cpp:351] esp_spp_cb(): ESP_SPP_CL_INIT_EVT
[I][BluetoothSerial.cpp:334] esp_spp_cb(): ESP_SPP_OPEN_EVT
Connected Succesfully!
[V][BluetoothSerial.cpp:307] esp_spp_cb(): ESP_SPP_DATA_IND_EVT len=90 handle=129
[V][BluetoothSerial.cpp:307] esp_spp_cb(): ESP_SPP_DATA_IND_EVT len=1 handle=129
[V][BluetoothSerial.cpp:307] esp_spp_cb(): ESP_SPP_DATA_IND_EVT len=64 handle=129
[V][BluetoothSerial.cpp:307] esp_spp_cb(): ESP_SPP_DATA_IND_EVT len=26 handle=129
L5,004000,0,0,4150,4151,4152,287,0,298,1,240,1699,13250,25930;
$FLB,850,99319,2113,0;
[...]
```

V2 Upstream (not OK):
```
entry 0x400805e4
[␀␘␂␂␂��m␑um����2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   562][I][BluetoothSerial.cpp:622] _init_bt(): device name set
[   567][I][BluetoothSerial.cpp:251] esp_spp_cb(): ESP_SPP_INIT_EVT
[   570][I][BluetoothSerial.cpp:134] btSetPin(): pin set
The device started in master mode, make sure remote BT device is on!
[   581][I][BluetoothSerial.cpp:867] connect(): master : remoteAddress
[  1683][I][BluetoothSerial.cpp:336] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT
[  1684][I][BluetoothSerial.cpp:338] esp_spp_cb(): ESP_SPP_DISCOVERY_COMP_EVT: spp connect to remote
[  1690][I][BluetoothSerial.cpp:363] esp_spp_cb(): ESP_SPP_CL_INIT_EVT
Failed to connect. Make sure remote device is available and in range, then restart app.
Failed to connect. Make sure remote device is available and in range, then restart app.
[ 31849][I][BluetoothSerial.cpp:283] esp_spp_cb(): ESP_SPP_CLOSE_EVT: 0
```
platformio-project:

[esp32-issue6061.zip](https://github.com/espressif/arduino-esp32/files/8027470/esp32-issue6061.zip)

BLE works fine with 2.0.2.

euphi, 2022-02-09:
Is it easily possible to increase debug-level of the IDF/BlueDroid? (using platformio).

I tried the IDF SPP_INITIATOR_DEMO and this also can't connect.
When connecting by directly using the adress, the connection is initiated (ESP_SPP_CL_INIT_EVT) but then fails with
`mW (36287) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 1 closed: Peer connection failed (res: 16)`

SuGlider, 2022-02-10:
Try calling ```setPin()``` before ```begin()```.

``` dart
void setup() {
  Serial.begin(115200);
  SerialBT.setPin(pin);    // setPin() shall come before begin()
  SerialBT.begin("ESPBT", true); 
  Serial.println("The device started in master mode, make sure remote BT device is on!");
  
  connected = SerialBT.connect(address);
  
  if(connected) {
    Serial.println("Connected Succesfully!");
  } else {
    while(!SerialBT.connected(10000)) {
      Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app."); 
    }
  }
}
```

euphi, 2022-02-12:
@SuGlider thanks for the hint, but it does not solve the issue. I suspect that cause is lying "deeper", so I opened an issue at espressif/esp-idf.

euphi, 2022-02-13:
**Workaround:**

I was able to connect with Arduino-ESP32 2.0.2, by using an own SDK configuration that disables SSP.

See https://community.platformio.org/t/using-latest-esp32-arduino-but-with-own-idf-configuration/26000/2 :

platformio.ini:
```ini
[env:esp32-v2-upstream]
platform = https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.2/platform-espressif32-2.0.2.zip
board = esp32doit-devkit-v1
framework = arduino, espidf
platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git
```

SDK-Config: https://github.com/euphi/ESP32_BTTacho/blob/main/sdkconfig.esp32-v2-upstream

Patch for BluetoothSerial.cpp (commenting out symbols that need SSP enabled): 
https://github.com/euphi/ESP32_BTTacho/blob/main/framework-arduino-esp32-BTSerial.patch

cattledogGH, 2022-04-04:
In my testing with trying to connect an ESP32 using the Arduino core as Bluetooth Serial Master to an HC05 in slave mode, failure to connect begins with any version after 1.0.6.  

I think that the issue is with the change from
ESP-IDF v3.3.5 to ESP-IDF 4.4.

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

euphi, 2022-04-11:
As already mentioned the problem lays "deeper":
The problem can be reproduced without using Arduino framework, but IDF only. So root cause is in the IDF: https://github.com/espressif/esp-idf/issues/8394 .

VojtechBartoska, 2022-04-12:
@euphi Thanks for linking IDF ticket, we will investigate this further with ESP-IDF team. Adding this to the Roadmap.

StockBrick, 2022-05-28:
Is mine the same issue?

[https://github.com/espressif/arduino-esp32/discussions/6785](url)

cattledogGH, 2022-10-11:
Ian--
Previously you mentioned a work around with the SDK  configuration and SSP settings. I tired to do what you suggested in the Arduino environment, but was not succesfull.

Can you provide me with any more information on what to settings to change?

Regards
Richard Marantz

The problem can be reproduced without using Arduino framework, but IDF only. So root cause is in the IDF: espressif/esp-idf#8394<https://github.com/espressif/esp-idf/issues/8394> .

Alssales, 2023-01-23:
Has anyone gotten anywhere with this issue? "Bluetoothserial with Password pairing only works on ESP32 v1.0.1 #6061" Really needing a fix for this issue. My program is on hold tell then.........Help

VojtechBartoska, 2023-01-25:
@PilnyTomas Can you please revisit this? Test it if it's still valid as a first step, thank you.

PilnyTomas, 2023-01-27:
I have a question - why are you trying to connect 2 masters together?

PilnyTomas, 2023-01-30:
I'm sorry, I'm confused with the other related issues.
All I can say is that having two ESP32s - one with `SerialToSerialBT` with one line added and one changed:
```
  SerialBT.begin("OBDII"); //Bluetooth device name
  SerialBT.setPin("1234");
```
is working with the default `SerialToSerialBTM`
BTW using the current master.
And if this helps anyone - don't try to connect ESP32 initialized in master mode with the phone, computer, or similar device which is also master - BT standard allows only one master in the pico-network. https://github.com/espressif/arduino-esp32/blob/23d715af1b185fac21558c907af4ff5b86526aa4/libraries/BluetoothSerial/src/BluetoothSerial.h#L42

cattledogGH, 2023-01-30:
I can confirm that with Arduino-ESP32 2.0.6. , BluetoothSerial 2.0.0  I can connect to an HC05 in slave mode with password "1234".

The issue appears to have been fixed by this commit and the Arduino core using idf 4.4.1.
[https://github.com/espressif/esp-idf/commit/70d892fbb81e351b38a11c95964e6df4e4edbceb](url)

PilnyTomas, 2023-01-31:
@cattledogGH I'm glad to hear that.
BTW the link might be broken - I cannot access it.

closed 2023-02-14 by PilnyTomas