espressif/arduino-esp32 #6063: GetServices crashes in 2.0.0+ when connecting to heart rate monitor

FIXMBR, 2021-12-25:
### Board

ESP32

### Device Description

Watchy smartwatch

### Hardware Configuration

[link to hardware documentation](https://watchy.sqfmi.com/docs/hardware)

### Version

latest master

### IDE Name

Arduino VS Code

### Operating System

Arch Linux

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

I'm trying to connect to Polar H10 heart rate monitor.
From what i gathered while it's connecting, function `pClient->getService` invokes `getServices`. Expected behavior is to return services map, instead it crashed with `Stack canary watchpoint triggered (BTU_TASK) `. 

### Sketch

```cpp
/**
   A BLE client example that is rich in capabilities.
   There is a lot new capabilities implemented.
   author unknown
   updated by chegewara
*/
// #define CONFIG_BT_NIMBLE_TASK_STACK_SIZE 32768
#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID(BLEUUID((uint16_t)0x180D));
// The characteristic of the remote service we are interested in.
static BLEUUID charUUID(BLEUUID((uint16_t)0x2A37));

// static BLEAddress *pServerAddress;

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic *pRemoteCharacteristic;
static BLEAdvertisedDevice *myDevice;

static void notifyCallback(
    BLERemoteCharacteristic *pBLERemoteCharacteristic,
    uint8_t *pData,
[... 249 lines cut ...]
0x4011cda7: bta_gattc_start_disc_char_dscp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c line 542
0x4011cd6d: bta_gattc_char_dscpt_disc_cmpl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c line 706
0x4011cda7: bta_gattc_start_disc_char_dscp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c line 542
0x4011cd6d: bta_gattc_char_dscpt_disc_cmpl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c line 706
0x4011cda7: bta_gattc_start_disc_char_dscp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c line 542
0x4011cd6d: bta_gattc_char_dscpt_disc_cmpl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c line 706
0x4011cf53: bta_gattc_disc_cmpl_cback at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c line 1067
0x4010664d: gatt_end_operation at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_utils.c line 2301
0x40101355: gatt_act_discovery at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_cl.c line 116
0x40101836: gatt_process_read_info_rsp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_cl.c line 484
0x401021d5: gatt_client_handle_server_rsp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_cl.c line 1125
0x401037a3: gatt_data_process at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_main.c line 1004
0x401037e2: gatt_le_data_ind at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_main.c line 587
0x401109a9: l2c_rcv_acl_data at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/l2cap/l2c_main.c line 275
0x400ff1a2: btu_hci_msg_process at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/btu/btu_task.c line 144
0x401195e3: osi_thread_run at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/thread.c line 67
```

### Other Steps to Reproduce

The code works fine on 1.0.x versions of library. I've seen other people having the same issue with 2.0.x lib and this hear rate monitor.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

jacobq, 2022-02-24:
Might it be possible to add an option to specify the stack size for the Bluetooth task (`CONFIG_BTU_TASK_STACK_SIZE`) from Arduino similar to what was done for `ARDUINO_LOOP_STACK_SIZE` in https://github.com/espressif/arduino-esp32/pull/5173?

I am running into this with a simple "remote shutter" (HID) device, and it doesn't look like a typical memory leak on my side. Sometimes it happens immediately when the device connects, other times it connects, runs, and even disconnects/reconnects many times before happening.

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

FIXMBR, 2022-04-14:
Unfortunately, I won't be able to test it for a week or so. I will report back once I get to test it.

FIXMBR, 2022-04-20:
Just tested it on the new version and I get the same error. 
I made a merge request that fixes this issue by increasing the `CONFIG_BT_BTU_TASK_STACK_SIZE` to 8k.
https://github.com/espressif/esp32-arduino-lib-builder/issues/54
Making it user configurable might a better solution though.

VojtechBartoska, 2022-04-21:
Thanks for the PR @FIXMBR! We will review it as soon as possible.

me-no-dev, 2022-05-09:
Can not be made user selectable because it's not part of Arduino and is prebuilt. @FIXMBR I will add you change to an upcoming update of the configurations for all chips. Ref: https://github.com/espressif/arduino-esp32/issues/6465

blind-oracle, 2023-07-30:
Hey guys, I'm having exactly the same issue on 2.0.11 and ESP32-S3 board. Is there anything I can tweak to make it work?

Dabolus, 2023-10-12:
Same issue here, even though with a different BT device. Everything works fine when using a standard ESP32, while an ESP32-S3 crashes when using the exact same code. I opened [a PR](https://github.com/espressif/esp32-arduino-lib-builder/pull/141) to increase the stack size for the ESP32-S3 as it was already done to fix this issue for the standard ESP32.