espressif/arduino-esp32 #6064: How app.bin file upload in ESP32-S2-FH2 chip

muzahid13, 2021-12-26:
### Board

ESP32-S2-FH2 chip

### Device Description

[image]
ESP32-S2-FH2 

### Hardware Configuration

App. bin file uploading problem 

### Version

v2.0.1

### IDE Name

Arduino IDE 

### Operating System

Windows 10

### Flash frequency

80mHZ

### PSRAM enabled

yes

### Upload speed

115200

### Description

Dear concern, 
We are going to do mass production soon with ESP32-S2-FH2 chip. We developed prototype with ESP32-S2 development board. Now we purchased ESP32-S2-FH2 chip to upload app.bin file using esp flash tool but the chip cannot be written. We connected
1. ESP chip Tx to TTL Rx
2. ESp chip Rx to TTL Tx
3. ESP chip VDDA to TTL 3.3V
4. ESP chip GND pin 57 to TLL GND
5. ESP chip GPIO0  boot pin to gnd during app.bin file uploading 
6. ESP33 chip EN pin to gnd after boot pin grounding, after few moment EN pin open. 
No app.bin file uploading on ESP32-S2-FH2 chip.

Above the same way to upload the app.bin file to development board but chip didn’t upload 

Need your help. It’s very urgent. We have to go mass production shortly. Thanks 

### Sketch

```cpp
It is not related to sketch
```

### Debug Message

```plain
Not uploaded
```

### Other Steps to Reproduce

Development board is working but chip not work 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Hello, sorry for answering late. Is this still valid? If yes, are you able to retest it with v2.0.3-rc1? Thanks!

VojtechBartoska, 2022-05-03:
Hi,

I'm closing the issue due to no response. 

If needed you can reopen it. 

Thanks for your contribution & understanding!

closed 2022-05-03 by VojtechBartoska