espressif/arduino-esp32 #6065: Compiler Options for assembler in platform.txt has missing longcalls option

helmut64, 2021-12-26:
### Board

ESP32, ESP32-S2, ESP32-C3

### Device Description

Any

### Hardware Configuration

Any

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

n<

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

The platforms.txt has the option -mlongcalls missing for assembler input files:
compiler.S.flags.esp32=**-mlongcalls** -ffunction-sections  ...****
For C, CPP it is included, for .S it is missing (affects ESP32, S2, C3)

### Sketch

```cpp
no
```

### Debug Message

```plain
no
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2021-12-27:
Could you please explain how that is a problem? Options are extracted from the ESP-IDF build system. If they do not show up here, that means that IDF did not use them. Do you have any particular issue?

helmut64, 2021-12-29:
Yes, if the assembler code has function references into the IDF and GCC runtime libraries, such as: ets_delay_us, gpio_output_set_high, you will receive linker errors:
**dangerous relocation: call8: call target out of range:** ets_delay_us, ...
Adding the -mlongcalls to the assembler option fixes the problem.

igrr, 2021-12-29:
Indeed this option is not added to assembler flags in ESP-IDF, we typically use the following pattern in assembly files:
```
l32i  a8, func
callx8 a8
```
I'm unsure if there's any downside enabling it globally. I suppose there could be a couple places in FreeRTOS port where we use `call0` and except that no registers other than a0 will be modified — in this specific case we might need to disable call relaxation. On the other hand, for most user-written assembly code I don't see a way this option could cause issues.

@me-no-dev if we enable this option for assembly files in IDF, it will likely be only for v5.0. So if you'd like to have this option in Arduino-esp32, you'll have to add it manually somewhere.

helmut64, 2021-12-29:
Thanks for the advise, I understand, however my assembler is auto generated (49k lines) and has code links this:
.LBE311:
	.loc 1 379 6 view .LVU80
	call8	gpio_output_set_high
.LVL12:
In prior Arduino ESP32 versions (<2.0.0) it was working fine.

helmut64, 2021-12-29:
BTW: I wrote an ESP32 evolution article for our Arduino Hannover web site: (read it via  google translate). I will fix the C3 deep sleep wakeup pins, anyways the ESP32 series and your native USB, RISC-V, etc. is superior, thank you for all your hard work on it:
https://arduino-hannover.de/2021/12/09/die-esp32-evolution-s2-s3-c3/
Regards Helmut

igrr, 2021-12-30:
I agree that for auto-generated code it's not viable to apply that pattern! I am not sure why it worked in previous versions of Arduino, perhaps the code layout was slightly different and these functions ended up within the range of call instructions immediate values.

helmut64, 2021-12-30:
Yes the code layout can be the reason. What I don't understand, the -mlongcalls option for compiled assembler should not modify the code, but somehow it must do something to arrange the code that the "call8: call target out of range" linker error does not happen.

As C and C++ enables this option, enabling it for the assembler rule will not hurt and resolves the problem.

closed 2025-01-16 by Parsaabasi