espressif/arduino-esp32 #6069: ESP WEB Server to Support HTTPS

tsctrl, 2021-12-27:
### Related area

esp32

### Hardware specification

DevkitC

### Is your feature request related to a problem?

this might be something that was well known but i am not able to justify on any information on the net for https web server on iot devices.

i am using esp webserver with wifi and it works well. no issue. but i am wondering if esp webserver library can be use with https connection or is there any recomendation for other web server library to support https secure connection. this might be not optimal as the cert has expiry on not suitable on iot devices. but i am open for any opinion why http is more suitable on this case where i could just take it as is or what could be the consideration to put the https in esp32

### Describe the solution you'd like

https web server running on esp

zekageri, 2022-09-20:
I would love to see something like this. Even with self signet CA my life would be so much easier.
If i could serve pages with HTTPS the esp could push notification to the user who is using the esp and the possibilities are endless!

Adrianotiger, 2023-03-01:
I would like to see this too. Also because browsers are always more restrictive.
On desktops there are some settings in the browser to allow insecure websockets for example.
On mobile phones this options aren't available and so also a simple PWA is not able to connect to an ESP32.

alexiri, 2024-06-09:
This is becoming more urgent now that web browsers are favoring https and making it harder and harder to use http.

lbernstone, 2024-06-09:
Note that there is a [3rd party library](https://github.com/fhessel/esp32_https_server/) as well as the C-based [ESP-IDF library](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32s3/api-reference/protocols/esp_https_server.html#https-server).  Rewriting WebServer to use https would separate the esp32 and esp8266 codebases even further... Better for someone to wrap the idf functionality into a new 3rd party implementation with backwards compatibility.

JAndrassy, 2024-06-09:
esp8266 supports secure server

closed 2025-01-16 by Parsaabasi