espressif/arduino-esp32 #6077: `WiFiClientSecure` handshake fails after `client.stop()`. Regression in 2.0.2

vshymanskyy, 2021-12-28:
### Board

ESP32

### Device Description

Any board

### Hardware Configuration

Any configuration

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Linux

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

921600

### Description

https://github.com/espressif/arduino-esp32/pull/5945 clears `ssl_client` in several places, which sets `handshake_timeout` to 0.
Consequently, `start_ssl_client` always fails here:
https://github.com/espressif/arduino-esp32/blob/caef4006af491130136b219c1205bdcf8f08bf2b/libraries/WiFiClientSecure/src/ssl_client.cpp#L266

### Sketch

```cpp
#include <WiFiClientSecure.h>

const char* ssid     = "structure8";
const char* password = "lebowski8";

const char*  server = "blynk.cloud";

const char* test_root_ca= R"EOF(
-----BEGIN CERTIFICATE-----
MIIDSjCCAjKgAwIBAgIQRK+wgNajJ7qJMDmGLvhAazANBgkqhkiG9w0BAQUFADA/
MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT
DkRTVCBSb290IENBIFgzMB4XDTAwMDkzMDIxMTIxOVoXDTIxMDkzMDE0MDExNVow
PzEkMCIGA1UEChMbRGlnaXRhbCBTaWduYXR1cmUgVHJ1c3QgQ28uMRcwFQYDVQQD
Ew5EU1QgUm9vdCBDQSBYMzCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEB
AN+v6ZdQCINXtMxiZfaQguzH0yxrMMpb7NnDfcdAwRgUi+DoM3ZJKuM/IUmTrE4O
rz5Iy2Xu/NMhD2XSKtkyj4zl93ewEnu1lcCJo6m67XMuegwGMoOifooUMM0RoOEq
OLl5CjH9UL2AZd+3UWODyOKIYepLYYHsUmu5ouJLGiifSKOeDNoJjj4XLh7dIN9b
xiqKqy69cK3FCxolkHRyxXtqqzTWMIn/5WgTe1QLyNau7Fqckh49ZLOMxt+/yUFw
7BZy1SbsOFU5Q9D8/RhcQPGX69Wam40dutolucbY38EVAjqr2m7xPi71XAicPNaD
aeQQmxkqtilX4+U9m5/wAl0CAwEAAaNCMEAwDwYDVR0TAQH/BAUwAwEB/zAOBgNV
HQ8BAf8EBAMCAQYwHQYDVR0OBBYEFMSnsaR7LHH62+FLkHX/xBVghYkQMA0GCSqG
SIb3DQEBBQUAA4IBAQCjGiybFwBcqR7uKGY3Or+Dxz9LwwmglSBd49lZRNI+DT69
ikugdB/OEIKcdBodfpga3csTS7MgROSR6cz8faXbauX+5v3gTt23ADq1cEmv8uXr
AvHRAosZy5Q6XkjEGB5YGV8eAlrwDPGxrancWYaLbumR9YbK+rlmM6pZW87ipxZz
R8srzJmwN0jP41ZL9c8PDHIyh8bwRLtTcm1D9SZImlJnt1ir/md2cXjbDaJWFBM5
JDGFoqgCWjBH4d1QB7wCCZAA62RjYJsWvIjJEubSfZGL+T0yjWW06XyxV3bqxbYo
Ob8VZRzI9neWagqNdwvYkQsEjgfbKbYK7p2CNTUQ
-----END CERTIFICATE-----
)EOF";

WiFiClientSecure client;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(100);

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  // attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    // wait 1 second for re-trying
    delay(1000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  client.stop();
  // Workaround:
  //client.setHandshakeTimeout(30);

  client.setCACert(test_root_ca);
  Serial.println("\nStarting connection to server...");
  if (!client.connect(server, 443)) {
    Serial.println("Connection failed!");
  } else {
    client.stop();
  }
}

void loop() {
  // do nothing
}
```

### Debug Message

```plain
[  5265][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.

Starting connection to server...
[  5275][V][ssl_client.cpp:61] start_ssl_client(): Free internal heap before TLS 268420
[  5275][V][ssl_client.cpp:67] start_ssl_client(): Starting socket
[  6675][V][ssl_client.cpp:143] start_ssl_client(): Seeding the random number generator
[  6676][V][ssl_client.cpp:152] start_ssl_client(): Setting up the SSL/TLS structure...
[  6680][V][ssl_client.cpp:175] start_ssl_client(): Loading CA cert
[  6726][V][ssl_client.cpp:244] start_ssl_client(): Setting hostname for TLS session...
[  6726][V][ssl_client.cpp:259] start_ssl_client(): Performing the SSL/TLS handshake...
[  6734][E][WiFiClientSecure.cpp:135] connect(): start_ssl_client: -1
[  6737][V][ssl_client.cpp:311] stop_ssl_socket(): Cleaning SSL connection.
Connection failed!
```

### Other Steps to Reproduce

As a workaround for the issue, one can add `client.setHandshakeTimeout(30);` before every connection attempt.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

roboticboyer, 2022-01-16:
> As a workaround for the issue, one can add client.setHandshakeTimeout(30);

The work around is working!
I think the issue come from those instructions which are not yet executed after a stop of the client

`[V][ssl_client.cpp:233] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:242] start_ssl_client(): Certificate verified.`

GioTB, 2022-03-09:
Hi!, i´m having the same issue, after a wifi reconect the client doesn´t connect back due to the timeout set to 0, i also added the client.setHandshakeTimeout() before each connect and it works, hopefully this can be fixed!
Thanks

VojtechBartoska, 2022-04-11:
Hello folks, can you please retest this on v2.0.3-rc1?

VojtechBartoska, 2022-05-03:
any news on this, thanks!

VojtechBartoska, 2022-05-18:
@vshymanskyy do you still face the issue on v2.0.4?

vshymanskyy, 2022-05-18:
Seems to be fixed in 2.0.3

closed 2022-05-18 by vshymanskyy