espressif/arduino-esp32 #6078: ESP32 core 2.0.x: SD read file functions <SD.h> messed up, core 1.06 is fine though

dsyleixa, 2021-12-28:
### Board

Adafruit ESP32 Feather

### Device Description

default Adafruit ESP32 Feather

### Hardware Configuration

ads1115 by i2c, 
Adafruit Featherwing TFT HX3857 with TS

### Version

latest master

### IDE Name

Arduino IDE 1.8.9

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

e.g. I am reading a file array list from my SD, select ist by buttonpad buttons up, down, and right.
When it's a .txt file then I display the content on the screen while blocking parallel running threads:

by 1.0.6 the content is displayed and then return to the calling routine.
By 2.0.2 nothing happens first, by repeated tries the programm completely hangs up and blocks.

BTW,
there is another extra  widget which displays ".bmp" picture files on the screen by Adafruit_Imagereader, which also is fine by 1..0.6 and what hangs up the same way like the .txt files  by 2.0.6.

### Sketch

```cpp
String filelist[100]; // contents the filenames found on my SD

void ExplorerExecFunc() {
   char filename[80]="";
   strcpy( filename, filelist[selectfilenr].c_str() );
   if(strEndsWith(filename,".txt"))
   {
      THREADSUSPENDED=1;
      msleep(1);
      display_mutex.lock();
      display.setRotation(2);
      display.fillScreen(BLACK);

      File myTxtFile = SD.open(filename);
      if (myTxtFile) {
         //display.setFont(&FreeSans9pt7b);
         display.setTextSize(2);
         display.setTextColor(WHITE);
         display.setCursor(0, 0);
         int c;
         while (myTxtFile.available()) {
            c=myTxtFile.read();
            Serial.write(c);
            display.write(c);
         }
         display.setTextColor(RED);
         display.write(20);  // EOF symbol
         display.setTextColor(WHITE);
         display.setFont();
         display.setTextSize(1);

        delay(2000); // or wait for switch button press

         // close the file:
         myTxtFile.close();
         delay(2);
         display_mutex.unlock();
         THREADSUSPENDED=0;
         display.setRotation(TFT_ROTAT);
         display.fillScreen(BLACK);
         ls(filelist, filecount, selectfilenr);
         markPos(cursorfilenr, cursorfilenr);
      }
      display.setRotation(TFT_ROTAT);
   } // if(strEndsWith(...))

}
```
```

### Debug Message

```plain
just first does nothing and later hangs up
```

### Other Steps to Reproduce

complete program: 
https://github.com/dsyleixa/Arduino/tree/master/ESP32_GBox/ESP32_Box023

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

FrankBoesing, 2021-12-28:
I can confirm this, the reason is this change: https://github.com/espressif/arduino-esp32/pull/5988
With enabled log, you'll get CRC errors over and over.

dsyleixa, 2021-12-28:
I checked it, this failure even goes back to 2.0.0.
The latest version before was 1.0.6 ba which all is fine.

FrankBoesing, 2021-12-28:
Then #5988 is a new, additional issue.

closed 2022-01-17 by me-no-dev

dsyleixa, 2022-01-17:
why closed?
any comments, hints, explanations, or links?

P-R-O-C-H-Y, 2022-01-17:
It was closed automatically by PR. 
But it seems, there was an issue before last bug in diskio.cpp.
Iam working on finding that :)

dsyleixa, 2022-01-17:
@me-no-dev 
why closed?
any comments, hints, explanations, or links?

dsyleixa, 2022-01-18:
in Arduino SD class standard conventions it is 
filelist[filecount] = (String)entry.name();
and that must not be changed to .path().

I am basically using the Arduino standard example
https://www.arduino.cc/en/Reference/FileOpenNextFile
modified to write to a String list addiolally to print out.

In the Arduino example there are only
File dir;
File entry;
and ESP32 SD is supposed to work compatible.

I cannot locate anything, I can just see that nothing happens first, by repeated tries the programm completely hangs up and blocks.

P-R-O-C-H-Y, 2022-01-19:
It's so weird that you don't get any error messages.
It will help if you try to catch where the program hangs up.
You can just add some serial prints to lines and see if the program even gets there.

dsyleixa, 2022-01-19:
no, sorry, that is far too complicated.
Original Arduino API don't need that weird sdSelectCard() stuff, I expect it actually to work with standard Arduino code out of the box.
Finally, the ESP API has to be cross-compatible to standard Arduino code examples.

dsyleixa, 2022-01-19:
no, sorry, that is far too complicated.
I also never patch any library code ever.
Original Arduino API don't need that weird sdSelectCard() stuff (and neither does your 1.06 version), I expect it actually to work with standard Arduino code out of the box.
Finally, the ESP API has to be cross-compatible to standard Arduino code examples.

P-R-O-C-H-Y, 2022-01-19:
There is the sdSelectCard() stuff in 1.0.6 but the 2 lines were added later in never version. On another issues with mounting the SD card commenting (removing) the 2 lines solves the initialization problem.

Actually when I run some Arduino example on ESP32 it works normal with version 2.0.2 even opening and writing to file.

Do you have Core debug level set at least to error or warning?