espressif/arduino-esp32 #6079: Arduino lib builder fails with `CONFIG_DISABLE_HAL_LOCKS=y`

Jason2866, 2021-12-28:
### Board

Esp32 board

### Device Description

Fails at compile time

### Hardware Configuration

none

### Version

latest master

### IDE Name

Arduino Lib Builder 

### Operating System

Ubuntu

### Flash frequency

not relevant

### PSRAM enabled

no

### Upload speed

none

### Description

Arduino Lib Builder fails when Option `CONFIG_DISABLE_HAL_LOCKS=y` is set
```
builder/components/arduino/cores/esp32/esp32-hal-i2c-slave.c
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/cores/esp32/esp32-hal-i2c-slave.c: In function 'i2cSlaveDeinit':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/cores/esp32/esp32-hal-i2c-slave.c:363:12: error: 'i2c_slave_struct_t' {aka 'struct i2c_slave_struct_t'} has no member named 'lock'
     if(!i2c->lock){
            ^~
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/components/arduino/cores/esp32/esp32-hal-i2c-slave.c: In function
```

### Sketch

```cpp
https://github.com/Jason2866/esp32-arduino-lib-builder
```

### Debug Message

```plain
Not relevant
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2021-12-29:
Duplicate of https://github.com/espressif/arduino-esp32/issues/5943.

Remove `CONFIG_DISABLE_HAL_LOCKS=y` in your sdkconfig file and it will work as expected.

@me-no-dev Can you fix https://github.com/espressif/arduino-esp32/blob/caef4006af491130136b219c1205bdcf8f08bf2b/cores/esp32/esp32-hal-i2c-slave.c#L363-L366 to be wrapped in a check for locks being configured?

Jason2866, 2021-12-29:
@atanisoft I know. Why is the option there, when I cannot use?
To be clear it will fail in Arduino code when you set the option.
It IS a bug!

atanisoft, 2021-12-29:
> Why is the option there, when I cannot use?

The option is there as there are some use cases for not having locks used, but it does introduce a number of potential issues at runtime when using a multi-core ESP32. I agree it is a bug that it is failing to compile currently and was acknowledged as such in the referenced issue (which shouldn't have been closed without the fix being merged).

Jason2866, 2021-12-29:
I know what it does and what can happen. Using with a single core ESP32 (solo1)
with a custom compiled Arduino framework.

Jason2866, 2022-01-08:
probably solved with #6108

VojtechBartoska, 2022-04-11:
Closing as solved. Thanks for contribuiton.

closed 2022-04-11 by VojtechBartoska