espressif/arduino-esp32 #6080: SD.h: rename File and/or copy File

dsyleixa, 2021-12-28:
### Related area

SD.h

### Hardware specification

for either ESP32 plus SD slot

### Is your feature request related to a problem?

I am facing a certain SD card <SD.h> functionality

### Describe the solution you'd like

similar to 
#include <SD.h>
File myFile; 
SD.remove("example.txt");

now additionally:

SD.rename("example.txt", "newname.txt");
and
SD.copy("example.txt", "newname.txt");

### Describe alternatives you've considered

nothing similar found

closed 2023-06-12 by P-R-O-C-H-Y

dsyleixa, 2023-06-12:
no, rename does not create a 2nd file additionally, it just renames the 1 and only existing one. 
So the copy command still needs to be impemented.

me-no-dev, 2023-06-12:
@dsyleixa `copy` is not part of Arduino SD/FS API. The reason is that there are different types of `copy` and is not necessary that the destination is on the same volume/filesystem as the source, so you should `copy` the data yourself. Just open the source for reading, the destination for writing and manually read/write from one to the other.

dsyleixa, 2023-06-12:
that is what has to be simplified for file copies on the same SD.
e.g.
copy( /Seamus.txt, /backup/Floyd.txt);

prof955, 2025-03-06:
i don't understand why they refuse to understand what is needed. its "copy" guys. very very normal thing to ask. 
I am new to Fs library please tell me if i am missing comething. But what @dsyleixa is asked very normal. I encountered same need for eg. and let me tell you many people are looking for the same thing. Rename as you @P-R-O-C-H-Y say is jusr deleting the original file. creating destination with same content and deleting original file. Why we can't have a "copy" ?? please explain me as i am 5yo.

why we can't have this as a very easy tihng: 
e.g.
copy( "/Seamus.txt", "/backup/Floyd.txt"); ----> "Seamus.txt" will not be removed from the folder. just "Floyd.txt" will be created.

at this point i can't believe that i am explaing how copy works :D this means i am definetly missing something
or this is very very hard to do so no one wants to add to library. this normal too.