espressif/arduino-esp32 #6081: With https://github.com/espressif/arduino-esp32/pull/5988 SD is not usable

FrankBoesing, 2021-12-28:
### Board

WT32-SC01

### Device Description

ESP-Wrover-Kit

### Hardware Configuration

SD Slot

### Version

latest master

### IDE Name

Platformio

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Enabled log shows many many SD- CRC Failures, reading is not possible wit 2.0.2

After revert the [mentioned change](https://github.com/espressif/arduino-esp32/pull/5988), SD starts working again.

### Sketch

```cpp
-----
Any that uses SD.h
```

### Debug Message

```plain
-----
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

AppsByDavideV, 2021-12-29:
I can confirm there is an issue with SD and core 2.02:
Core 2.01 reading text file from sd, to be sent to client took 15 ms
Core 2.02 , same task, took 6000+ ms 
Needed to go back to 2.01

FrankBoesing, 2021-12-29:
@AppsByDavideV: It is OK to just remove these lines: https://github.com/espressif/arduino-esp32/pull/5988/commits/f32c0da7e2e8336c05171cf312541369b45737f2

AppsByDavideV, 2021-12-29:
> @AppsByDavideV: It is OK to just remove these lines: [f32c0da](https://github.com/espressif/arduino-esp32/commit/f32c0da7e2e8336c05171cf312541369b45737f2)

Yes, it works fine now with core 2.02, thanks

schreibfaul1, 2021-12-29:
Likewise, if the lines in https://github.com/espressif/arduino-esp32/commit/f32c0da7e2e8336c05171cf312541369b45737f2 are removed, SD works in V2.0.2, tested with PlatformIO, thanks to @FrankBoesing

VojtechBartoska, 2022-01-04:
Hello @FrankBoesing, we will take a look on this issue. Thanks for reporting.

VojtechBartoska, 2022-01-04:
Related to #6080 and #6078. These 3 issues can be validate and tested together, @P-R-O-C-H-Y.

edouardreg, 2022-01-06:
Hello, the same here : SD read and write very slow since 2.0.2. Speed was around 400-500 ko/s before, now 4 ko/s after 2.0.2 upgrade. Thanks

tobozo, 2022-01-06:
getting random `[sd_diskio.cpp:186] sdCommand(): token error [23] 0x4` and `[sd_diskio.cpp:174] sdCommand(): no token received` under heavy load too.

During a screen capture test, 6 files out of 256 failed to write to SD with an average size of 40kb, this is after applying @FrankBoesing's fix.

P-R-O-C-H-Y, 2022-01-10:
Hello, can you try editing sd_diskio.cpp as PR #6103 and test it?
It should fix the bug.
Thanks

tobozo, 2022-01-10:
@P-R-O-C-H-Y confirmed unbroken SD after applying this PR

Vigeant, 2022-01-14:
Well, after upgrading to 2.02 and applying the fix from PR #6103 as suggested by @P-R-O-C-H-Y my sketch worked for a while and I could read the card again. However, when I resumed work today, it would fail again in the sd_diskio.cpp in the sdWait()<-sdSelectCard() trace. I tried reloading and resetting many times and it would always fail to mount. I reverted my esp32 library back to 1.06 as I had to do before to fix my sd card mounting issues and my card mounted fine. Once the card mounted, I could re-update to 2.02 and apply the PR #6103 fix and my card would keep working. Not sure what happens in the card but it seems like something happens to the card that makes the sdWait() fail in 2.02 that is simply ignored or fixed in 1.06. When running a diff between the 1.06 and 2.02 sd_diskio.cpp you see that the sdSelectCard() is slightly different. In 1.06 it logs that sdWait() failed and keeps going and it still mounts successfully (and fixes the card somehow). In 2.02, if this sdWait fails, it deasserts the ss and errors out stopping the initialization of the card. All that to say that there seem to be more to it than just the PR #6103. There seem to be a fail condition that 1.06 handles better than 2.02 during initialization. Any help from a dev here would be appreciated.

closed 2022-01-17 by me-no-dev

P-R-O-C-H-Y, 2022-01-17:
Status check bug fix is merged now so this issue gets automatically closed. 

Working on reported issues with mounting sd cards :)

P-R-O-C-H-Y, 2022-01-18:
@Vigeant Can you try removing(commenting) these 2 lines from `sd_diskio.cpp` in function `sdSelectCard`?
This can be the issue you are getting. I cannot reproduce the issue so I will be glad if you or someone can test that.
```
bool sdSelectCard(uint8_t pdrv)
{
    ardu_sdcard_t * card = s_cards[pdrv];
    digitalWrite(card->ssPin, LOW);
    bool s = sdWait(pdrv, 300);
    if (!s) {
        log_e("Select Failed");
        //digitalWrite(card->ssPin, HIGH);     comment these 2
        //return false;             
    }
    return true;
}
```
These 2 lines are not in 1.0.6.
Please let me know :) Thanks

tobozo, 2022-01-18:
@P-R-O-C-H-Y this seems to solve the SD mounting issue on ESP32-Wroom and ESP32-S2 and the fix persists across power cycles \o/

P-R-O-C-H-Y, 2022-01-18:
> @P-R-O-C-H-Y this seems to solve the SD mounting issue on ESP32-Wroom and ESP32-S2 and the fix persists across power cycles \o/

Thanks @tobozo for testing. Will look why it was added and if its necessary to have these lines.

Vigeant, 2022-01-18:
Hi @P-R-O-C-H-Y , This is what I am running with since my post and haven't had problems mounting. This seems to be the issue.

P-R-O-C-H-Y, 2022-01-20:
@tobozo @Vigeant When you commented out these 2 lines, the SD wait still fails right? Can you try making sdWait with longer timeout(lets try double = 600 or 1000)? I would appreciate any help with that. I cannot reproduce the issue with my HW and SD cards. Thanks!

```
bool sdSelectCard(uint8_t pdrv)
{
    ardu_sdcard_t * card = s_cards[pdrv];
    digitalWrite(card->ssPin, LOW);
    bool s = sdWait(pdrv, 600); <--- edit to 600 or 1000
    if (!s) {
        log_e("Select Failed");
        digitalWrite(card->ssPin, HIGH); 
        return false;             
    }
    return true;
}
```

tobozo, 2022-01-20:
@P-R-O-C-H-Y mounting fails both at 600 and 1000 on ESP32-S2, but I can't see the error as it happens before USBCDC is ready, and using setDebugOutput(true) seems to create other problems.

ESP32-Wroom doesn't seeem to care about this change, works fine in both situations.

P-R-O-C-H-Y, 2022-01-20:
@tobozo For me on ESP32-S2 even if I change the timeout to 10 it still works... Can you try that with different sd cards? Maybe its more related to HW.

tobozo, 2022-01-20:
@P-R-O-C-H-Y I have three microsd cards (4GB kingston, 32GB sandisk, 16GB sandisk), the 4GB is a faulty one stuck in readonly mode I keep to test edgy cases, all three show similar symptoms.

The hardware isn't very special, the ESP32-S2 device [is an USB dongle with hall sensor connected to gpio0](https://github.com/AprilBrother/ab-hardware/blob/master/wud/wud.pdf) and a SD card. 

In the sketch it's started using `SD.begin(SD_CS, SPI, 40000000);`

```C
#define SD_MISO  37
#define SD_MOSI  35
#define SD_SCK   36
#define SD_CS    34
```

Maybe @AprilBrother can send you a sample of their [ESP32-WUD](https://wiki.aprbrother.com/en/wud.html#wud)? I got mine shipped to europe in less than 10 days.

Vigeant, 2022-01-20:
@P-R-O-C-H-Y here are the requested tests.
## test1 Uncommented the 2 lines to return to 2.02 source:
'''bool sdSelectCard(uint8_t pdrv)
{
    ardu_sdcard_t * card = s_cards[pdrv];
    digitalWrite(card->ssPin, LOW);
    bool s = sdWait(pdrv, 300);
    if (!s) {
        log_e("Select Failed");
        digitalWrite(card->ssPin, HIGH); 
        return false;             
    }
    return true;
}'''
and ran the sd example without power cycling the sparkfun esp32 thing. The card mounted.

## test2 surviving a power cycle
The card failed to mount following a power cycle with following log trace:
```
[   335][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[   335][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[   335][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[   340][E][sd_diskio.cpp:796] sdcard_mount(): f_mount failed: (3) The physical                                                                                                              drive cannot work
[   649][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[   649][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount Failed
```
## test3 increasing wait to 600
Card failed to mount same error as test2

## test3 increasing wait to 1000
Card failed to mount same error as test2

## test4 commenting both lines and trying wait of 100:
'''
bool sdSelectCard(uint8_t pdrv)
{
    ardu_sdcard_t * card = s_cards[pdrv];
    digitalWrite(card->ssPin, LOW);
    bool s = sdWait(pdrv, 100);
    if (!s) {
        log_e("Select Failed");
        //digitalWrite(card->ssPin, HIGH);
        //return false;
    }
    return true;
}
'''
Card mounted and worked fine with smaller delay.
'''
SD Card Type: SDHC
SD Card Size: 30436MB
Listing directory: /
  DIR : System Volume Information
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
...
Removing Dir: /mydir
Dir removed
Writing file: /hello.txt
File written
Appending to file: /hello.txt
Message appended
Reading file: /hello.txt
Read from file: Hello World!
Deleting file: /foo.txt
File deleted
Renaming file /hello.txt to /foo.txt
File renamed
Reading file: /foo.txt
Read from file: Hello World!
1048576 bytes read for 1420 ms
1048576 bytes written for 2756 ms
Total space: 30419MB
Used space: 25MB
'''

P-R-O-C-H-Y, 2022-01-20:
@tobozo  So on the ESP32 it works and ESP32S2 it does not.
Can you try to measure or just ignore the timeout in sdWait and how long it takes to success or if it ever success?

```
bool sdWait(uint8_t pdrv, int timeout)
{
    char resp;
    uint32_t start = millis();

    do {
        resp = s_cards[pdrv]->spi->transfer(0xFF);
    } while (resp == 0x00); //removed timeout, just waiting for resp==0x00

    if (!resp) {
        log_w("Wait Failed");
    }
    return (resp > 0x00);
}
```
If that success, can you measure the time it needed to pass?
```
bool sdWait(uint8_t pdrv, int timeout)
{
    char resp;
    uint32_t start = millis();

    do {
        resp = s_cards[pdrv]->spi->transfer(0xFF);
    } while (resp == 0x00);

    log_w("Time to pass: %d",millis()-start);   //enable core debug level to warning or edit the line :)
   
    if (!resp) {
        log_w("Wait Failed");
    }
    return (resp > 0x00);
}
```

Vigeant, 2022-01-20:
@P-R-O-C-H-Y

## test6 I tried powercycling to see if I could at least get the sdWait() to log the error.
No luck it never logs that it fails ; ( even though its ignored.

## test 7 return to 2.02 source and power cycle to force failed mount then try to mount in a loop.
```
void setup() {
  Serial.begin(115200);
  while (!SD.begin(16, SPI, 25000000)){
    Serial.println("Card Mount loop Failed");
    delay(100);
  }
  /*
  if (!SD.begin(16, SPI, 25000000)) {
    Serial.println("Card Mount Failed");
    return;
  }
  */
...
}
```

trace is

```
Card Mount loop Failed
[ 21520][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 21520][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 21520][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[ 21525][E][sd_diskio.cpp:796] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 21633][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 21633][E][sd_diskio.cpp:126] sdSelectCard():
[ 21844][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 21844][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 21844][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[ 21849][E][sd_diskio.cpp:796] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 21957][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 21957][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount loop Failed
[ 22168][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 22168][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 22168][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[ 22173][E][sd_diskio.cpp:796] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 22281][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 22281][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount loop Failed
[ 22492][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 22492][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 22492][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[ 22497][E][sd_diskio.cpp:796] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 22605][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 22605][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount loop Failed
[ 22816][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 22816][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 22816][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[ 22821][E][sd_diskio.cpp:796] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 22929][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 22929][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount loop Failed
[ 23140][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 23140][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 23140][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[ 23145][E][sd_diskio.cpp:796] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 23253][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 23253][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount loop Failed
[ 23464][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 23464][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 23464][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[ 23469][E][sd_diskio.cpp:796] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 23577][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 23577][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount loop Failed
[ 23788][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 23788][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[ 23788][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[ 23793][E][sd_diskio.cpp:796] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 23901][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[ 23901][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount loop Failed
```

so fails forever...

P-R-O-C-H-Y, 2022-01-20:
@Vigeant 
can you specify SPI pins in setup? Can you try this exact setup? Without setting SD frequency?

```
//change pins you use
#define SDCARD_SPI_MISO 2
#define SDCARD_SPI_MOSI 15
#define SDCARD_SPI_SCK  14
#define SDCARD_SPI_CS   13

void setup() {
  Serial.begin(115200);
  delay(100);
  SPI.begin(SDCARD_SPI_SCK,
            SDCARD_SPI_MISO,
            SDCARD_SPI_MOSI,
            SDCARD_SPI_CS); //SCK, MISO, MOSI, SS (CS)
  while (!SD.begin(SDCARD_SPI_GPIO_CS, SPI)) {
    Serial.println("Card Mount loop Failed");
    delay(100);
  }
...
}

```

Vigeant, 2022-01-20:
Interesting!

```
//change pins you use (Sparkfun esp32 thing with gpio 16 for my card)
#define SDCARD_SPI_MISO 19
#define SDCARD_SPI_MOSI 23
#define SDCARD_SPI_SCK  18
#define SDCARD_SPI_CS   16

void setup() {
  Serial.begin(115200);
  delay(100);
  SPI.begin(SDCARD_SPI_SCK,
            SDCARD_SPI_MISO,
            SDCARD_SPI_MOSI,
            SDCARD_SPI_CS); //SCK, MISO, MOSI, SS (CS)
  while (!SD.begin(SDCARD_SPI_CS, SPI)) {
    Serial.println("Card Mount loop Failed");
    delay(100);
  }
...
}
```

and the trace !

```
[   135][W][sd_diskio.cpp:174] sdCommand(): no token received
[   236][W][sd_diskio.cpp:174] sdCommand(): no token received
[   337][W][sd_diskio.cpp:174] sdCommand(): no token received
[   437][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[   437][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[   438][E][sd_diskio.cpp:796] sdcard_mount(): f_mount failed: (3) The physical                                                                                         drive cannot work
[   448][W][sd_diskio.cpp:174] sdCommand(): no token received
[   553][W][sd_diskio.cpp:174] sdCommand(): no token received
[   654][W][sd_diskio.cpp:174] sdCommand(): no token received
[   754][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
Card Mount loop Failed
[   861][W][sd_diskio.cpp:174] sdCommand(): no token received
[   962][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1063][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1163][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  1163][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[  1164][E][sd_diskio.cpp:796] sdcard_mount(): f_mount failed: (3) The physical                                                                                         drive cannot work
[  1173][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1279][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1380][W][sd_diskio.cpp:174] sdCommand(): no token received
[  1480][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
Card Mount loop Failed
SD Card Type: SDHC
SD Card Size: 30436MB
Listing directory: /
  DIR : System Volume Information
  FILE: test.txt  SIZE: 0
  FILE: foo.txt  SIZE: 13
...
```

Important to note that the size of test.txt should be 1048576 Bytes!!!. So it eventually mounted but clobbered the test.txt file???

P-R-O-C-H-Y, 2022-01-20:
@Vigeant Hmm we are getting somewhere.. Now there is no problem with sdWait but its still failing in mount for some time.

Vigeant, 2022-01-20:
@P-R-O-C-H-Y This time it worked flawlessly

```
SD Card Type: SDHC
SD Card Size: 30436MB
Listing directory: /
  DIR : System Volume Information
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
  DIR : edit
  DIR : recordings
  DIR : www
  FILE: index.htm  SIZE: 5450
Creating Dir: /mydir
Dir created
Listing directory: /
  DIR : System Volume Information
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
  DIR : mydir
  DIR : edit
  DIR : recordings
  DIR : www
  FILE: index.htm  SIZE: 5450
Removing Dir: /mydir
Dir removed
Writing file: /hello.txt
File written
Appending to file: /hello.txt
Message appended
Reading file: /hello.txt
Read from file: Hello World!
Deleting file: /foo.txt
File deleted
Renaming file /hello.txt to /foo.txt
File renamed
Reading file: /foo.txt
Read from file: Hello World!
1048576 bytes read for 3306 ms
1048576 bytes written for 4570 ms
Listing directory: /
  DIR : System Volume Information
  FILE: test.txt  SIZE: 1048576
  FILE: foo.txt  SIZE: 13
  DIR : edit
  DIR : recordings
  DIR : www
  FILE: index.htm  SIZE: 5450
Total space: 30419MB
Used space: 29MB
```

Vigeant, 2022-01-20:
Oh but fails for ever following a power cycle...

P-R-O-C-H-Y, 2022-01-20:
And than is again problem in sdWait().. so commenting out fixes that but that's not a fix :D

P-R-O-C-H-Y, 2022-01-20:
Speed is slower because when you don't specify frequency, it will be set to 4Mhz. Before you used 25Mhz.

Vigeant, 2022-01-20:
exactly. The weird part is that it will work until a power cycle so it points to something with the SPI module initialization maybe?

Vigeant, 2022-01-20:
is there a way to dump the spi module status? I would dump it and compare.

P-R-O-C-H-Y, 2022-01-20:
What if you call SPI.end() ?

Vigeant, 2022-01-20:
ya

P-R-O-C-H-Y, 2022-01-20:
Can you call SD.end() and SPI.end() before restart?

Vigeant, 2022-01-20:
so call SPI.end() before begin()?

P-R-O-C-H-Y, 2022-01-20:
You can try that :) but I mean before you restart CPU end the SD and SPI and after that restart cpu

Vigeant, 2022-01-20:
not sure what you mean with restart cpu. In software using arduino API?

P-R-O-C-H-Y, 2022-01-20:
How are you doing the power cycle?

Vigeant, 2022-01-20:
I disconnect from usb and battery

P-R-O-C-H-Y, 2022-01-20:
When I was testing a was just using software to restart the CPU, will try with disconnecting to. 
So I mean after the routine of listing and writing to file call SD.end() and SPI.end().
After that disconnect and connect board again.
If still doesn't work you can try to call SPI.end before calling begin. But I think it won't make any change

Vigeant, 2022-01-20:
Ya soft reset does not make it fail. Only a full power cycle.

Adding SD.end() and SPI.end() did not help (Had to comment out the lines for this to happen and then uncomment en powercycle...)

P-R-O-C-H-Y, 2022-01-20:
I still have no problems.. Cannot reproduce the issues :/ even I disconnect the board.

P-R-O-C-H-Y, 2022-01-20:
Isn't that related to Pullups resistors now?

Vigeant, 2022-01-20:
not sure. Here is the trace on the first reboot that fixes the card:
```
10:26:38.516 -> [   434][W][sd_diskio.cpp:104] sdWait(): Wait Failed
10:26:38.516 -> [   434][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
10:26:38.516 -> [   434][W][sd_diskio.cpp:180] sdCommand(): crc error
10:26:38.610 -> [   537][W][sd_diskio.cpp:180] sdCommand(): crc error
10:26:38.845 -> SD Card Type: SDHC
10:26:38.892 -> SD Card Size: 30436MB
```

Vigeant, 2022-01-20:
Do you want me to try and configure my CS pin with a pullup before I start my spi bus and sd card?

P-R-O-C-H-Y, 2022-01-20:
Try that on MOSI pin according to this: [sdspi-espidf](https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card/sdspi)

Vigeant, 2022-01-20:
hmm Guess I would need to set MISO to `OUTPUT_OPEN_DRAIN` and add a physical 10k pullup.
Mod needs to be on line 230 of `espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.c`. 
Cannot try this ATM. Does your setup have an external 10k pullup? There is no such thing as OUTPUT_PULLUP...

P-R-O-C-H-Y, 2022-01-20:
I have this SD SPI module [image](https://www.gme.cz/data/product/1024_1024/pctdetail.774-012.1.jpg) 
It seems to have 10k resistors on pins.

Vigeant, 2022-01-20:
ok I am using https://www.sparkfun.com/products/13743
so I do not seem to have a 10k on MISO.
https://cdn.sparkfun.com/datasheets/BreakoutBoards/Shifting_microSD_v10.pdf

P-R-O-C-H-Y, 2022-01-21:
Do you have 5V power to the microSD board?
I'm actually getting out of any ideas why its not working for you. It has to be HW related when I don't have any issue on any ESP boards..

tobozo, 2022-01-21:
@P-R-O-C-H-Y removing the timeout in sdWait created an endless loop, it waits forever.

Tried SD.end / SD.begin in a loop with various delays with no effect. 

It feels like something (SPI driver?) is setting up the pins in a way that persists across reboots.

I'm out of ideas.

P-R-O-C-H-Y, 2022-01-21:
Yeah it seems the problem is with SPI driver, but more weird is that I can't reproduce any issue on official ESPs Devkits boards (tried on ESP32 and ESP32S2).
It has to be HW related. Even more weird is that ignoring the sdWait() failure makes the SD card to fully work right?

AppsByDavideV, 2022-01-21:
Even CS pin should have a 10K pull-up tied to 3.3V

P-R-O-C-H-Y, 2022-01-21:
One more thing to try @tobozo, @Vigeant.
Can you try same test with 2 different SPI speeds? Read somewhere that some cards don't like this freq for init.
Change in file `SPI.cpp` SPICLASS _freq
```
SPIClass::SPIClass(uint8_t spi_bus)
    :_spi_num(spi_bus)
    ,_spi(NULL)
    ,_use_hw_ss(false)
    ,_sck(-1)
    ,_miso(-1)
    ,_mosi(-1)
    ,_ss(-1)
    ,_div(0)
    ,_freq(1000000) <----- can you try higher speed 16000000 (16Mhz) and lower speed 300000 (300kHz)
    ,_inTransaction(false)
{}
```

tobozo, 2022-01-21:
> I can't reproduce any issue on official ESPs Devkits boards 

@P-R-O-C-H-Y maybe enable USBCDC at boot and don't use the hardware serial? 

However I got some freezes when using `USBSerial.setDebugOutput(true)`, maybe it's unrelated, maybe it's part of the problem, but I recommend to not enable it and use occasional `USBSerial.printf` statements instead.

P-R-O-C-H-Y, 2022-01-21:
One more test if you comment out this in `DSTATUS ff_sd_initialize(uint8_t pdrv)` in `sd_diskio.cpp`
```
if (sdTransaction(pdrv, GO_IDLE_STATE, 0, NULL) != 1) {
        log_w("GO_IDLE_STATE failed");
        goto unknown_card;
    }
```

tobozo, 2022-01-21:
@P-R-O-C-H-Y commenting out does not help it just fails on the next transaction.

P-R-O-C-H-Y, 2022-01-21:
> @P-R-O-C-H-Y commenting out does not help it just fails on the next transaction.

Yes failed for me too. 
Can you try changing the SPI speed?

tobozo, 2022-01-21:
tried a few speeds from 16, 27, 40MHz, even 400KHz, using FSPI, HSPI, constructor, pointer, all fail with sdWait, all succeed when I comment out the `return false` in `sdSelectCard`.

So if the problem on ESP32-S2 is SPI related, it's probably wrapped between some `#if CONFIG_IDF_TARGET_ESP32S2` macros (maybe even with an extra ` ||  CONFIG_IDF_TARGET_ESP32`) right?

[9 more replies]

PR #6103 (2022-01-17): BUGFIX - Sd check status
## Summary
Fix ff_sd_status() in sd_diskio.cpp from previous PR, which broke SD card functionality.

## Impact
Fix:
#6081
#6078

## Related links
Previous PR #5988