espressif/arduino-esp32 #6086: ESP32 sethostname in arduino core 2.x not correct

kroon040, 2021-12-31:
### Board

LilyGO TTGO T-Camera ESP32-WROVER

### Device Description

LilyGO TTGO T-Camera ESP32-WROVER

### Hardware Configuration

nothing

### Version

v2.0.1

### IDE Name

Aruido IDE

### Operating System

Linux

### Flash frequency

80mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

In arduino core 1.06  is WiFi.setHostname ok and works with sketch below.

When switching to arduin core 2.x, it fail I see the hostname: esp32-88DB80

I should expect ESP32-Test, this I get switching to core 1.06

### Sketch

```cpp
#include <Arduino.h>
#include <WiFi.h>

const char *_ssid = "kroonen";
const char *_password = "password";

void setup() {
  WiFi.disconnect(true, true);
  Serial.begin(115200);
  while(!Serial) yield();
  Serial.println();

  Serial.println("### === ESP32 Wifi-Test v1.0 === ###");

  Serial.printf(" WiFi mode set to WIFI_STA: %s\n", WiFi.mode(WIFI_STA) ? "Success!" : "Failed!");
  Serial.printf(" Setting hostname: %s\n", WiFi.setHostname("ESP32-Test") ? "Success!" : "Failed!");
  WiFi.begin(_ssid, _password);

  // Warte auf Verbindung
  while ((WiFi.status() != WL_CONNECTED)) {
    delay(100);
    Serial.print(" ");
    Serial.print(WiFi.status());
  }
  if (WiFi.status() == WL_CONNECTED) {
    Serial.println("");
    Serial.print(" Verbunden mit: ");
    Serial.println(WiFi.SSID());
    Serial.print(" IP address: ");
    Serial.println(WiFi.localIP());
    Serial.print("\n");
  }
  else {
    Serial.println("\n ERROR: WiFi Connection not possible!\n");
  }

}

void loop() {

}
```

### Debug Message

```plain
10:41:47.280 -> 
10:41:47.280 -> ### === ESP32 Wifi-Test v1.0 === ###
10:41:47.479 ->  WiFi mode set to WIFI_STA: Success!
10:41:47.479 ->  Setting hostname: Success!
10:41:47.644 ->  6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 6 0 3
10:41:50.133 ->  Verbunden mit: kroonen
10:41:50.133 ->  IP address: 192.168.180.25
10:41:50.133 ->
```

### Other Steps to Reproduce

core 1.06 works, 2.0 and higher fails

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

20162026, 2022-01-03:
Looks like setHostname() function only writes into temporary buffer that is then copied into low level network layer by the first WiFi.mode(WIFI_STA) call.

In your example moving WiFi.setHostname() above WiFi.mode() should solve the issue.

kroon040, 2022-01-04:
Ok, that worked. Thnx.

closed 2022-01-04 by kroon040