espressif/arduino-esp32 #6090: BLEScan->start(0) Async none blocking scanning

ktonder, 2022-01-01:
### Board

ESP32C3

### Device Description

ESP32C3 internal jtag.

### Hardware Configuration

no

### Version

latest master

### IDE Name

Visual Studio Code

### Operating System

Ubuntu 20.04

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

In  prev versions (1.x.x) I used to start scanning with scan(0).

This allowed the code to scan in background, but now its simply stops there. Is this an intended change to 2.x and is there another way to acheive the same functionallity now?

### Sketch

```cpp
void TTBleScanner::begin() {
    BLEDevice::init("");

    pBLEScan = BLEDevice::getScan(); //create new scan
    pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
    pBLEScan->setInterval(80);
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), true, true);
    pBLEScan->setWindow(30);  // less or equal setInterval value
    Serial.pritnln("Starting");
    pBLEScan->start(0);
    Serial.pritnln("Started");
}
```

### Debug Message

```plain
No error happens
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

eos1d3, 2022-01-02:
You are actually using blocking version of `start`. See the header file `"BLEScan.h"`

```
bool           start(uint32_t duration, void (*scanCompleteCB)(BLEScanResults), bool is_continue = false);
BLEScanResults start(uint32_t duration, bool is_continue = false);
```

Only the first one is async version. I am using ESP32 and both work.

ktonder, 2022-01-03:
I can confirm that this is the case, I have changed the code to the following for later reference.

`void TTBleScanner::begin() {
    BLEDevice::init("");

    pBLEScan = BLEDevice::getScan(); //create new scan
    pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
    pBLEScan->setInterval(80);
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks(), true, true);
    pBLEScan->setWindow(30);  // less or equal setInterval value
    Serial.pritnln("Starting");
    pBLEScan->start(0, nullptr, true);
    Serial.pritnln("Started");
}`

Thanks for the help in this matter.

closed 2022-01-03 by ktonder