espressif/arduino-esp32 #6093: OTA fails with ESP32-S2 (ESP.partitionRead)

misery, 2022-01-03:
### Board

ESP32-S2 / ESP12k from AI Thinker

### Device Description

I'm using a naked ESP12k dev board from AI Thinker.

### Hardware Configuration

Nothing connected

### Version

v2.0.1

### IDE Name

esphome

### Operating System

Linux

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

460800

### Description

ESP.partitionRead won't read that data that was written with ESP.partitionWrite.

I'm using esphome and the OTA fails. So I digged into the code and saw that the binary of esphome is correct. Even the md5 check of the Updater.cpp works and is valid.

The data ```buf``` is wrong and the check for the magic byte in line 47 fails. So the whole OTA fails.
```
    if(!ESP.partitionRead(partition, 0, (uint32_t*)buf, ENCRYPTED_BLOCK_SIZE)) {
        return false;
    }

   if(buf[0] != ESP_IMAGE_HEADER_MAGIC) {
        return false;
    }
```
https://github.com/espressif/arduino-esp32/blob/2.0.2/libraries/Update/src/Updater.cpp#L43

I can see that in UpdateClass::_verifyHeader the magic byte is correct and also the buffer contains the correct magic byte ```0xE9``` for ESP.partitionWrite. But if ESP.partitionRead reads it, it contains something like this.

```
\xfc\xed\xdf\xed\xff\xef\xee\xef\xcc\xfe\xed\xef\xfe\xdf\xfd\xec0
```

### Sketch

```cpp
Well, no sketch... just an esphome yaml

esp32:
  board: esp32-s2-kaluga-1

esphome:
  name: esp_demo_s2
  platformio_options:
    platform_packages:
      - framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.2
    platform:
      - https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream

status_led:
  pin: GPIO2

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
```
```

### Debug Message

```plain
I get  Read Failed as errorString. If I skip the magic header check in ```_partitionIsBootable``` I get ```Could Not Activate The Firmware```

```
[D][esp-idf:000]: E (32183) esp_image: image at 0x1d0000 has invalid magic byte (nothing flashed here?)
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2022-01-11:
Since adding #5426 for all unicore esp32x. OTA works in [Tasmota](https://github.com/arendst/Tasmota) for S2 / C3 and the ESP32solo1 (Solo1 needs self compiled Arduino libs)

Pedroalbuquerque, 2022-01-16:
Hi I seem to have the same/identical problem.

I am using the board
"ESP32-S2- WROVER-I ESP32-S2 SMD module 3.3V 2MB PSRAM 4MB SPI flash IPEX antenna connector"

Just performed several uploads using Serial port with success.

when using OTA (I use it for long time for ESP8266 and ESP32), the upload runs fine up to the point  when it is checking, and fails

"> Executing task: platformio run --target upload --environment esp32-s2_ota <

Processing esp32-s2_ota (platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream; framework: arduino; board: esp32-s2-generic)
---------------------------------------------------------------------------------------
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
Verbose mode can be enabled via `-v, --verbose` option
* Adding toolchain toolchain-riscv32-esp with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32 with version 8.4.0+2021r2
* Adding toolchain toolchain-xtensa-esp32s2 with version 8.4.0+2021r2
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s2-generic.html
PLATFORM: Espressif 32 (3.3.1+sha.3784198) > Espressif ESP32-S2 generic
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 0.0.0+sha.caef400 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-mkspiffs 2.230.0 (2.30) 
 - toolchain-esp32s2ulp 1.22851.191205 (2.28.51) 
 - toolchain-riscv32-esp 8.4.0+2021r2 
 - toolchain-xtensa-esp32s2 8.4.0+2021r2
Converting tesp32_s2.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 142 compatible libraries"
...

"esptool.py v3.1
Merged 2 ELF sections
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Uploading .pio/build/esp32-s2_ota/firmware.bin
19:53:39 [DEBUG]: Options: {'esp_ip': '192.168.4.1', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 12916, 'auth': '', 'image': '.pio/build/esp32-s2_ota/firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
19:53:39 [INFO]: Starting on 0.0.0.0:12916
19:53:39 [INFO]: Upload size: 729888
Sending invitation to 192.168.4.1 
Authenticating...OK
19:53:40 [INFO]: Waiting for device...

Uploading: [                                                            ] 0% 
Uploading: [                                                            ] 0% 
Uploading: [                                                            ] 0% 
Uploading: [                                                            ] 0% 
Uploading: [                                                            ] 0% 
"
....
"Uploading: [============================================================] 99% 
Uploading: [============================================================] 100% Done...

19:54:00 [INFO]: Waiting for result...
Traceback (most recent call last):
  File "/Users/pedroalbuquerque/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/espota.py", line 205, in serve
    data = connection.recv(32).decode()
socket.timeout: timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/pedroalbuquerque/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/espota.py", line 353, in <module>
    sys.exit(main(sys.argv))
  File "/Users/pedroalbuquerque/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/espota.py", line 348, in main
    return serve(options.esp_ip, options.host_ip, options.esp_port, options.host_port, options.auth, options.image, command)
  File "/Users/pedroalbuquerque/.platformio/packages/framework-arduinoespressif32@src-537c58760dafe7fcc8a1d9bbcf00b6f6/tools/espota.py", line 220, in serve
    except e:
NameError: name 'e' is not defined
*** [upload] Error 1
"

so seems it is waiting for some info from the board that never arrives.

What Mysery is saying above would justify this same behaviour.

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

misery, 2022-04-22:
@VojtechBartoska Seems to work for me now. :-) Thanks!

VojtechBartoska, 2022-04-26:
Closing as solved, thanks for testing @misery.

If needed you can reopen the issue.

closed 2022-04-26 by VojtechBartoska