espressif/arduino-esp32 #6094: HTTPClient: disconnect() issues

FrankBoesing, 2022-01-03:
### Board

all

### Device Description

all

### Hardware Configuration

all

### Version

latest master

### IDE Name

Platformio

### Operating System

WIndows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

The httpclient disconnect method has two issues:

a)   
```
 if(connected()) {
        if(_client->available() > 0) {
            log_d("still data in buffer (%d), clean up.\n", _client->available());
            while(_client->available() > 0) {
                _client->read();
            }
        }
```
With a fast stream, under certain circumstances (some tasks running), this can takes ages - if not endless - the stream continues to fill the buffer, while the code tries to empty it -> The connection should be stopped before that.

b)
The mehtodf does exactly nothing if this is true:
```
        **if(_reuse && _canReuse) {**
            log_d("tcp keep open for reuse");
        } else {
            log_d("tcp stop");
```
Is that intended? If one calls disconnect, one would expect a disconnect, regardless of _reuse(?)

### Sketch

```cpp
Any
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

FrankBoesing, 2022-01-03:
I now think, this is just to hide a problem in wificlient. It seems to be a workaround...
Wificlient.stop():
```
void WiFiClient::stop()
{
    clientSocketHandle = NULL;
    _rxBuffer = NULL;
    _connected = false;
}
```
So.. what is that? It just sets all pointers to NULL? Especially for _rcBuffer, this is a fat bug, I think :)
But we have flush():
```
// Though flushing means to send all pending data,
// seems that in Arduino it also means to clear RX
void WiFiClient::flush() {
    int res;
    size_t a = available(), toRead = 0;
    if(!a){
        return;//nothing to flush
    }
    uint8_t * buf = (uint8_t *)malloc(WIFI_CLIENT_FLUSH_BUFFER_SIZE);
    if(!buf){
        return;//memory error
    }
    while(a){
        toRead = (a>WIFI_CLIENT_FLUSH_BUFFER_SIZE)?WIFI_CLIENT_FLUSH_BUFFER_SIZE:a;
        res = recv(fd(), buf, toRead, MSG_DONTWAIT);
        if(res < 0) {
            log_e("fail on fd %d, errno: %d, \"%s\"", fd(), errno, strerror(errno));
            stop();
            break;
        }
        a -= res;
    }
    free(buf);
}
```
So... this does, what stop should do, too.

I'd suggest do add a call to flush() to wificlient.stop().
Then , we can remove the ugly workaround in httpclient which tries to empty the buffer (these lines:
```
        if(_client->available() > 0) {
            log_d("still data in buffer (%d), clean up.\n", _client->available());
            while(_client->available() > 0) {
                _client->read();
            }
        }
```
and instead just call stop() without trying to expty the buffer which goes wrong with a continuing stream (i.e. audio datastream, a large download, etc)

FrankBoesing, 2022-01-03:
..this is all a bit...ugly. 
why do we need to malloc even more memory if we try to flush??? This makes not much sense..and shows that the api needs some more care... :)

FrankBoesing, 2022-01-17:
Any thoughts?

VojtechBartoska, 2022-05-03:
any news on this @FrankBoesing ?

FrankBoesing, 2022-05-03:
Due to missing reaction I dont use that anymore and have no time to test it.

VojtechBartoska, 2022-05-04:
Ok, understood. Let's keep the issue open for some time if any other users recall the same problem.

VojtechBartoska, 2022-08-23:
Closing as expired. Thanks for understanding, if needed you can reopen the issue.

closed 2022-08-23 by VojtechBartoska

FedericoBusero, 2022-08-24:
I did a test, and the problem seems to be solved in v2.0.4

VojtechBartoska, 2022-08-25:
@FedericoBusero That's great, thanks for testing.