espressif/arduino-esp32 #6097: Ticker::active() is declared but not implemented.

everslick, 2022-01-04:
### Board

ALL

### Device Description

NONE

### Hardware Configuration

NO

### Version

latest master

### IDE Name

NONE

### Operating System

GNU/Linux

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

see title

### Sketch

```cpp
Ticker t;

bool act = t.active();
```

### Debug Message

```plain
undefined reference to `Ticker::active()'
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

everslick, 2022-01-04:
https://github.com/espressif/arduino-esp32/pull/5919

VojtechBartoska, 2022-01-05:
Hi @everslick, what's the purpose of this issue? Seems to me it can be discussed directly in mentioned PR above.

everslick, 2022-01-05:
I posted it, before I found the PR, I'm fine with closing the issue.

VojtechBartoska, 2022-01-05:
Oki, nevermind. We can move to the PR then. Thanks for your contribution.

closed 2022-01-05 by VojtechBartoska