espressif/arduino-esp32 #6099: Default hostname non unique ?

tablatronix, 2022-01-04:
### Board

esp32 wroom32d

### Device Description

Custom

### Hardware Configuration

no

### Version

latest master

### IDE Name

pio

### Operating System

osx

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

921600

### Description

I expected default hostnames to be ESP-deviceid but I see this instead ?

`[WIFI] HOST: esp32-arduino`

Other stuff like ota is as expected `[OTA] Hostname: esp32-f008d1be63ac`
etc.

```cpp
static bool hostname(const String& aHostname) { return setHostname(aHostname.c_str()); }
...
static char default_hostname[32] = {0,};
static const char * get_esp_netif_hostname(){
	if(default_hostname[0] == 0){
	    uint8_t eth_mac[6];
	    esp_wifi_get_mac((wifi_interface_t)WIFI_IF_STA, eth_mac);
	    snprintf(default_hostname, 32, "%s%02X%02X%02X", CONFIG_IDF_TARGET "-", eth_mac[3], eth_mac[4], eth_mac[5]);
	}
	return (const char *)default_hostname;
}
static void set_esp_netif_hostname(const char * name){
	if(name){
		snprintf(default_hostname, 32, "%s", name);
	}
}
```

Is this an expected change, or documented?

### Sketch

```cpp
#include <WiFi.h>

const char* ssid = "..........";
const char* password = "..........";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
}

  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.print("Hostname: ");
  Serial.print(WiFi.getHostname());
}

void loop() {
}
```

### Debug Message

```plain
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.39, MASK: 255.255.255.0, GW: 192.168.1.1

[WIFI] CONNECTED
[WIFI] IP: 192.168.1.39
[WIFI] HOST: esp32-arduino
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Hello, sorry for answering late. Is this still valid?

VojtechBartoska, 2022-04-11:
Please retest this on v2.0.3-rc1 of Arduino esp32 core, thanks!

tablatronix, 2022-05-05:
Missed this will test right now

VojtechBartoska, 2022-05-18:
Hello @tablatronix, any news?

tablatronix, 2022-05-18:
Looks ok
`HOSTNAME: esp32-480E40`

I usually use full chipid not just 32bit int, but whatever makes more sense. I would have to check what esp8266 does.
`BED108F0` etc

VojtechBartoska, 2022-06-14:
Can I consider this as solved @tablatronix? Thanks for answer!

closed 2022-06-14 by tablatronix

tablatronix, 2022-06-25:
@VojtechBartoska please reference the commit that fixed this, I think I found a regression

tablatronix, 2022-06-25:
It is also missing from release notes

me-no-dev, 2022-06-25:
what is that regression @tablatronix ? `esp32-arduino` no longer exists as string anywhere and hostname is set based on MAC

tablatronix, 2022-06-25:
It is not about the default hostname, but WHEN we can set a custom hostname.
It was that on esp8266 and esp32 previous we could set hostname AFTER enablesta, but now it has to occur before.

So maybe this was an sdk implementation change because there is also a specific note about having to restart 

>  * Please note that when the hostname is altered after interface started/connected the changes
> * would only be reflected once the interface restarts/reconnects

Ill test some more and open a new issue if I find that there is a bad order of operations or overwrite happening instead.

me-no-dev, 2022-06-25:
This is how it is done in IDF. Hostname must be set before the STA interface contacts the DHCP server. 8266 has nothing to do with anything here anymore :)

tablatronix, 2022-06-26:
Ok i found 2 issues kind of related to this issue
( the hostname code is a bit convoluted at the moment it seems, could use some cleanups)

- [ ] I now get 2 different default hostnames!

```
setup(){
 Serial.println("[WIFI] HOSTNAME: " + (String)WiFi.getHostname());
}

[WIFI] HOSTNAME: esp32-02403F

setup(){
 WiFi.mode(WIFI_STA);
 Serial.println("[WIFI] HOSTNAME: " + (String)WiFi.getHostname());
}

[WIFI] HOSTNAME: esp32-02403F
```

My code was checking hostname to restart for changes, but I found an issue there also

- [ ] wifigeneric set/get hostnames affects default_hostname static var.. NOT SDK netif hostnames.
  * hostname is only set if mode() changes, so not always on changes, confusing
  * gethostname gets whatever you set it to last, caveat to the previous..
  
```C++
// get the actual hostname as you would see it on the network
// testing, no safeties or netif or init checking here
const char * WiFiGenericClass::getHostname()
{
	const char * hostname = NULL;
	esp_netif_get_hostname(get_esp_interface_netif(ESP_IF_WIFI_STA), &hostname);
	return hostname;
    // return get_esp_netif_hostname();
}
```

me-no-dev, 2022-06-26:
the thing is that in order to set the hostname, first the network subsystem must be initialized. To actually have that hostname visible by the DHCP server, it must be set before the initial request for IP. Even if you set it after DHCP has started, it will not be applied. So all in all, we have tried to make it as workable as possible, so it is set when the proper event occurs.

About getHostname, maybe we can add `getActiveHostname` to query the netif directly, else there will be inconsistencies between what you set and what you get with the other API

tablatronix, 2022-06-26:
When is the request for ip , do you know when that happens, is it after esp_wifi_start or inside it?

Yeah the biggest issue is that there is no way for user code to know when and if the hostname was set, since it occurs abstracted inside mode() and only if it makes it past the mode already set check. I will work on a workaround for alwyas checking mode when setting hostname ( I assume this all occurs inside esp_wifi_start ). and see what delays are introduced when having to wifistop/wifistart.

a `getActiveHostname` would certainly help there.

me-no-dev, 2022-06-26:
I just looked at the source in IDF. The hostname is sent to the DHCP server on DISCOVER, SELECT, RENEW, REBIND and REBOOT. Maybe restarting the DHCP can help change the hostname.
Precursor to setting the hostname is calling `wifiLowLevelInit` and it can be set any time before connecting.
ESP-IDF has a default hostname "espressif" on the version that we use. I did not check in IDF5.