espressif/arduino-esp32 #6100: ESP32 - BLE Communication - Connection issues with different frequency value from crystal oscillator

arunprakash2611, 2022-01-05:
### Related area

Bluetooth 

### Hardware specification

ESP32 WROOM - 32U

### Is your feature request related to a problem?

Hi,

Thanks for your time,

We are working on multiple ESP32 module to finalize the one of bluetooth related project.

while working with multiple ESP32 development kit, we facing different result from different board instead one unique results since we are using same firmware all the hardware and same scenarios 

Case 1 - While checking some of the board with own developed software application (both mobile as well as windows) - its working fine
Case 2 - But some of the board, working fine on mobile application alone instead of windows also

While checking in deeply, we found different kind of frequency available for different boards like 39.9999 MHz and 39.9969 MHz, 

So exact frequency required to connect properly for Bluetooth communication? And what is the exact frequency to operate the bluetooth communication

Note : I mean 40 Mhz is mandatory for BLE Communication? or what is the operating frequency value ?

[image]
[image]
 
 

### Describe the solution you'd like

Need a exact or particular range of crystal frequent to operate BLE communication properly

h2zero, 2022-01-14:
BLE operates on the same frequency as WiFi, in the 2.4ghz range. I highly doubt the 3khz difference you're seeing from the clock between boards is the cause of your issue.

chegewara, 2022-01-14:
@h2zero actually it may be the problem, because ble specs has more strict requirements about +/-ppm.
I would ask this on esp-idf as it is not related to arduino and you can find nore educated and experienced expert in there.

h2zero, 2022-01-14:
@chegewara Yes, you're correct. I was too tired to do the math but it looks like the xtal in this case is -75ppm, most specs I have seen call for +/- 40-60ppm.

chegewara, 2022-01-14:
oh, i didnt mean you, i was tellin about myself not being experienced and educated, sorry for misunderstanding.

h2zero, 2022-01-14:
>oh, i didnt mean you, i was saying about myself not being experienced and educated, sorry for misunderstanding.

Lol I didn't take it that way, but you're still correct anyway :)

VojtechBartoska, 2022-04-11:
Hello, is this still valid?

VojtechBartoska, 2022-05-04:
Closing as expired due to no answer.

If needed, reopen the issue.

Thanks for understanding.

closed 2022-05-04 by VojtechBartoska