espressif/arduino-esp32 #6101: BasicOTA with ESP32 and Arduino 2.0.0 rc3 = espota.py error

tombo9999, 2022-01-05:
### Board

Esp32-Cam

### Device Description

In Arduino Boards Manager selected ESP32 Wrover Module

### Hardware Configuration

Nothing

### Version

other

### IDE Name

Arduino 2.0.0 rc3

### Operating System

macOS 12.1 and Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I like BasicOTA because it's simple ;)
Tried on Windows 10 + Arduino IDE 1.8.19 and MacOS Monterey 12.1 + Arduino IDE 1.8.18 and it works without any problem.
Wanted to try Arduino 2.0.0 rc3 nightly build, can compile, but can't upload.
I'm just a hobbyist, not a programmer, surely I'm making some silly mistake.

MacOS
Sketch uses 690349 bytes (52%) of program storage space. Maximum is 1310720 bytes.
Global variables use 41348 bytes (12%) of dynamic memory, leaving 286332 bytes for local variables. Maximum is 327680 bytes.

Compilation complete.
Usage: espota.py [options]

espota.py: error: option -p: invalid integer value: '{upload.port.properties.port}'
Upload error: Failed uploading: uploading error: exit status 2

Windows 10 (same error)
...
espota.exe: error: etc

### Sketch

```cpp
Just BasicOTA
```

### Debug Message

```plain
espota.py: error: option -p: invalid integer value: '{upload.port.properties.port}'
Upload error: Failed uploading: uploading error: exit status 2
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

jandrassy, 2022-01-05:
https://github.com/arduino/arduino-ide/issues/740

VojtechBartoska, 2022-04-11:
Hello, cis this still valid?

per1234, 2022-04-11:
Hi @VojtechBartoska. Unfortunately yes it still is.

**`@me-no-dev`** did an investigation and reported the findings here:
https://github.com/arduino/arduino-ide/issues/740#issuecomment-1086595394

VojtechBartoska, 2022-04-11:
Adding to the roadmap and we will investigate this more.

lsdlsd88, 2024-03-10:
still happening. ide 2.3.2 on win 10

rftafas, 2025-01-10:
This is referring version 2.0.x and seems to have a solution appointed above. Please, re-test this with the latest release, update the bug report to current requirements and re-open if the problem persists.