espressif/arduino-esp32 #6105: ESP32 ETH.h linkUp never reprots true

MartinMueller2003, 2022-01-07:
### Board

ESP32 All

### Device Description

LoLin D32 Pro. ESP32-CAM, TTGO - all

### Hardware Configuration

LAN 8720 Ethernet Adapter

### Version

v2.0.1

### IDE Name

Arduino 1.8.1

### Operating System

ESP32

### Flash frequency

1

### PSRAM enabled

yes

### Upload speed

900k+

### Description

ETH.h has no code that ever sets the link up flag to true. This results the linkUp function to always report false even when the link is up and passing data. The link up flag is used when compiled for any ESP32 platform.

### Sketch

```cpp
Example sketch from the repository modified to call ETH.linkUp and print result.
```

### Debug Message

```plain
No debug message needed. Code review identified this issue
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

MartinMueller2003, 2022-01-07:
I have fixes for this and would be happy to submit them to the branch.

TD-er, 2022-01-12:
Yep, this is a bug indeed, introduced recently as it used to work just fine.

https://github.com/espressif/arduino-esp32/blob/caef4006af491130136b219c1205bdcf8f08bf2b/libraries/Ethernet/src/ETH.cpp#L517-L524

The enum `eth_link` is never set other than what the constructor does.

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

TD-er, 2022-04-11:
> Hello, can you please retest this on v2.0.3-rc1?

This also still has the same code as when it was reported:

2.0.3-RC1: https://github.com/espressif/arduino-esp32/blob/1e388a24cef1b336990b8f67dbce63cc8a4794a4/libraries/Ethernet/src/ETH.cpp#L541-L548

VojtechBartoska, 2022-04-11:
Adding to the roadmap.

DoctorDeatch, 2022-08-23:
As I understand it, it has not been repaired to this day?

TD-er, 2022-08-23:
> As I understand it, it has not been repaired to this day?

That's correct.
In my software I act on the events. 
However this is a bit tricky, because no Ethernet event can be used more than once.
So on a "disconnect" event, I destruct the Eth object and create it again.

It is a seriously hacky fix, but at least it is workable.

DoctorDeatch, 2022-08-23:
I have so far bypassed this problem by using a low-quality pin and connecting it directly to the LEDlink of the network card :)

As I understand it, the same implementation problem does not allow changing the settings from a static address to DHCP, without reinitializing ETH.begin();

DoctorDeatch, 2022-08-24:
> That's correct. In my software I act on the events. However this is a bit tricky, because no Ethernet event can be used more than once. So on a "disconnect" event, I destruct the Eth object and [...]

please tell me what procedure you are performing this?
and also, I can't figure out in any way what type of variable is the value that subnet Mask() returns?

TD-er, 2022-08-24:
Well my code is rather complex and to be honest I hate how complex it has become to overcome all quirks regarding WiFi and now also Ethernet events and things not working how you'd expect them to.
One of the things I did was to make an object to keep track of what states and events I've seen and processed.
For ethernet, this object is `EthEventData` (in my project, not part of the SDK)

Example of how I keep track of states while establishing a connection:
https://github.com/letscontrolit/ESPEasy/blob/2923ec74a58c0933d337c98eabed0424f70bf294/src/src/ESPEasyCore/ESPEasyEth.cpp#L138-L181

Implementation of my `EthLinkUp()`
https://github.com/letscontrolit/ESPEasy/blob/2923ec74a58c0933d337c98eabed0424f70bf294/src/src/ESPEasyCore/ESPEasyNetwork.cpp#L241-L252

The subnet mask is also just an IP address:
https://github.com/letscontrolit/ESPEasy/blob/2923ec74a58c0933d337c98eabed0424f70bf294/src/src/ESPEasyCore/ESPEasyNetwork.cpp#L87-L99

ppescher, 2022-12-13:
My own quick fix was:
```cpp
bool ETHClass::fullDuplex()
{
#ifdef ESP_IDF_VERSION_MAJOR
    eth_duplex_t link_duplex;
    esp_eth_ioctl(eth_handle, ETH_CMD_G_DUPLEX_MODE, &link_duplex);
    return (link_duplex == ETH_DUPLEX_FULL);
#else
    return eth_config.phy_get_duplex_mode();
#endif
}

bool ETHClass::linkUp()
{
#ifdef ESP_IDF_VERSION_MAJOR
    return WiFiGenericClass::getStatusBits() & ETH_CONNECTED_BIT;
#else
    return eth_config.phy_check_link();
#endif
}
```

Not sure if this could be definitive, but it looks good to me, except the `#ifdef` that should be replaced with the correct version: every other place in ETH.cpp uses `#if > 3` but I don't know when those changes have been introduced in the ESP-IDF.

TD-er, 2022-12-13:
Thanks @ppescher 
I will try this code to see if it is working.
If so, then it would be great to see it in the main repo.

PR #7593 (2022-12-19): Fix Ethernet library link status and missing functionality with recent IDF versions
## Description of Change
The variable `eth_link` was never updated, so it was always reporting link down.
Proposed fix is to use the status bits managed by WiFi callbacks, that are updated correctly.
Link duplex mode did not have an implementation for IDF versions > 3 and actually the version check was done incorrectly with the preprocessor.
Proposed fix is using the same ioctl mechanism as for the link speed.

## Tests scenarios
Tested with PlatformIO and the latest stable release 2.0.5 on a custom hardware us [... cut]
closed 2022-12-19 by me-no-dev