espressif/arduino-esp32 #6107: Extend SD card select timeout

NomAnor, 2022-01-07:
### Board

DOIT ESP32 DEVKIT V1

### Device Description

I'm using a common Arduino SD Card PCB that is connected with 14cm wires to the ESP32

### Hardware Configuration

I'm using the VSPI Pins 18, 19, 23 and 5

### Version

other

### IDE Name

Arduino IDE

### Operating System

Arch Linux

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Updating to arduino-esp32 v2.0.2 caused the SD-Card to not work anymore (previously I used 1.0.6). The solutions in the other issues didn't work.

Reverting all changes in the file sd_diskio.cpp gave me the same functionality as in 1.0.6. 

Looking at the code I suspected the sdWait timeout in the [sdSelectCard](https://github.com/espressif/arduino-esp32/blob/caef4006af491130136b219c1205bdcf8f08bf2b/libraries/SD/src/sd_diskio.cpp#L124) function, because in 1.0.6 a timeout was just ignored but this is now fixed. In other places ([sdWriteBytes](https://github.com/espressif/arduino-esp32/blob/caef4006af491130136b219c1205bdcf8f08bf2b/libraries/SD/src/sd_diskio.cpp#L229)) a timeout of 500ms is used so I changed the value to that and now my card is recognized.

For me just chaning the timeout in v2.0.2 fixed the issue.

### Sketch

```cpp
Example SD_Test sketch
```

### Debug Message

```plain
Before changing the timeout:

``
[   313][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[   313][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[   313][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[   318][E][sd_diskio.cpp:795] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[   627][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[   627][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Card Mount Failed

After changing the timeout:

```
[    14][W][sd_diskio.cpp:186] sdCommand(): token error [59] 0x5
SD Card Type: SDSC
SD Card Size: 1947MB
```
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-01-07:
Hello, it's  a duplicate of #6081 and will be closed by merging #6103

P-R-O-C-H-Y, 2022-01-20:
@NomAnor Hi, changing timeout to 500ms solves the mounting issue for you? :) Thanks

NomAnor, 2022-01-20:
Sadly not. At first it worked but after a re-upload I had the same problem again. In the end I reverted all changes to `sd_diskio.cpp` back to 1.0.6. I still got the "Select failed" message but because the return was missing in

https://github.com/espressif/arduino-esp32/blob/caef4006af491130136b219c1205bdcf8f08bf2b/libraries/SD/src/sd_diskio.cpp#L125-L129

that was ignored. The missing return is clearly a bug but ignoring the failure kept the code going and I could read files. My SD-Card was rather old maybe it has to do with that.

PR #6162 (2022-01-31): Fix - SD mount issue
## Summary
On many issues with SD card mount, there is problem with sdWait() only before sending the 1st init cmd 0 = GO_IDLE_STATE. After this command sdWait() works as expected.

Edited GO_IDLE_STATE command:
-> sdWait() is ignored there, only warning message is logged.

Only in sdSelectCard timeout was 300 so added longer timeout as in other calls for sdWait.
According to issue #6107, making a timeout of 500 as in other calls should solve the issue.

Cannot reproduce issues with mounting SD cards.
Will appreciate someone to try this fix.

## Impact
Fixing SD card mounti [... cut]
closed 2022-01-31 by me-no-dev