espressif/arduino-esp32 #6114: no OTA example is working for AI Thinker ESP32-CAM (wrong partition table?)

beta-tester, 2022-01-08:
### Board

AI Thinker ESP32-CAM

### Device Description

- USB-to-Serial adapter. 3.3V logaic level, 5V power.

### Hardware Configuration

- WLAN router,
- Raspberry Pi as http server.

### Version

latest master

### IDE Name

Arduino IDE 1.8.19 (Linux 64 bits & Linux ARM 32 bits)

### Operating System

Ubunti 21.10 (64bit) & RasPiOS Buster / Bullseye (32bit)

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

i can successfuly update my **ESP32 Dev Module**s via OTA by using the examples skeches: WebUpdate.ino/httpUpdate.ino.

but when i switch to th Arduino IDE to **AI Thinker ESP32-CAM** and compile the skeches for that and flash it to the Ai Thinker ESP32-CAM (compatible) modules and try to update the device via OTA it crashes. with 
```
abort() was called at PC ...

ELF file SHA256: 0000000000000000

Backtrace: ...
```

### Sketch

```cpp
- /libraries/ArduinoOTA/examples/OTAWebUpdater/OTAWebUpdater.ino
- /libraries/HTTPUpdate/examples/httpUpdate/httpUpdate.ino
- /libraries/WebServer/examples/WebUpdate/WebUpdate.ino
```

### Debug Message

```plain
abort() was called at PC ...

ELF file SHA256: 0000000000000000

Backtrace: ...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

beta-tester, 2022-01-09:
i think the partition table settings for that AI Thinker ESP32-CAM board in arduino esp32 is not compatible for OTA skeches.
i read something about it.

**WORKAROUND:**
i got OTA working for **AI Thinker ESP32-CAM** boards by editing the `~/.arduino15/packages/esp32/hardware/esp32/2.0.2/boards.txt` file:

- i duplicated the block of `esp32cam`
- and renamed `esp32cam` to `esp32camota` of the duplicated block
- then i changed `esp32camota.name=AI Thinker ESP32-CAM` to `esp32camota.name=AI Thinker ESP32-CAM (for OTA)`
- then i changed `esp32camota.build.partitions=huge_app` to `esp32camota.build.partitions=default`.

after a new start of Arduino IDE i now have an additional entry in the ESP boards list with the name: "**AI Thinker ESP32-CAM (for OTA)**".
by using this board settings, compiling and flashing the OTA example skeched to the **AI Thinker ESP32-CAM** this device is now able to successfully update over the air.

i don't know if it is the right way, if `default` is the right and optimal setting for OTA in the **AI Thinker ESP32-CAM** modules but it works as far as i can see.
my suggestion is to add an official kind of permanent fix to the repository.

Hieromon, 2022-02-03:
The AI-Thinker ESP32-CAM board definition described in the [`boards.txt`](https://github.com/espressif/arduino-esp32/blob/3a96fc0e4a166a9221f4a4ab259747d484f19499/boards.txt#L8042) only has a partition schema with no reservation for OTA space as a [`huge_app.csv`](https://github.com/espressif/arduino-esp32/blob/master/tools/partitions/huge_app.csv). A modification to `boards.txt` to reserve the OTA area is to list the partition schema choices with `menu.PartitionScheme` in the `esp32cam` section.

```ini
esp32cam.name=AI Thinker ESP32-CAM

esp32cam.upload.tool=esptool_py
esp32cam.upload.maximum_size=3145728
esp32cam.upload.maximum_data_size=327680
esp32cam.upload.flags=
esp32cam.upload.extra_flags=
esp32cam.upload.speed=460800

esp32cam.serial.disableDTR=true
esp32cam.serial.disableRTS=true

esp32cam.build.tarch=xtensa
esp32cam.build.bootloader_addr=0x1000
esp32cam.build.target=esp32
esp32cam.build.mcu=esp32
esp32cam.build.core=esp32
esp32cam.build.variant=esp32
esp32cam.build.board=ESP32_DEV
esp32cam.build.flash_size=4MB
esp32cam.build.partitions=default
esp32cam.build.defines=-DBOARD_HAS_PSRAM -mfix-esp32-psram-cache-issue -mfix-esp32-psram-cache-strategy=memw
esp32cam.build.extra_libs=
esp32cam.build.code_debug=0

esp32cam.menu.PartitionScheme.default=Default(3MB No OTA/1MB SPIFFS)
esp32cam.menu.PartitionScheme.default.build.partitions=huge_app
esp32cam.menu.PartitionScheme.default.upload.maximum_size=3145728
esp32cam.menu.PartitionScheme.no_ota=No OTA (2MB APP/2MB SPIFFS)
esp32cam.menu.PartitionScheme.no_ota.build.partitions=no_ota
esp32cam.menu.PartitionScheme.no_ota.upload.maximum_size=2097152
esp32cam.menu.PartitionScheme.min_spiffs=Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)
esp32cam.menu.PartitionScheme.min_spiffs.build.partitions=min_spiffs
esp32cam.menu.PartitionScheme.min_spiffs.upload.maximum_size=1966080

esp32cam.menu.CPUFreq.240=240MHz (WiFi/BT)
esp32cam.menu.CPUFreq.240.build.f_cpu=240000000L
esp32cam.menu.CPUFreq.160=160MHz (WiFi/BT)
esp32cam.menu.CPUFreq.160.build.f_cpu=160000000L
esp32cam.menu.CPUFreq.80=80MHz (WiFi/BT)
esp32cam.menu.CPUFreq.80.build.f_cpu=80000000L
esp32cam.menu.CPUFreq.40=40MHz (40MHz XTAL)
esp32cam.menu.CPUFreq.40.build.f_cpu=40000000L
esp32cam.menu.CPUFreq.26=26MHz (26MHz XTAL)
esp32cam.menu.CPUFreq.26.build.f_cpu=26000000L
esp32cam.menu.CPUFreq.20=20MHz (40MHz XTAL)
esp32cam.menu.CPUFreq.20.build.f_cpu=20000000L
esp32cam.menu.CPUFreq.13=13MHz (26MHz XTAL)
esp32cam.menu.CPUFreq.13.build.f_cpu=13000000L
esp32cam.menu.CPUFreq.10=10MHz (40MHz XTAL)
esp32cam.menu.CPUFreq.10.build.f_cpu=10000000L

esp32cam.menu.FlashMode.qio=QIO
esp32cam.menu.FlashMode.qio.build.flash_mode=dio
esp32cam.menu.FlashMode.qio.build.boot=qio
esp32cam.menu.FlashMode.dio=DIO
esp32cam.menu.FlashMode.dio.build.flash_mode=dio
esp32cam.menu.FlashMode.dio.build.boot=dio
esp32cam.menu.FlashMode.qout=QOUT
esp32cam.menu.FlashMode.qout.build.flash_mode=dout
esp32cam.menu.FlashMode.qout.build.boot=qout
esp32cam.menu.FlashMode.dout=DOUT
esp32cam.menu.FlashMode.dout.build.flash_mode=dout
esp32cam.menu.FlashMode.dout.build.boot=dout

esp32cam.menu.FlashFreq.80=80MHz
esp32cam.menu.FlashFreq.80.build.flash_freq=80m
esp32cam.menu.FlashFreq.40=40MHz
esp32cam.menu.FlashFreq.40.build.flash_freq=40m
```

With this fix, the Tool menu in the Arduino IDE will show the partition table available for the OTA.

ChristopheLaurent, 2022-02-18:
Using PlatformIO, I managed to do OTA on this board by specifying the right partition.

In `platformio.ini:`

```
[env:esp32cam]
platform = espressif32
board = esp32cam
framework = arduino
build_type = debug
monitor_filters = esp32_exception_decoder
board_build.partitions = partitions_custom.csv
```

In `partitions_custom.csv`:

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x180000,
app1,     app,  ota_1,   0x190000,0x180000,
spiffs,   data, spiffs,  0x310000,0xF0000,
```

easytarget, 2022-03-10:
The AI-THINKER board definition in the IDE is most definitely not OTA compatible and offers no partition scheme selection.
You can work around this by selecting the vanilla dev board and choosing a OTA compatible partition scheme. 
I have an OTA compatible fork of the standard webcam sketch and have documented the relevant settings here: https://github.com/easytarget/esp32-cam-webserver#programming

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

beta-tester, 2022-05-03:
> Hello, can you please retest this on v2.0.3-rc1?

sorry for my late response, i couldn't try it yet.
how can i swap from regular release to release candidate v2.0.3-rc1?

VojtechBartoska, 2022-05-04:
Take a look on installing guide please. 

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

easytarget, 2022-05-07:
Ok; this appears to be resolved, in the boards manager for 2.0.3 release the `AI-Thinker ESP32-CAM` entry now exposes the partition scheme, you can select an OTA compatible partition scheme, and everything works as expected.

I tested this by selecting the board + ota partition (minimal spiffs) used for my sketch and flashing the 'BasicOTA' example over it (over the air), then re-flashing my sketch back over the example (again, over the air).

I also checked the boards definition in the 2.0.3 package's `boards.txt`, and verified that it has psram enabled.

This is easy enough to document for my projects users, I consider it closed, thankyou.

readnotify, 2023-05-28:
Here's the only working OTA for this board I've seen:  https://github.com/wjsanek/wjsanek

He's cleverly using the unused OTA partition as the SPIFFS for the camera data, which is how and why it can work.

GnomiBerlin, 2023-06-18:
Also if using the right partition for OTA usage I found out, that the ESP32-CAM board is rebooting on firmware update, if the frame buffer for big pictures is configured like: config.fb_count = 2;
if config.fb_count = 1; then all web and  Arduino OTA firmware updates are working with PSRAM usage.
Can someone explain, why the frame buffer allocation impacts the OTA function?

beta-tester, 2023-06-19:
> ... if config.fb_count = 1; then all web and Arduino OTA firmware updates are working with PSRAM usage.
Can someone explain, why the frame buffer allocation impacts the OTA function?

what is your `config.fb_location` setting, [`CAMERA_FB_IN_PSRAM` or `CAMERA_FB_IN_DRAM`](https://github.com/espressif/arduino-esp32/blob/72c41d09538663ebef80d29eb986cd5bc3395c2d/tools/sdk/esp32/include/esp32-camera/driver/include/esp_camera.h#L91) ?

me-no-dev, 2023-06-19:
when fb_count is more than 1 and you have set `CAMERA_GRAB_LATEST` the DMA+ISR of the camera are constantly triggered to get the latest frame, so that could cause some issues with caching and PSRAM usage. Generally not expected, but ESP32 does have some workarounds for issues with PSRAM. S3 should work OK regardless

beta-tester, 2023-06-19:
@me-no-dev , i just stumbled over this condition in the [esp32-camera/driver/cam_hal.c](https://github.com/espressif/esp32-camera/blob/e689c3b082985ee7b90198be32d330ce51ac5367/driver/cam_hal.c#L366-L370) code:
```c
esp_err_t cam_config(const camera_config_t *config, framesize_t frame_size, uint16_t sensor_pid)
{
    ...
#if CONFIG_IDF_TARGET_ESP32
    cam_obj->psram_mode = false;
#else
    cam_obj->psram_mode = (config->xclk_freq_hz == 16000000);
#endif
    ...
```
does it mean, when the target is ESP32, psram is not used at all, or is that something totally different?

GnomiBerlin, 2023-06-19:
@beta-tester: I used the basis from the ESP32 Camera example: CameraWebServer.
There we have:
```
.....
config.fb_location = CAMERA_FB_IN_PSRAM;
  config.jpeg_quality = 12;
  config.fb_count = 1;
  
  // if PSRAM IC present, init with UXGA resolution and higher JPEG quality
  //                      for larger pre-allocated frame buffer.
  if(config.pixel_format == PIXFORMAT_JPEG){
    if(psramFound()){  
      config.jpeg_quality = 10;
      config.fb_count = 1; //2
      config.grab_mode = CAMERA_GRAB_LATEST;
    } else {
      // Limit the frame size when PSRAM is not available
      config.frame_size = FRAMESIZE_SVGA;
      config.fb_location = CAMERA_FB_IN_DRAM;
    }

```
As the ESP32-CAM uses PSRAM psramFound()  should be true and so 
```
config.fb_location = CAMERA_FB_IN_PSRAM;
config.fb_count = 1; //not 2 for OTA ?
      config.grab_mode = CAMERA_GRAB_LATEST;
```
is the used cobfiguration.
As I tested with no PSRAM usage (either in IDE disabled or fixing this this "if psramFound()" I think the answer to your second comment is "no". In this case all OTA variants were working, independent from the frame buffer setting. (of course with an OTA usable partition scheme) 

@me-no-dev: ESP32-CAM only used ESP32, not S3. And if DMA+ISR is always triggered in case of buffer frames >1 I do not understand, why PSRAM is disturbing the OTA, which in my understanding is happening in the DRAM?

waynepiekarski, 2023-11-09:
Just for the record, if you are using arduino-cli it is possible to also get the correct partition table configured without editing any config files:
```
arduino-cli --fqbn=esp32:esp32:esp32cam compile --build-property build.partitions=min_spiffs --build-property upload.maximum_size=1966080
```