espressif/arduino-esp32 #6115: esp32s3 can't driver sph-0645 microphone sensor

yangislet, 2022-01-09:
### Board

ESP32-S3

### Device Description

I add the code `REG_SET_BIT(I2S_TIMING_REG(CONFIG_EXAMPLE_I2S_CH), BIT(9));
    REG_SET_BIT(I2S_CONF_REG(CONFIG_EXAMPLE_I2S_CH), I2S_RX_MSB_SHIFT);`  it not work , mybe  the register of esp32s3 address is different? I don't know!  however the code work in esp32D board.

### Hardware Configuration

i2s_pin_config_t pin_config = {
        .mck_io_num = I2S_PIN_NO_CHANGE,
        .bck_io_num = 12,
        .ws_io_num = 10,
        .data_out_num = I2S_PIN_NO_CHANGE,
        .data_in_num = 11,
    };

### Version

latest master

### IDE Name

Platformio

### Operating System

windows

### Flash frequency

240M

### PSRAM enabled

yes

### Upload speed

115200

### Description

the microphone can get the correct data of voice!

### Sketch

```cpp
ESP_ERROR_CHECK(i2s_driver_install(0, &i2s_config, 0, NULL));
#ifdef xx
    REG_SET_BIT(I2S_RX_TIMING_REG(0), BIT(9));
    REG_CLR_BIT(I2S_RX_CONF_REG(0), I2S_RX_MSB_SHIFT);
#else
    REG_SET_BIT(I2S_TIMING_REG(0), BIT(9));
    REG_SET_BIT(I2S_CONF_REG(0), I2S_RX_MSB_SHIFT);
#endif
    ESP_ERROR_CHECK(i2s_set_pin(0, &pin_config));
```

### Debug Message

```plain
implicit declaration of function 'I2S_TIMING_REG'; did you mean 'I2S_RX_TIMING_REG'? [-Werror=implicit-function-declaration]
implicit declaration of function 'I2S_CONF_REG'; did you mean 'I2S_TX_CONF_REG'? [-Werror=implicit-function-declaration]
```

### Other Steps to Reproduce

esp32D board is ok, but esp32s3 board is not ok.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

yangislet, 2022-01-09:
nothing at all

closed 2022-01-09 by yangislet

VojtechBartoska, 2022-01-10:
same as here, please follow S3 support issue.

Tinyu-Zhao, 2024-02-06:
> nothing at all

我也遇到了这个问题,在s3的sdk中这个定义被删除了,你可以手动加上.
#define I2S_CONF_REG(i) (REG_I2S_BASE(i) + 0x0008)
#define I2S_TIMING_REG(i) (REG_I2S_BASE(i) + 0x001c)

[image]

Tinyu-Zhao, 2024-02-06:
> nothing at all

The i2s register of s3 has been modified, and this can be used as a replacement
    // Delay by falling edge
    REG_SET_BIT(I2S_RX_TIMING_REG(I2S_PORT), BIT(1));
    // Force Philips mode
    REG_SET_BIT(I2S_RX_CONF1_REG(I2S_PORT), I2S_RX_MSB_SHIFT);