espressif/arduino-esp32 #6116: not support esp32s3 board in esp32-hal-adc.c files

yangislet, 2022-01-09:
### Board

ESP32-S3

### Device Description

when you use the Arduino as a ESP-IDF component  in [https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html](url), the file can't build

### Hardware Configuration

not need

### Version

latest master

### IDE Name

ESP-IDF  plugin in vscode

### Operating System

windows

### Flash frequency

240M

### PSRAM enabled

no

### Upload speed

115200

### Description

can't compile !

### Sketch

```cpp
not need
```

### Debug Message

```plain
../components/arduino/cores/esp32/esp32-hal-adc.c:41:2: error: #error Target CONFIG_IDF_TARGET is not supported
 #error Target CONFIG_IDF_TARGET is not supported
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

atanisoft, 2022-01-09:
ESP32-S3 is not yet supported by arduino-esp32. It is on the roadmap still.

VojtechBartoska, 2022-01-10:
Hello @yangislet, please for info about S3 support, follow #5594 task. I'm closing this one.

closed 2022-01-10 by VojtechBartoska