espressif/arduino-esp32 #6117: Connection problem via https ESP32S2

Chicco7, 2022-01-09:
### Board

ESP32S2 WROVER

### Device Description

Only the board, not pcb attached

### Hardware Configuration

Nothing attached

### Version

latest master

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

OTA or 80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Hi, I use the latest 2.02 arduino core but I have an issue with the https connect. It work well one-two time, after the connection failed.
If I use the 2.01 version all work well. 
The debug show connection failed with -1 error

I tried various option (PSRAM, no PSRAM), and various code configuration but the issue still happaning. My only solution is to downgrade to 2.01. I don't tried the 2.00.  

### Sketch

```cpp
WiFiClientSecure client;

  client.setInsecure();

  if(client.connect(server, 443)) {
    client.setTimeout(30);
    TelnetStream.println("Connected to server");
    client.println("POST /api/measurements HTTP/1.0");
    client.println("Host: *******");
    client.println("Authorization: Bearer " + String(token));
    client.println("Content-length: " + String(parameters.length()));
    client.println("Connection: Close");
    client.println("Content-Type: application/json;");
    client.println();
    client.println(parameters);
    client.stop();
  }
```

### Debug Message

```plain
If needed I can get it
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

maspetsberger, 2022-01-24:
I can confirm the same problem on ESP32-C3.
A `WiFiClientSecure` object cannot be used after closing a connection. This breaks several libraries depending on it.

The problem is caused by https://github.com/espressif/arduino-esp32/commit/f29f4485b56b8aab1bb9ecc9094fbfbfe5bedc32 and the changes introduced in lines:
https://github.com/espressif/arduino-esp32/blob/f29f4485b56b8aab1bb9ecc9094fbfbfe5bedc32/libraries/WiFiClientSecure/src/ssl_client.cpp#L49
https://github.com/espressif/arduino-esp32/blob/f29f4485b56b8aab1bb9ecc9094fbfbfe5bedc32/libraries/WiFiClientSecure/src/ssl_client.cpp#L331

This not only resets all pointers (as intended) but it also sets all other fields in the struct to 0, including `handshake_timeout`. This in turn never has a chance to complete it in 
https://github.com/espressif/arduino-esp32/blob/f29f4485b56b8aab1bb9ecc9094fbfbfe5bedc32/libraries/WiFiClientSecure/src/ssl_client.cpp#L265

VojtechBartoska, 2022-05-04:
I'm closing the issue as expired due to no answer.

If needed, please reopen it.

Thanks for understanding.

closed 2022-05-04 by VojtechBartoska