espressif/arduino-esp32 #6118: Arduino compiler string warnings when Preferences set to None

palmerr23, 2022-01-10:
### Board

DevkitC ESP32 WROOM

### Device Description

DevkitC

### Hardware Configuration

N/A

### Version

latest master

### IDE Name

Arduino

### Operating System

Win 11

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

921600

### Description

Frederic, 

Even when Arduino compiler message Preferences are set to None, messages like the following appear:

warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
       screenError("Bad EEPROM read\nFailed factory reset\nAborting", MY_RED, 10, false);

The message itself is not the issue (as it is correct), the problem is that it should not appear when compiler warnings are turned off. 

Perhaps a compiler flag is not set correctly in the board file?

It's a minor issue, but it's annoying to have to hunt through all the warnings for an actual error!

BTW: thanks for the Arduino ESP32 implementation, it generally works pretty flawlessly.

Richard

### Sketch

```cpp
void screenError(char * message, uint16_t bgcol, uint16_t onTime, bool logo)
```

### Debug Message

```plain
N/A
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-04-11:
Hello, can you please retest this on v2.0.3-rc1?

palmerr23, 2022-04-12:
Issue is still present in v2.0.3-rc1

PilnyTomas, 2022-04-12:
Hi @palmerr23, could you please provide complete sketch with all the `#include`s and other setup necessary to replicate your issue?

palmerr23, 2022-04-13:
```
void setup() {
badkitty("ABC");
}
void badkitty(char * msg){
}
void loop() {
}
![prefs](https://user-images.githubusercontent.com/14856369/163073937-8d8855b4-7710-4bbe-a002-1681e30ec1df.png)

```

C:\Users\xyzzy.DESKTOP-J4ESB3N\Documents\arduino\sketch_apr13a\sketch_apr13a.ino: In function 'void setup()':
C:\Users\xyzzy.DESKTOP-J4ESB3N\Documents\arduino\sketch_apr13a\sketch_apr13a.ino:2:11: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
 badkitty("ABC");

Arduino 1.8.19 with v2.0.3-rc1

Preferences image attached.

me-no-dev, 2022-05-09:
Closing via https://github.com/espressif/arduino-esp32/pull/6596

closed 2022-05-09 by me-no-dev