espressif/arduino-esp32 #6119: Problem with yieldIfNecessary()

zenobik, 2022-01-10:
### Board

ESP32-S2

### Device Description

BLDC driver

### Hardware Configuration

n/a

### Version

v2.0.1

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

In the main Arduino loop, above loop() function, in the file cores/esp32/main.cpp we see yieldIfNecessary() function. It works on ESP32-S2 for example as it only has one core. Function adds a 5ms delay every two seconds. Under some specific circumstances, this causes problem with loop() execution time.
I'm using SimpleFOC library, which calculates PWM duty cycles to drive BLDC motor in the main loop. Every two seconds BLDC motor stops for milliseconds, it could be felt by a hand. This was caused by yieldIfNecessary() function.
When I added another while(1) inside loop() function, so yieldIfNecessary() didn't execute, program starts working fine.

vTaskDelay should be shortened to 1ms and some warnings should be added for more demanding users.

### Sketch

```cpp
void yieldIfNecessary(void){
    static uint64_t lastYield = 0;
    uint64_t now = millis();
    if((now - lastYield) > 2000) {
        lastYield = now;
        vTaskDelay(5); //delay 1 RTOS tick
    }
}

void loopTask(void *pvParameters)
{
    setup();
    for(;;) {
#if CONFIG_FREERTOS_UNICORE
        yieldIfNecessary();
#endif
        if(loopTaskWDTEnabled){
            esp_task_wdt_reset();
        }
        loop();
        if (serialEventRun) serialEventRun();
    }
}
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2022-05-03:
Hi,

I'm closing the issue due to no response. 

If needed you can reopen it. 

Thanks for your contribution & understanding!

closed 2022-05-03 by VojtechBartoska

Andy1978, 2024-02-22:
This problem still exists in 2.0.14 or even 3.0.0-apha3

me-no-dev, 2024-02-22:
Maybe we could come up with better solution. I can agree that this is not pretty. Testing is necessary though, because I have the feeling that your sketch would not run OK if you use WiFi and force a busy loop

Andy1978, 2024-02-23:
I've made several tests (WiFi mode WIFI_STA, client sends in own timer and so on)  with the busy loop, which prevents yieldIfNecessary from getting called but haven't found yet a combination which causes problems.

I would like to know why yieldIfNecessary and it's called has been added at all. This looks like some workaround but 5 ticks (the comment says 1 tick) every 2 seconds looks rather arbitrary for me.

Attached some screenshots of just a loop and single pin toggling. You'll see the 5ms delay every 2000ms
[image]
[image]

me-no-dev, 2024-02-23:
We added this because empty/busy loop was causing watchdog resets and network issues. We could revisit this now that we are upgrading many things and can check for better behavior.

With that said...

It is never a good idea to run a busy loop on an ESP device. It is running an OS and by design it is not meant to have anything blocking for long. I'm not sure what your code is doing and what is needed, but I am pretty sure that your functionality can be done in non-blocking way by using one of the peripherals (probably RMT if you need to update the PWM values often)

PepeTheFroggie, 2024-03-29:
This can be avoided by runnung time sensitive code in another task.

Arduino:
TaskHandle_t TaskHandle_1;
xTaskCreate(MyTask1, "Task1", 2000, NULL, 1, &TaskHandle_1); 

static void MyTask1(void* pvParameters)
{
  while(1)
  {
    doyourstuffhere;
  }
}

dmitrykez, 2024-04-07:
Can yieldIfNecessary() function be overwritten to do nothing and avoid delays? 

My whole code is time sensitive as I need to control 16 PWM loops. 
I also use WIFI sometimes (enabled by request), so I can't just add while(1) in main loop() function.

me-no-dev, 2024-04-07:
You can spin another task with higher priority and do your sensitive work there

xiaopeng12138, 2024-12-08:
I have the same problem here. I am using LEDC function to generate timecode signal, i struggled over 3 days and then figured out that the issue is yieldIfNecessary(). Im using esp32c3 and only single core, so its possible for me to just disable FREERTOS_UNICORE, but it requieres recompile the arduino-esp32 which is not that convinient... 

[image]