espressif/arduino-esp32 #6120: Change GPIO to use ESP-IDF API (C)

VojtechBartoska, 2022-01-10:
This issue tracks refactoring of GPIO to use ESP-IDF API.

No related issues have been found.

sylvioalves, 2022-01-11:
@VojtechBartoska I think OneWire feature isn't working as it fails for ESP32C3 when building some DS18B20 temperature sensor. I guess it is GPIO related.

Jason2866, 2022-01-11:
@sylvioalves  DS18B20 does work with the C3. Probably your used OneWire lib is not core 2.0.x ready. See changes made in Stickbreakers OneWire lib made here https://github.com/arendst/Tasmota/commit/3cb48f1f0878acf9d59dee48f72a0fecb88f6aee

sylvioalves, 2022-01-11:
@Jason2866 I understand. However, for common users, how would they know that or which OneWire lib to use? If you type OneWire to install a new library, there happens to be more then 5 libs and at least 2 of then does not work. So what would be the recommended way of having any OneWire library working? Well, I am not following the development as is, so my question is really as a end-user one.

Jason2866, 2022-01-12:
@sylvioalves That's in general a Arduino library problem. Many do not work reliable since the are designed to be used with many different MCU. Maybe espressif can recommend a library and supports to maintain?

Jason2866, 2022-01-17:
@sylvioalves  The maintainer of Tasmota fixed the onewire lib.
https://github.com/arendst/Tasmota/commit/165880f06b478c0b6b0f3a78fa2455ee0dd39fee

sylvioalves, 2022-01-17:
@Jason2866 thanks for the heads up!

SuGlider, 2022-01-17:
@VojtechBartoska @me-no-dev
What do you think about adding an official OneWire library to ESP Arduino Framework?

me-no-dev, 2022-01-19:
@SuGlider maybe a good idea :) need to refresh my memories on the protocol. Maybe we can use a peripheral, like RMT for that

igrr, 2022-01-19:
FWIW, there's this IDF-specific RMT-based onewire driver you could reference: https://github.com/DavidAntliff/esp32-owb/blob/master/owb_rmt.c

bbinet, 2022-01-28:
@igrr this IDF-specific RMT-based onewire driver does not work with esp32-c3 though:
https://github.com/DavidAntliff/esp32-owb/issues/26

SuGlider, 2022-01-28:
I think it may not compile, but it could work with the right adjustment.

C3 has a more limited version of the RMT peripheral. It certainly needs code adjusting.

Jason2866, 2022-02-15:
Would be great to have a adopted OneWire driver for release of v2.0.3.

PR #6259 (2022-02-16): GPIO refactoring
## Summary
GPIO now using ESP-IDF API on all chips.
LEDC interrupt fix removed - no longer needed.
Edited pins_arduino.h in variants according to changes in gpio.

Known issue with GPIO interrupt on chip Esp32s3. Will be fixed in ESP-IDF.
For now GPIO interrupts work only on CORE 0.

## Impact
Close #6249
Close #6120

## Related links
closed 2022-02-16 by me-no-dev