espressif/arduino-esp32 #6124: NVS Memory Leak I know not feature request but new issue page broken

kayleejacob1, 2022-01-10:
### Related area

NVS 

### Hardware specification

ESP32rover

### Is your feature request related to a problem?

yes the issue page is broke so i put here

NVS.getstring has memory leak issue if you look at code and output below then see on the last call to getstring the memory drops every time i make this call.

Serial.printf("  before heap at %d\n", ESP.getFreeHeap());
eprom.readboot();
Serial.printf(" after heap at %d\n", ESP.getFreeHeap());

bootdata Eprom::Readboot(void) {
   
    Serial.printf(" after1 heap at %d\n", ESP.getFreeHeap());
    bootdata val;
    Serial.printf(" after 2heap at %d\n", ESP.getFreeHeap());
   // nvs.getBytes("bootdat", &val, sizeof(val));
    val.uintitype = nvs.getUInt("type", 0);
        Serial.printf(" after 3heap at %d\n", ESP.getFreeHeap());
   // Serial.print("got type");
   // Serial.println(val.uintitype);
    val.setup = nvs.getUInt("setup", 254);
    Serial.printf(" after 4heap at %d\n", ESP.getFreeHeap());
    val.area = nvs.getFloat("area", 0.0f);
    Serial.printf(" after5 heap at %d\n", ESP.getFreeHeap());
    val.pass = nvs.getString("pass", "");
    Serial.printf(" after6 heap at %d\n", ESP.getFreeHeap());
    val.ssid = nvs.getString("ssid", "");
    Serial.printf(" after7 heap at %d\n", ESP.getFreeHeap());

    val.Config1Active = nvs.getBool("conact", false);
    Serial.printf(" after 8heap at %d\n", ESP.getFreeHeap());
    val.Config1RE = nvs.getBool("conre", false);
    Serial.printf(" after 9heap at %d\n", ESP.getFreeHeap());
    val.ErrorActive = nvs.getBool("erract", false);
    Serial.printf(" after 10heap at %d\n", ESP.getFreeHeap());
    val.AirVBME = nvs.getBool("abme", false);
    Serial.printf(" after 11heap at %d\n", ESP.getFreeHeap());
    val.BME = nvs.getBool("bme", true);
    Serial.printf(" after 12heap at %d\n", ESP.getFreeHeap());
    val.firmware = nvs.getBool("fir", true);
    Serial.printf(" after 13heap at %d\n", ESP.getFreeHeap());
    val.spiffware = nvs.getBool("spiff", false);
    Serial.printf(" after 14heap at %d\n", ESP.getFreeHeap());
    val.RV = nvs.getBool("rv", false);
    Serial.printf(" after 15heap at %d\n", ESP.getFreeHeap());
    val.setver = nvs.getFloat("setver", -1.0f);
    Serial.printf(" after 16heap at %d\n", ESP.getFreeHeap());
    val.unver = nvs.getFloat("unver", -1.0f); 
    Serial.printf(" after 17heap at %d\n", ESP.getFreeHeap());
    val.freon410 = nvs.getBool("freon", true);
    Serial.printf(" after 18heap at %d\n", ESP.getFreeHeap());
    val.activated = nvs.getBool("actvat", true);
    Serial.printf(" after 19heap at %d\n", ESP.getFreeHeap());

    val.dodefaultset = nvs.getBool("ddset", true);
    Serial.printf(" after 20heap at %d\n", ESP.getFreeHeap());
    val.doset = nvs.getBool("dset", false);
    Serial.printf(" after21 heap at %d\n", ESP.getFreeHeap());
    val.dosnap = nvs.getBool("dsn", false);
    Serial.printf(" after 22heap at %d\n", ESP.getFreeHeap());
    val.douint = nvs.getBool("dun", false);
    Serial.printf(" after23 heap at %d\n", ESP.getFreeHeap());

    val.MatchingDevice = nvs.getString("mdev", "nodevice");
    Serial.printf(" after 24heap at %d\n", ESP.getFreeHeap());
   
    return val;
}

Output:
 before heap at 86152
 after1 heap at 86152
 after 2heap at 86152
 after 3heap at 86152
 after 4heap at 86152
 after5 heap at 86152
 after6 heap at 86152
 after7 heap at 86152
 after 8heap at 86152
 after 9heap at 86152
 after 10heap at 86152
 after 11heap at 86152
 after 12heap at 86152
 after 13heap at 86152
 after 14heap at 86152
 after 15heap at 86152
 after 16heap at 86152
 after 17heap at 86152
 after 18heap at 86152
 after 19heap at 86152
 after 20heap at 86152
 after21 heap at 86152
 after 22heap at 86152
 after23 heap at 86152
 after 24heap at 86120
 after heap at 86120

### Describe the solution you'd like

Fix

### Describe alternatives you've considered

qwe

mrengineer7777, 2022-01-12:
It's a feature, not a bug.  The value has to be stored somewhere in order to be returned.  Having played with the Preferences.cpp library, I can explain.

Here is the implementation of getString():
`String mPreferences::getString(const char* key, const String defaultValue){
    char * value = NULL;
    size_t len = 0;
    if(!_started || !key){
        return String(defaultValue);
    }
    esp_err_t err = nvs_get_str(_handle, key, value, &len);
    if(err){
        log_e("nvs_get_str len fail: %s %s", key, nvs_error(err));
        return String(defaultValue);
    }
    char buf[len];
    value = buf;
    err = nvs_get_str(_handle, key, value, &len);
    if(err){
        log_e("nvs_get_str fail: %s %s", key, nvs_error(err));
        return String(defaultValue);
    }
    return String(buf);
}`

During the first call "nvs_get_str(_handle, key, value, &len)" value==NULL, so the function returns the length.
Then "char buf[len]" allocates space by creating a char array.
The next call to "nvs_get_str(_handle, key, value, &len)" copies the chars into the array.
Finally "String(buf)" converts the char array to a String class.  What you get back is a pointer to that String object.
I suspect buf[] is released, but String{} (your result) stays in memory.  I wonder if the heap size goes down once you release "val" (val.MatchingDevice)?

kayleejacob1, 2022-01-12:
> just a bit offtopic question but what do you mean by "yes the issue page
> is broke so i put here"? You mean issue template?

No problem tried filling it out yester day 5 or 6 times but the submit
button stayed greyed out and i check several times and reloaded several
time to make sure everything was filled out but still did not work

VojtechBartoska, 2022-04-11:
Is this still valid issue?

VojtechBartoska, 2022-05-18:
@kayleejacob1 any updates?

mrengineer7777, 2023-03-29:
Closing stale issue

closed 2023-03-29 by mrengineer7777